esp8266/Arduino #2221: HTTP_POST Problem

llams, 2016-07-04:
### Basic Infos

#### Hardware
Hardware:			?ESP-12F?
Core Version:      	?git master?

### Description

HTTP_POST Problem
Does not work the return of status after upgrading the from the library ESP8266HTTPUpdateServer.
### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?

### Sketch
ESP8266HTTPUpdateServer/examples/WebUpdater/

igrr, 2016-07-04:
This looks like a regression caused by recent WiFiClient refactoring. Now that WiFiClient doesn't prevent Nagle algorithm from being used, some TCP segments may end up not being sent before restart is called.
Affects ArduinoOTA as well.

reaper7, 2016-07-04:
this problem exists too on "traditional" ota upadate (2.3-latest github)
new firmware loaded to device propetrly but arduino ide still waiting for confirmation and after some time give [ERROR]: No Result!

igrr, 2017-05-09:
Created #3219 to fix this

HarmOtten, 2017-12-07:
I have a very similar issue. When uploading the example BasicOTA sketch via OTA, the new firmware is uploaded 100%, and ESP reboots, but ArduinoIDE (Running in windows7) keeps waiting.
I use ESP8266 v2.4.0-rc2 from github...

closed 2018-01-02 by igrr

delwinbest, 2019-01-11:
> I have a very similar issue. When uploading the example BasicOTA sketch via OTA, the new firmware is uploaded 100%, and ESP reboots, but ArduinoIDE (Running in windows7) keeps waiting.
> I use ESP8266 v2.4.0-rc2 from github...

HarmOtten did you find the cause? I have this problem repeatedly:

20:24:27 [INFO]: Sending invitation to: 192.168.10.103
20:24:27 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

20:24:39 [INFO]: Waiting for result...
20:26:42 [ERROR]: No Result!

20:26:45 [INFO]: Waiting for device...
Uploading: [============================================================] 100% Done...

20:26:57 [INFO]: Waiting for result...

Where Upload size: 335056
Any ideas?

HarmOtten, 2019-01-11:
@delwinbest,
 I currently use ESP8266 v2.4.2 and I do not have the issue anymore.