esp8266/Arduino #2225: ESP8266WebServer::collectHeaders can not be used from libraries

igrr, 2016-07-04:
`collectHeaders` methods expects the headers list to be supplied in a single call. This effectively prevents other libraries from using `collectHeaders`. I.e. ESP8266HTTPUpdateServer should be able to request Content-Length header independently from other headers which may be requested from the sketch.

Add `collectHeader` method to collect just one header. Replace C array (`_currentHeaders`) with `std::map<String, String>` and `insert` into the map. Make `collectHeaders` call `collectHeader` for each array item.

me-no-dev, 2016-08-15:
how about we use [this](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/StringArray.h)

me-no-dev, 2016-09-02:
Ivan, please check this PR and tell me what you think

closed 2018-01-02 by igrr