esp8266/Arduino #2227: Add ESP-NOW library

igrr, 2016-07-04:
Add an Arduino-style wrapper around ESP-NOW library and at least some documentation explaining what ESP-NOW is.

vanjo9800, 2016-07-07:
Is there any documentation available somewhere because I need to learn how the encryption of the messages sent works?

roboticboyer, 2016-08-05:
There is a prototype code at [https://github.com/HarringayMakerSpace/IoT/tree/master/ESP-Now]
The documentation is [https://espressif.com/sites/default/files/documentation/esp-now_user_guide_en.pdf]

ESPlorador, 2017-01-07:
I have explained al the functions with examples at http://www.esploradores.com/practica-6-conexion-esp-now/

Its in Spanish. I´m sorry.

devyte, 2017-10-17:
@igrr The linked guide is actually not bad, It could be condensed into a document and examples could be built using the code there as basis.
However, I somehow don't see this making into 2.4.0. I suggest pushing it back.

yoursunny, 2018-01-02:
I made an ESP-NOW library. You may try it while waiting for the official library.
https://github.com/yoursunny/WifiEspNow

devyte, 2018-01-03:
@igrr @d-a-v I briefly looked at @yoursunny 's implementation. I would request several changes, but overall it looks sound.
How should this be handled? Possibilities:
-we integrate it into our libs
-we add it as a subrepo, similar as SoftwareSerial
-we just document the existence of the external repo and point to it

igrr, 2018-01-03:
Linking to this library from libraries.rst sounds like a good first approach. The library can be easily installed via boards manager.

If one of the contributors becomes familiar enough with the library (to be able to debug issues with it), then pulling it into the project might be justified.

mischmerz, 2018-07-06:
It's very easy to do ESP now with a simple

mischmerz, 2018-07-06:
It's very easy to do basic ESP now with a simple #include <espnow.h> (extern C) . Works quite nice, as long there's no special crypto involved. There seems to be a bug in Espressif 's core allowing controllers to send unencrypted messages even IF the slave sets it's peer to a special crypto key.

PR #5322 (2018-11-08): Update libraries.rst with ESP-NOW
Fixes #2227
commit 53edcfd9 (2018-11-08): Update libraries.rst with ESP-NOW (#5322)

Fixes #2227
closed 2018-11-08 by devyte