esp8266/Arduino #2229: Unable to change mDNS name

Ediolot, 2016-07-04:
### Basic Infos

Only able to connect to ` http://esp_d2b4d7.lan/`.

#### Hardware
Hardware:			ESP-01
Core Version:      	Stable 2.3.0

### Description

I was trying to make it work the example provided in `ESP8266mDNS/mDNS_Web_Server` (Only ssid and password are edited) but I was not able to connect to the module using `http://esp8266.local/` . Instead,` http://esp_d2b3d7.lan/` works just right.

Im assuming that if I change the `mdsn.begin(...)` parameter I should be able to call it whatever I would like to. Isnt that right or I am missing something here ?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512K (64K SPIFFS)
CPU Frequency:	80Mhz
Flash Mode:		DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Debug Messages

Serial output: 
```
þ
..
Connected to SomeNetwork
IP address: 192.168.2.177
mDNS responder started
TCP server started
```

nslookup command in ubuntu:
```
ragnarok@ubuntu:~$ nslookup 192.168.2.177
Server:		127.0.1.1
Address:	127.0.1.1#53

Non-authoritative answer:
177.2.168.192.in-addr.arpa	name = ESP_D2B4D7.lan.

Authoritative answers can be found from:
```

Screenshots: 

https://cdn.pbrd.co/images/7zyX1inyL.png
https://cdn.pbrd.co/images/7zz8n9ZqS.png

pipi61, 2016-07-04:
On PC installed mdns software?
https://en.wikipedia.org/wiki/Avahi_(software)

igrr, 2016-07-05:
Tried mDNS_Web_Server example (changed SSID and password), works fine.
Here is the output i get on Ubuntu 14.04 desktop:

```
ivan@castor:~$ avahi-browse --resolve --parsable _http._tcp
+;eth0;IPv4;esp8266;Web Site;local
=;eth0;IPv4;esp8266;Web Site;local;esp8266.local;192.168.77.24;80;
```

```
ivan@castor:~$ curl --noproxy *.local http://esp8266.local
<!DOCTYPE HTML>
<html>Hello from ESP8266 at 192.168.77.24</html>
```
(I have `--noproxy` option because i have proxy server set through environment variables)

Ediolot, 2016-07-05:
Thanks for your answers. I have been taking a look at this (My first time working with mDNS and I was still learning how DNS works) and it is kind of working now. 

I am working on a Windows PC with a virtual Ubuntu machine (Bridged connection). Now, the Ubuntu machine can reach the esp8266 server in http://esp8266.local (I honestly dont know why it didnt work before). 

But I mainly use Windows so I installed Bonjour Browser too see if it was able to locate the module:

```
Available Bonjour Services:
Type: _http._tcp 
Name: esp8266

IP Addresses
192.168.2.177:80 (Esp address)
```
But still only `http://esp_d2b4d7.lan ` (domain name given by the router as DNS server?) was working and `http://esp8266.local ` returns _ERR_NAME_NOT_RESOLVED_. I found that the mDNS protocol is not fully implemented (??) in windows and you have install the [Bonjour software](https://support.apple.com/kb/DL999?viewlocale=en_US&locale=es_ES)  for printers (for example). 

I took this information mostly from [here](http://www.howtogeek.com/167190/how-and-why-to-assign-the-.local-domain-to-your-raspberry-pi/):

> If you’re running a Windows machine that does not have iTunes installed (which would have installed a companion Bonjour client for mDNS resolution), you can resolve the lack of native mDNS support by downloading Apple’s Bonjour Printer Service helper app here. Although the download page makes it sound like it’s a printer-only tool, it effectively adds mDNS/Bonjour support across the board to Windows.

Anyway it seems to have a lot of problems during installation in windows. But somehow I managed to run the process one time and __it worked !__

closed 2017-02-22 by Ediolot