esp8266/Arduino #2232: WPS problem

craon7, 2016-07-05:
### Basic Infos

WPS does not work on BT (UK provider) router

#### Hardware
Hardware:			Amica esp8266 ESP 12E
Core Version:      	        2.2.0 and 2.3.0

### Description

I have a problem with WPS, I cannot connect to BT Home Hub 5 router. I can successfully connect to 3 different routers but only that BT Home Hub 5 makes problems. I tried 2 different Home Hub 5 routers - same story. Windows 10 can connect to it via WPS with no problems. 

I am using code from samples I have found on github.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		 SERIAL
Reset Method:		nodemcu

### Sketch

```

#include <ESP8266WiFi.h>

bool StartWPSPBC() 
{
  Serial.println("WPS config start");

  WiFi.mode(WIFI_STA);
  //WiFi.setAutoConnect(true);
  //WiFi.setAutoReconnect(true);

  delay(5000);
  bool wpsSuccess = WiFi.beginWPSConfig();

  delay(2000);

  if (wpsSuccess) {
    
    delay(2000);
    Serial.println("WPS success: " + String(WiFi.status()));
    Serial.println("SSID" + String(WiFi.SSID()));
    Serial.println("PSK" + String(WiFi.psk()));    
    
    // Well this means not always success :-/ in case of a timeout we have an empty ssid
    String newSSID = WiFi.SSID();
    if (newSSID.length() > 0) {
      // WPSConfig has already connected in STA mode successfully to the new station. 
      Serial.printf("\nWPS finished. Connected successfull to SSID '%s'\n", newSSID.c_str());
    }
    else {
      wpsSuccess = false;
      Serial.println("WPS finished. SSID empty. ");
    }
  }
  return wpsSuccess;
}

void setup() {
  
  Serial.begin(115200);
  Serial.setDebugOutput(1);

  Serial.println("");
  Serial.println("starting ... ");

  delay(5000);

  StartWPSPBC();
}

void loop() {
  
  if (WiFi.SSID().length() > 0 ) 
    {
      Serial.printf("\nTry connecting to WiFi with SSID '%s'\n", WiFi.SSID().c_str());
      
  
      WiFi.mode(WIFI_STA);
      WiFi.begin(WiFi.SSID().c_str(), WiFi.psk().c_str()); // reading data from EPROM, last saved credentials
      while (WiFi.status() == WL_DISCONNECTED) {
        delay(500);
        Serial.print(".");
      }
  
      wl_status_t status = WiFi.status();
      if (status == WL_CONNECTED) {
        Serial.printf("\nConnected successfull to SSID '%s'\n", WiFi.SSID().c_str());
        
      }
      else 
      {
        Serial.printf("\nCould not connect to WiFi. state='%d'", status);
       
      }
    }

  delay (10000);
}
```

### Debug Messages

```
starting ... 
WPS config start
wps begin
wifi_wps_enable
wps scan
build public key start
build public key finish
f r0, wps discover [BTHub5-25N6]
scandone
WPS: neg start
f r0, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
state: 5 -> 0 (2)
rm 0
reconnect
f r0, wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
state: 5 -> 0 (2)
rm 0
reconnect
f r0, wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
wps cb status: 2
wps TIMEOUT
wps discover [BTHub5-25N6]
state: 5 -> 0 (0)
rm 0
WPS: neg start
f r0, wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
WPS success: 6
SSIDBTHub5-25N6
PSK

WPS finished. Connected successfull to SSID 'BTHub5-25N6'

Try connecting to WiFi with SSID 'BTHub5-25N6'
sta config unchanged........................................................................state: 5 -> 0 (2)
rm 0
reconnect
f r0, wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
.
Could not connect to WiFi. state='4'
Try connecting to WiFi with SSID 'BTHub5-25N6'
sta config unchanged
Could not connect to WiFi. state='4'
Try connecting to WiFi with SSID 'BTHub5-25N6'
sta config unchanged
Could not connect to WiFi. state='4'
Try connecting to WiFi with SSID 'BTHub5-25N6'
sta config unchanged
Could not connect to WiFi. state='4'state: 5 -> 0 (2)
rm 0
reconnect
f r0, wifi evt: 1
STA disconnect: 204
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 
```

igrr, 2016-07-05:
This part of above output indicates that SSID is `BTHub5-25N6` and PSK is empty:
```
SSIDBTHub5-25N6
PSK
```

Is that actually true?

craon7, 2016-07-05:
Yes, exactly.

igrr, 2016-07-05:
What encryption mode is set to on the router?
Could you please try with a PSK of at least 8 characters? I don't think empty PSK is supported for WPA2, at least.

craon7, 2016-07-05:
Hi
Encryption is set to "WPA2 only", PSK length is 10 characters.

igrr, 2016-07-05:
Okay, when I asked "Is that actually true" above I was referring to the real values, i.e. i was surprised that PSK set on router was somehow empty. It is clear now, so WPS doesn't save PSK into flash config.
I'll check whether it should be, in fact, saving PSK there.

craon7, 2016-07-05:
OK, I see, When I connect to different router, I am sure that PSK is saved into flash memory. I tested 3 other routers and above code was returning SSID and PSK. Those settings were remembered after module restart as well.

craon7, 2016-07-05:
here I have a debug from Draytek router where above code is working:

`
starting ... 
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
chg_B1:-40
WPS config start
state: 5 -> 2 (fc0)
rm 0
wifi evt: 1
STA disconnect: 15
reconnect
state: 2 -> 0 (0)
f r-40, scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with draytek, channel 1
dhcp client start...
wifi evt: 0
state: 5 -> 0 (0)
rm 0
wps begin
wifi_wps_enable
wps scan
build public key start
build public key finish
f r-40, wps discover [draytek]
scandone
WPS: neg start
f r-40, wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 
process pubkey start
process pubkey finish
WPS: key[MySecretPassword]  
wps finished
wps cb status: 0
wifi_wps_disable
state: 5 -> 0 (0)
rm 0
f r-40, wifi evt: 1
STA disconnect: 8
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with draytek, channel 1
dhcp client start...
wifi evt: 0
ip:192.168.1.10,mask:255.255.255.0,gw:192.168.1.1
wifi evt: 3
WPS success: 3
SSID draytek
PSK MySecretPassword

WPS finished. Connected successfull to SSID 'draytek'

Try connecting to WiFi with SSID 'draytek'
sta config unchanged
Connected successfull to SSID 'draytek'
pm open,type:2 0
`

If you compare both debugs you will notice that first debug (the one which cannot connect) stops at:
`
add 0
aid 1
cnt 
`
and then repeats few times?? there is no 'process pubkey start' logic running.

Thanks for looking at it.

timw1971, 2017-06-15:
I have exactly this issue with a BT Hub 5. WPS gets the SSID almost immediated, but not the 10-character password, and the WPS callback receives WPS_CB_ST_TIMEOUT after about 2 minutes.

Was there ever a resolution to this???

craon7, 2017-06-15:
Nope

devyte, 2017-10-16:
@craon7 some googling hints that the BT home hub 5 may have issues with spaces in the password, especially leasing or trailing spaces (i.e.: spaces that may have been left before or after your real password).
Given that WPS works with other routers, this issue looks like either a bug on the router side, or an incompatibility/bug within the SDK, in which case there is nothing we can do in this repo (the wps code is a thin wrapper around SDK api calls).
Closing.

closed 2017-10-16 by devyte