esp8266/Arduino #2235: WiFi Reconnet only after power cycle

DasBasti, 2016-07-05:
### Basic Infos
Connecting to Wifi does only work after powercycle. After Reset via external button or programming Wifi does not get an IP adress. DHCP Server shows updated lease!
#### Hardware
Hardware:			ESP-12E
Core Version:      	2.1.0-rc2

### Description
Code as in example

After initial power up the connection is established. After reset it stays at the `WiFi.status() != WL_CONNECTED` part and prints out dots.

The router has an entry for a refreshed lease for the MAC adress.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp
void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("MAC: ");
  Serial.println(WiFi.macAddress());
  Serial.print("Connecting to ");
  Serial.println(wifiName);

  WiFi.begin(wifiName, wifiPass);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}
```

### Debug Messages
```
Connecting to TP-LINK_3E08
......
WiFi connected
IP address: 
192.168.0.111
```
after pressing RESET button
```
Connecting to TP-LINK_3E08
..........................................................................................................
```

igrr, 2016-07-05:
2.1.0-rc2 is pretty old, could you please try with 2.3.0?

DasBasti, 2016-07-05:
ok, tried with 2.3.0 still no difference. updated above submission.

SalCimmino, 2016-08-04:
Same issue here. Any update? Thanks.

MartijnBraam, 2016-09-22:
I'm having the same issues. This is not only fixable by rebooting but also by removing the wifi connection in the router that you connect to (or rebooting the router if that option is not available). Since the ESP doesn't send a DEAUTH message to the router so it cannot connect again.

drmpf, 2016-09-22:
Try V2.2.0,  it fixed my issue with WiFi.status() != WL_CONNECTED

SalCimmino, 2016-09-22:
It may be simplistic, but this is how I fixed it. 
I call this routine when I need to connect. Basically send a disconnect first.

`
 void startWIFI(void) {

       WiFi.disconnect();
       
       WiFi.mode(WIFI_STA); // added in V 3.1a to disable AP_SSID publication in Client mode - default was WIFI_AP_STA
       WiFi.begin(WIFI_SSID, WIFI_PASS);  // connect to local WIFI Access Point
       digitalWrite(BLNK_PIN, HIGH);
       delay(1000);

 
}

`
It works great for me.

krzychb, 2016-09-22:
@MartijnBraam - release 2.3.0 suffers from specific issue to establish Wi-Fi connection that looks similar to what you observe. The root cause and solution is provided in https://github.com/esp8266/Arduino/issues/2186#issuecomment-228581052

vijaysy, 2016-10-10:
Hi All , we were also facing same issue 
we tried below solution 
``` WiFi.mode(WIFI_STA);
  WiFi.mode(WIFI_STA); 
   WiFi.begin(WIFI_SSID, WIFI_PASS); 
   delay(1000);```

It did not work for us, (I am not sure whether it is a right way to fix this issue)
But when I reduced Address Lease Time of DHCP to 1 min in router setting of my router it got connected without restarting my router.

markus-fischbacher, 2017-04-15:
Same problem here. Any news about a solution?

devyte, 2017-10-16:
@DasBasti is this issue still valid with latest git?

DasBasti, 2017-10-20:
I have no way to check. The project went in another direction.

devyte, 2017-10-20:
I don't observe this, but then again my dhcp setup is a bit special.
Closing. If there is still a problem, please open a new issue and fill in the requested info.

closed 2017-10-20 by devyte

Neoster, 2018-01-28:
I have a small project using the noceMCU that holds the ESP8266MOD.
I do experience the same issue; I can not re-connect to my WiFi AP once I have already connected to it, and disconnected from it. I only want to connect once specific statements are met. None of the above mentioned fixes fixed it for me.

ofekp, 2018-08-18:
@SalCimmino At https://github.com/esp8266/Arduino/issues/2235#issuecomment-248916617, after the `disconnect()` I placed a delay (1.2 sec) and now this works. Before using the delay it stalled forever trying to connect to the WiFi.

d-a-v, 2018-08-18:
This is interesting. Can you print `WiFi.status()` every 100ms 13 times after `WiFi.disconnect()` ?
If the value is predictable, that test could be included in a variant of `WiFi.disconnect()`.

ofekp, 2018-08-20:
```
  0 ms => status [0]
  100 ms => status [0]
  200 ms => status [0]
  300 ms => status [0]
  400 ms => status [0]
  500 ms => status [0]
  600 ms => status [0]
  700 ms => status [0]
  800 ms => status [0]
  900 ms => status [0]
  1000 ms => status [0]
  1100 ms => status [0]
  1200 ms => status [0]
  1300 ms => status [0]
```
sometimes I got status 3 for 0 ms which is WL_CONNECTED.
**And yet**, when I remove the delay completely after disconnect() and then being(), my program stalls on `while (WiFi.status() != WL_CONNECTED)`

```
connectWiFi
................................................................... ...
```

It stalls forever, so add as many points as you like :)

Swedish-Coder, 2018-12-19:
Having a simular (or the same) issue happening.
Using the following Code

```
void Reconnect(){
WiFi.disconnect(true);
			WiFi.mode(WIFI_STA);
			
			While(WiFi.status() != WL_CONNECTED){
//do nothing, yield a bit...
Yield();
			}
			Serial.println(String(F("[WiFi] Connect now... heap: ")) + String(ESP.getFreeHeap()) + String(F(", ts: ")) + String(millis()));
		
			//… load ssid & pass from spiffs
			WiFi.begin(ssid.c_str(), psk.c_str());
}

```

This works fine 99% of the time. Running this test on ~50 ESP8266. Reconnect is called every 30seconds. 
After some undefined time (ranging from 1h to 10days..) the Wifi will not be able to reconnect to the Wlan.
The Status stays like this until the device is reset (hardware reboot). Rebooting the AP does not help.
The signal strength is also not the issue (ran a test with AP in front of 42 ESP8266 with ~2m away from devices, they all had ~ -65 dba). 

Have tried various approaches such as
```
WiFi.disconnect(true);
 	forceSleepBegin(35000); 
	//…do other tasks for 25ms
	WiFi.forceSleepWake();
WiFi.persistent(false);
…
WiFi.begin(ssid.c_str(), psk.c_str());

```
This sometimes causes crashes on the WiFi.begin(..) line.

Is there a way to “properly” do a reconnect on WiFi, or a way to reset the WiFi to “hardware boot” status (delete all DHCP information, known AP, etc.)?

maxbanton, 2019-01-01:
Hi! I ran into a similar problem and found that described issue related to DHCP lease time. After increasing the lease time, my NodeMCU v2 fall off after 24 hours, previously there were 60 minutes, which is consistent with the settings of the DHCP server. It is likely that after the expiration of the lease of the IP address, the board cannot fulfill the request for a new lease. Recently I excluded the level of the Wi-Fi signal, which is at the level of -60 dBm, as well as the network delays that make up 4ms on average. Will try to add `WiFi.disconnect();` call as @SalCimmino mentioned, but I wonder why this does not happen automatically.

Regards.

d-a-v, 2019-01-17:
This thread is closed, please continue the discussion in #5527.