esp8266/Arduino #2238: analogWrite(pin, value); causes a reboot module

llams, 2016-07-06:
### Basic Infos

#### Hardware
Hardware: ESP-12F
Core Version: git master

### Description

analogWrite(pin, value); causes a reboot module.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	40Mhz

igrr, 2016-07-06:
Could you please provide the full sketch code? If the module reboots, please check serial output for the stack dump, then decode the stack dump and paste the result here. Thanks.
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

llams, 2016-07-06:
listing is very large. I tried a simple example works correctly. At the same time on the core 2.3.0 all fine.
When setting any value for PWM LED starts to flicker slightly, and after some time the module is restarted.
In serial output only:
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld

devyte, 2017-10-16:
@llams No MCVE sketch, you didn't specify which pin.
 I am using analogWrite() with latest git, and I can't reproduce this (ref: WittyCloud pins 12/13/15).
- You have to set the pinMode to OUTPUT first (not doing it doesn't cause a reset, though).
- There are certain pins that can't be used, e.g.: pin 9 on an ESP12F.
- You have to mind your power supply. For development I use a 5V20A supply (yes, 20 amps) to make sure that any reboots are not due to voltage dips.
Closing as can't reproduce.

closed 2017-10-16 by devyte