esp8266/Arduino #2242: Reconnect Issues

kaeferfreund, 2016-07-07:
With the following sketch I have trouble reconnecting to the web server (from the browser) sometimes.  There is no wdt reset, I just can't reach the ESP for a while. Then a few hours later (no changes made) it might work again.
When I repower the module, it's instantly reachable again.
```
setup(){
 byte wifiCNT = 0;
 while (WiFi.status() != WL_CONNECTED && wifiCNT < 25) {
      delay(500);
      wifiCNT++;
 }
}
```

If I change that it to this, all problems are gone and there is no dropouts at all.
```
setup(){
while (WiFi.status() != WL_CONNECTED) {
      delay(500);
 }
}
```

For now this solves the problem, but I'm still wondering why?

In other words: How do set a timeout, so that when there is not network available during setupso the ESP goes to normal operation (loop) without the wifi connection and connects later...?

#### Hardware
Hardware:		ESP-12e
Core Version:      	todays-Git Version

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:	160Mhz
Flash Mode:		qio
Flash Frequency:	80Mhz
Upload Using:		OTA
Reset Method:		ck

liquidfalcon, 2016-07-07:
If you have `persistent` enabled, the SDK handles all of this automatically. You don't ever need to call `begin()` or any WiFi functions, unless you need to change modes, ssid, etc. Just jump straight to your code, and watch the status to determine if you should hold off on networking activities. 

Something like:

```
void loop() {
    if (WiFi.status() == WL_CONNECTED) {
         // Do networking code 
    } 
    // Do code that doesn't need networking
}
```

devyte, 2017-10-16:
@kaeferfreund the previous comment is correct. Also, it is best to use the WiFi events to figure out if there is connectivity or not (as opposed to the while loop + delay(500) ). You can use a counter there in a similar manner.
Finally, I doubt that just removing the counter is the fix to your problem. Most likely your problem is elsewhere, and removing the counter changes the symptom.
Without a MCVE sketch, I can't help further.
Closing per #3655 .

closed 2017-10-16 by devyte