esp8266/Arduino #2245: Problem:  getVCC reading lower values since the last update...

egzola, 2016-07-07:
I've been testing in many ESP-12F, and I can guarantee that since the last software update , getVcc function has returned a lower 0.3V value than the previous software version.

igrr, 2016-07-08:
There was an SDK update between these two versions. Apparently adc read routine from the SDK got "fixed".

wyasuda, 2017-01-08:
Hi,
I have the same problem with 2.3.0.
When I measured Vcc with a voltmeter, it was 3.3V.
However the value I got from getVcc function was 3.02V.
It was lower than real value.
Would you please tell me why this might happen?

Humancell, 2017-03-27:
Has anything been done on this?  I too am seeing ~0.2-0.3v low reading on all of my ESP-12F.

devyte, 2017-10-15:
There is nothing much that can be done within this repo. The issue seems to be in the SDK side.
I suggest implementing a linear calibration via software, which is something that should be done anyways in these cases.
Closing.

closed 2017-10-15 by devyte