esp8266/Arduino #2247: WifiMulti.addAP code inspection : Incoherent test for max length of passphrase

SteveToulouse, 2016-07-07:
### Description

WifiMulti.addAP limits the passphrase to 63 characters by the test at [L175](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiMulti.cpp#L174).

On the other hand, Wifi.begin recognises a length of 64 chars as a PSK instead of a passphrase, see [L118](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp#L118).

This seems to be incoherent (wifimulti.run calls wifi.begin) unless we aren't allowed to pass PSK to wifimulti by design? 

NOTE : SSID is handled the same in both places (31 char max).

zenmanenergy, 2017-05-12:
According to [wl_definitions.h](https://github.com/esp8266/Arduino/blob/4897e0006b5b0123a2fa31f67b14a3fff65ce561/libraries/ESP8266WiFi/src/include/wl_definitions.h#L32) it looks like it should be 63. Does that mean the WifiMulti.addAP is correct with 63 and the ESP8266WiFiSTA.cpp is incorrect with 64?

devyte, 2017-10-15:
Per Espressif API reference v2.1.2, section 7.2.1:
char	password[64]

The above includes the null terminator, though.

devyte, 2018-01-01:
It took me a while to fully understand what the correct behavior is supposed to be, and what is right/wrong in the code. At irst I thought this was just a boundary condition that needed fixing. As it turns out, there are two cases for passphrases: 63 char printable ascii with null terminator, and 64 byte hex with/without null terminator.
WiFi implements both cases, with certain assumptions for the 64byte hex case, while WiFiMulti implements only the 63 ascii case.
The Espressif SDK field for the passphrase is 64bytes in size, so the hex case must be handled in a special manner, and not as a null terminated string.
This is not an inconsistency or a bug, but a limitation in the implementation, of WiFiMulti, and is therefore an enhancement. Because there are certain assumptions made in WiFi, which are not quite clear, and which must be propagated, I consider the enhancement in WiFiMulti a bit delicate.
@igrr I suggest moving this to 2.5.0.

devyte, 2018-01-03:
@SteveToulouse @zenmanenergy could someone please test this?

devyte, 2018-01-11:
Closed via #4076 .

closed 2018-01-11 by devyte