esp8266/Arduino #2248: ESP8266mDNS crash when doing MDNS.queryService("arduino", "tcp")

martinayotte, 2016-07-07:
### Basic Infos

#### Hardware
Hardware:			Wemos-D1 Mini (ESP-12)
Core Version:      	Stable 2.3.0

### Description

ESP8266mDNS crash when doing MDNS.queryService("arduino", "tcp")
but doesn't with other service like MDNS.queryService("telnet", "tcp")

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Stacktrace

x4020e551: MDNSResponder::_parsePacket() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:656
0x4020e489: MDNSResponder::_parsePacket() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:611
0x40101daa: pp_post at ??:?
0x40101daa: pp_post at ??:?
0x40105087: lmacRxDone at ??:?
0x40101daa: pp_post at ??:?
0x40102abd: trc_NeedRTS at ??:?
0x40102c4c: trc_NeedRTS at ??:?
0x40102c4c: trc_NeedRTS at ??:?
0x40102c4c: trc_NeedRTS at ??:?
0x4010308a: wDev_ProcessFiq at ??:?
0x4000050c: ?? ??:0
0x40102e24: wDev_ProcessFiq at ??:?
0x402215a5: ieee80211_ht_updateparams at ??:?
0x40244668: etharp_send_ip at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c:435
0x40101daa: pp_post at ??:?
0x40105087: lmacRxDone at ??:?
0x4024494d: etharp_output_to_arp_index at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c:890
0x40102c4c: trc_NeedRTS at ??:?
0x40101daa: pp_post at ??:?
0x40101a55: ppEnqueueRxq at ??:?
0x4010193e: ppEnqueueRxq at ??:?
0x4010170b: ppProcessTxQ at ??:?
0x40101daa: pp_post at ??:?
0x40105087: lmacRxDone at ??:?
0x40102abd: trc_NeedRTS at ??:?
0x4010020c: _umm_free at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/umm_malloc/umm_malloc.c:1287
0x40102b6f: trc_NeedRTS at ??:?
0x4000050c: ?? ??:0
0x40102e24: wDev_ProcessFiq at ??:?
0x4020e88d: MDNSResponder::update() at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:219
0x4021689b: std::_Function_handler<void (), std::_Bind<std::_Mem_fn<void (MDNSResponder::*)()> (MDNSResponder*)> >::_M_invoke(std::_Any_data const&) at /home/martin/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional:2073
0x40000f58: ?? ??:0
0x4020c6e4: UdpContext::_s_recv(void*, udp_pcb*, pbuf*, ip_addr*, unsigned short) at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp:225
0x40244140: udp_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/udp.c:343
0x401070a4: pvPortMalloc at /home/martin/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266/heap.c:13
0x4024588c: ip_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip.c:553
0x40244c85: ethernet_input at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/netif/etharp.c:1379
0x4021b483: pp_tx_idle_timeout at ??:?
0x4021b3f6: pp_tx_idle_timeout at ??:?
0x4022f487: ets_snprintf at ??:?
0x40000f49: ?? ??:0
0x40000f49: ?? ??:0

me-no-dev, 2016-07-07:
what is the exception number/cause?

martinayotte, 2016-07-07:
Soft WDT ets Jan 8 2013,rst cause:2, boot mode:(3,6)

I've got the crash with MDNS.queryService("http", "tcp") too.

me-no-dev, 2016-07-07:
that is not the cause :) the cause was at the top of the stack trace (above the actual trace)
exceptions are outlined here: https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.md

martinayotte, 2016-07-07:
Is that what you means (just above the stacktrace) ?

Soft WDT reset

ctx: sys 
sp: 3ffffa60 end: 3fffffb0 offset: 01b0

martinayotte, 2016-07-07:
BTW, with HTTP, it is intermittent, sometimes it doesn't crash.

liquidfalcon, 2016-07-07:
Same issue as #2169 and #2020 - It's hitting an infinite loop when it's iterating the linked list of answers, and never breaks, causing the WDT to bark

martinayotte, 2016-07-08:
Glad to see that I'm not the only one facing this bug ...

me-no-dev, 2016-07-28:
Can you guys test [this PR](https://github.com/esp8266/Arduino/pull/2333)?
Let me know if you still get resets
```
mDNS query _arduino._tcp
2 service(s) found
1: (192.168.254.182:8266) esp-test
2: (192.168.254.156:22) piduino1
mDNS query _http._tcp
2 service(s) found
1: (192.168.254.182:80) esp-test
2: (192.168.254.194:80) HP LaserJet CP1025nw
mDNS query _ipp._tcp
1 service(s) found
1: (192.168.254.194:631) HP LaserJet CP1025nw
mDNS query _afpovertcp._tcp
1 service(s) found
1: (192.168.254.49:548) Hristo's Time Capsule
mDNS query _smb._tcp
1 service(s) found
1: (192.168.254.49:445) Hristo's Time Capsule
mDNS query _ftp._tcp
0 service(s) found
```

martinayotte, 2016-07-28:
Hi me-no-dev,
I still getting the same Soft WDT crash when using MDNS.queryService("arduino", "tcp") or MDNS.queryService("http", "tcp").

me-no-dev, 2016-07-28:
can you please enable RX debug in the cpp file in the lib and post some output?
I tried to cover all possible lockups that I came by...

martinayotte, 2016-07-28:
Here is some output with the MDNS_DEBUG_RX : 

`Reading answers RX: REQ, ID:0, Q:0, A:5, NS:0, ADD:0
                                                     8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 000c rdlength: 9
                      PTR 9 domino�

type: 0010 rdlength: 38
                       TXT 38 board=domino boarddistro_version=0.1
type: 0021 rdlength: 15
                       SRV 6 64 6f 6d 69 6e 6f 
                                               domino
                                                     type: 001c rdlength: 16
                                                                            Ignoring unsupported type 78
                                                                                                        type: 0001 rdlength: 4
                                                                                                                              All answers parsed, adding to _answers list..
RX: REQ, ID:0, Q:1, A:0, NS:0, ADD:0
                                    REQ: _arduino._tcp.local.   PTR   IN 
                                                                         Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
                                                                                                                              8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 000c rdlength: 41
                       PTR 41 ESPIRBlaster-A6523_arduino_tcplocal
 19 ESPIRBlaster-A65234
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0010 rdlength: 62
                       TXT 62 
ssh_upload=noboard=ESP8266_ESP01auth_upload=no
 19 ESPIRBlaster-A65234
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0021 rdlength: 33
                       SRV 19 65 73 70 69 72 62 6c 61 73 74 65 72 2d 61 36 35 32 33 34 
                                                                                       espirblaster-a65234
                                                                                                           19 espirblaster-a65234
 5 local
type: 0001 rdlength: 4
                      All answers parsed, adding to _answers list..
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
                                                     8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 000c rdlength: 38
                       PTR 38 iCondoESP-A08D0_arduino_tcplocal
 16 iCondoESP-A08D0B
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0010 rdlength: 56
                       TXT 56 
board=genericauth_upload=no   tcp_check=no
 16 iCondoESP-A08D0B
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0021 rdlength: 30
                       SRV 16 69 63 6f 6e 64 6f 65 73 70 2d 61 30 38 64 30 62 
                                                                              icondoesp-a08d0b
                                                                                               16 icondoesp-a08d0b
 5 local
type: 0001 rdlength: 4
                      All answers parsed, adding to _answers list..
Reading answers RX: REQ, ID:0, Q:0, A:4, NS:0, ADD:0
                                                     8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 000c rdlength: 38
                       PTR 38 iCondoESP-9C2A7_arduino_tcplocal
 16 iCondoESP-9C2A71
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0010 rdlength: 56
                       TXT 56 
board=genericauth_upload=no   tcp_check=no
 16 iCondoESP-9C2A71
 8 _arduino
found matching service: arduino
                                4 _tcp
 5 local
type: 0021 rdlength: 30
                       SRV 16 69 63 6f 6e 64 6f 65 73 70 2d 39 63 32 61 37 31 
                                                                              icondoesp-9c2a71
                                                                                               16 icondoesp-9c2a71
 5 local
type: 0001 rdlength: 4
                      All answers parsed, adding to _answers list..

Soft WDT reset

ctx: sys 
sp: 3ffffa50 end: 3fffffb0 offset: 01b0
`

martinayotte, 2016-07-30:
Adding some debug print, I figured where it stalling in a loop until WDT : 
https://github.com/me-no-dev/Arduino/blob/41ebe39b5bde969add15bd5d44f6b3f6b489e3ec/libraries/ESP8266mDNS/ESP8266mDNS.cpp#L674
But I didn't figured out yet why "answer->next" isn't becoming null ...

larsenglund, 2017-06-08:
Good find!

beegee-tokyo, 2017-09-15:
One year later and I still get the same problem.
Is it not possible to fix this.
I use the example code and sometimes it works, specially if there is no device or only one device that offers the service I try to query. But as soon as there are several devices offering the same service it crashes with:
```
ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset
```
Board manager shows: 
`esp8266 by ESP8266 Community version 2.3.0`

martinayotte, 2017-09-15:
The 2.3.0 version has been released on Jun 23 2016, well, a month before this fix has been merged.
To get the fix, you need to install 2.4.0-rc1 which has been released last May 21th.

beegee-tokyo, 2017-09-15:
Perfect! Thanks for the info. Switched to 2.4.0-rc1 and it works now!

closed 2017-09-15 by devyte