esp8266/Arduino #2249: Question: analogWrite() as PWM for ShiftRegister

Gorkde, 2016-07-08:
I'm using a shift register library (http://shiftregister.simsso.de/) to expand my ESP ports.

Since I need to fade some outputs on the shift register I just wrote my own PWM function that's probably not the best.....

Therefore is there a way to make analogWrite not output directly to a pin but call a function of the shift register library?

Thanks!

Gorkde, 2016-07-13:
Anyone?

Btw... i don't know how to mark something as Question maybe someone can tell me how to as well.

igrr, 2016-08-01:
`analogWrite` function is a weak function, so you can override it from your sketch or from a library:
```
extern "C" void analogWrite(int, int) {
  // custom stuff
}
```

closed 2016-08-01 by igrr

Gorkde, 2016-08-01:
@igrr 
I did already write my own but my routine would block the program so I thought to use the analogwrite since it doesn't block as far as I know.

Gorkde, 2016-08-01:
Btw. am  I somehow able to mark a question as such or are just Admins allowed to?

igrr, 2016-08-01:
analogWrite blocks for a bit because it needs to calculate periods, sort them, etc.

Only contributors can label issues afaik.

Gorkde, 2016-08-01:
A bit would be no problem but my routine would block all the time while it's running since it doesn't use timers/interrupt. 

Is there a simple way to make analogwrite call a function instead setting a pin directly?