esp8266/Arduino #2250: The problem with the version 2.3.0

krzbor, 2016-07-08:
My project was compiled correctly in version 2.2.0. After the installation of version 2.3.0 I have error (Windows XP):
In file included from c:\documents and settings\krzys\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\functional:48:0,

                 from C:\Documents and Settings\krzys\Ustawienia lokalne\Dane aplikacji\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiGeneric.h:27,

                 from C:\Documents and Settings\krzys\Ustawienia lokalne\Dane aplikacji\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFiSTA.h:28,

                 from C:\Documents and Settings\krzys\Ustawienia lokalne\Dane aplikacji\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/ESP8266WiFi.h:34,

                 from C:\programy\adruino\projekty\brama\brama.ino:2:

c:\documents and settings\krzys\ustawienia lokalne\dane aplikacji\arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\xtensa-lx106-elf\bits\c++config.h:426:29: fatal error: bits/os_defines.h: No such file or directory

 #include <bits/os_defines.h>

                             ^

compilation terminated.
-----------------------------------------
What happened?

krzychb, 2016-07-08:
Hi @krzbor,
Check this one https://github.com/esp8266/Arduino/issues/1631

krzbor, 2016-07-08:
Thank you!
I copied the "g ++ \ 4.8.2 \ xtensa-lx106-elf \ bits" to "c ++ \ 4.8.2 \ bits"
Now everything works!

closed 2016-08-01 by igrr