esp8266/Arduino #2252: Binaries contain __divdi3 and other soft float functions in .text, but it is already in the ROM

sven337, 2016-07-09:
Hello,

the ESP8266 ROM (the real one, the one we can't write) already contains soft float function such as __divdi3 and friends. From eagle.rom.addr.v6.ld:
PROVIDE ( __adddf3 = 0x4000c538 );
PROVIDE ( __addsf3 = 0x4000c180 );
PROVIDE ( __divdf3 = 0x4000cb94 );
PROVIDE ( __divdi3 = 0x4000ce60 );
PROVIDE ( __divsi3 = 0x4000dc88 );
...

However, when building a sketch, the ELF file includes its own implementation. This is redundant, as it could well call the ROM function. Is this done or purpose or not?

The problem is that these functions take a bit of program space, about 2kB. That doesn't sound like much, but for chips with 512kB if you want to use OTAs, it is important.

I will try to remove this redundancy and am filing this issue to track it and open a discussion about it.

### Basic Infos

#### Hardware
Hardware:			ESP-201
Core Version:      	commit 234a8743c9533d1b12d7962802869fe68e83179f

igrr, 2016-08-01:
I tested this with latest git version and it seems to be working correctly:
```
4000c538 A __adddf3
4000cb94 A __divdf3
4000c8f0 A __muldf3
4000c688 A __subdf3
4000c180 A __addsf3
4000c268 A __subsf3
4000ce60 A __divdi3
4000dc88 A __divsi3
```

ahuillet, 2016-08-02:
I've just updated to commit 8cd331a8bae04a6f1443ff0c93539af4720d8ddf and compiled the Blink example. The problem persists:

$ nm Blink.ino.elf  | grep divdi3
40106bbc T __divdi3

igrr, 2016-08-26:
Okay, the reason why i wasn't able to reproduce this was due to stray libgcc.a file which was stripped from those symbols, in my linker path. Noticed that while debugging an issue with SDK 2.0.0 build failure on Travis. I've added the stripped libgcc.a file, so once the SDK 2.0.0 branch is merged, this should be fixed as well.

closed 2018-01-02 by igrr