esp8266/Arduino #2253: section `.text' will not fit in region `iram1_0_seg'

reaper7, 2016-07-09:
### Basic Infos

#### Hardware
Hardware:			Wemos-D1 Mini (ESP-12)
Core Version:      	2.3 (daily updated from github)

### Description

I have problem with my sketches in latest 2.3 version,
(everything works correctly on versions prior to the date Jun 23, 2016,
last good version for me is from Jun 21 with commit: esptool update to 0.4.9) 

When I try compile attached simple sketch with debug "all" on Serial
then I got error:

> c:/programy/arduino/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\tmp\build5a185a896db7ad49d7b5e61a1d9a467d.tmp/MYENERGY_SDM_v4.ino.elf section `.text' will not fit in region `iram1_0_seg'

> collect2.exe: error: ld returned 1 exit status

but when I switch OFF debug then sketch compiled without problems!
and another curiosity, when I ON debug but comment //#define BRZO
for using normal i2c wire implementation then sketch again compiled ok!
...strange but this is not all :)
for checking I disable all debug and compile in two modes with and without brzo i2c implementation:
version with brzo implementation uses flash: 258 961 and ram: 34 248
vers.without brzo implementation uses flash: 259 065 and ram: 34 392
as You see, version with brzo i2c implementation (instead wire) consumes less ESP8266 resources!
So, why I can't compile it when I enable debug: all ???

I also observed, that on v2.3 drastically decreased the size of free heap size for the same projects!
for compare, attached sketch raport free heap size at the end of setup:
v2.3 ~44984
v2.2 ~47096
More complicated sketches like https://github.com/reaper7/Esp-radio not working because something ate ~2k from free heap (compared to versions before Jun 23)

Something is going in the wrong direction (sdk?), still less and less resources available for user :(

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu
Debug port:		Serial
Debug level:		All

### Sketch

```
#include <SoftwareSerial.h>

#define ONLYASCII
#define BRZO

#if defined ( ONLYASCII )
#include "SSD1306Ascii.h"
  #if defined ( BRZO )
  #include "SSD1306AsciiBrzo.h"
  #else
  #include "SSD1306AsciiWire.h"
  #endif
#else
#include "font.h"
  #if defined ( BRZO )
  #include "SSD1306Brzo.h"
  #else
  #include "SSD1306Wire.h"
  #endif
#endif

#define OLED_SDA    4
#define OLED_SDC    5
#define OLED_ADDR   0x3C

#if defined ( ONLYASCII )
  #if defined ( BRZO )
  SSD1306AsciiBrzo oled;
  #else
  SSD1306AsciiWire oled;
  #endif
#else
  #if defined ( BRZO )
  SSD1306Brzo oled(OLED_ADDR, OLED_SDA, OLED_SDC);
  #else
  SSD1306Wire oled(OLED_ADDR, OLED_SDA, OLED_SDC);
  #endif
#define FSIZE 10      //font size
#define XMARGIN 32    //x offset for small display 64x48 
#define YMARGIN 16    //y offset for small display 64x48 
#endif

SoftwareSerial sdmSer(12, 13, false, 64);   //TX-D7-13, RX-D6-12
//------------------------------------------------------------------------------
#define SDM_BAUD  4800
#define MAX_MILLIS_TO_WAIT 1000 //max time to wait for responce from SDM

#define SDM_B_01 0x01
#define SDM_B_02 0x04

#define SDM_VOLTAGE 0x0000                        //V
#define SDM_CURRENT 0x0006                        //A
#define SDM_POWER 0x000C                          //W
#define SDM_ACTIVE_APPARENT_POWER  0x0012         //VA
#define SDM_REACTIVE_APPARENT_POWER  0x0018       //VAR
#define SDM_POWER_FACTOR  0x001E                  //
#define SDM_PHASE_ANGLE  0x0024                   //DEGREE
#define SDM_FREQUENCY  0x0046                     //Hz
#define SDM_IMPORT_ACTIVE_ENERGY  0x0048          //Wh
#define SDM_EXPORT_ACTIVE_ENERGY  0x004A          //Wh
#define SDM_TOTAL_ACTIVE_ENERGY  0x0156           //Wh
#define SDM_IMPORT_REACTIVE_ENERGY  0x004C        //VARh
#define SDM_EXPORT_REACTIVE_ENERGY  0x004E        //VARh
#define SDM_TOTAL_REACTIVE_ENERGY  0x0158         //VARh

#define SDM_B_05 0x00
#define SDM_B_06 0x02

#define FRAMESIZE 9                               //size of out/in array

struct sdm_values {
  float voltage;
  float current;
  float power;
  float frequency;
} sdmval;

//------------------------------------------------------------------------------
uint16_t calculateCRC(uint8_t *array, uint8_t num) {
  uint16_t temp, flag;
  temp = 0xFFFF;
  for (uint8_t i = 0; i < num; i++) {
    temp = temp ^ array[i];
    for (uint8_t j = 8; j; j--) {
      flag = temp & 0x0001;
      temp >>= 1;
      if (flag)
        temp ^= 0xA001;
    }
  }
  return temp;
}
//------------------------------------------------------------------------------
float getvalue(uint16_t reg) {
  uint16_t temp;
  unsigned long resptime;
  uint8_t sdmarr[FRAMESIZE] = {SDM_B_01, SDM_B_02, 0, 0, SDM_B_05, SDM_B_06, 0, 0, 0};
  float res = -9999.99;

  sdmarr[2] = highByte(reg);
  sdmarr[3] = lowByte(reg);

  temp = calculateCRC(sdmarr, FRAMESIZE - 3);       //calculate out crc only from first 6 bytes

  sdmarr[6] = lowByte(temp);
  sdmarr[7] = highByte(temp);
  
  sdmSer.write(sdmarr, FRAMESIZE - 1);              //send 8 bytes

  resptime = millis();

  while ( (sdmSer.available() < FRAMESIZE) && ((millis() - resptime) < MAX_MILLIS_TO_WAIT) ) {      
    delay(1);
  }

  if(sdmSer.available() == FRAMESIZE) {
    for(int n=0; n<FRAMESIZE; n++) {
      sdmarr[n] = sdmSer.read();
    }

    if (sdmarr[0] == SDM_B_01 && sdmarr[1] == SDM_B_02 && sdmarr[2] == SDM_B_02) {
      if ((calculateCRC(sdmarr, FRAMESIZE - 2)) == ((sdmarr[8] << 8) | sdmarr[7])) {     //calculate crc from first 7 bytes and compare with received crc (bytes 7 & 8)
        ((uint8_t*)&res)[3]= sdmarr[3];
        ((uint8_t*)&res)[2]= sdmarr[4];
        ((uint8_t*)&res)[1]= sdmarr[5];
        ((uint8_t*)&res)[0]= sdmarr[6];      
      }
    }
  }
  sdmSer.flush();
  return (res);  
}
//------------------------------------------------------------------------------
void setup() {
  Serial.begin(115200);
  sdmSer.begin(SDM_BAUD);

#if defined ( ONLYASCII )
  #if defined ( BRZO )
  oled.begin(&MicroOLED64x48, OLED_ADDR, OLED_SDA, OLED_SDC);
  #else
  Wire.begin(OLED_SDA, OLED_SDC);
  oled.begin(&MicroOLED64x48, OLED_ADDR);
  #endif
  oled.setFont(System5x7);
  oled.clear();
  oled.print("MODBUS");
#else
  oled.init();                                        
  oled.setFont(Cousine_Bold_10);        
  oled.setTextAlignment(TEXT_ALIGN_LEFT);          
  oled.flipScreenVertically();                      
  oled.displayOn();                                   
  oled.clear();
  oled.drawString(XMARGIN + 0, YMARGIN + 0, "MODBUS");
  oled.display();
#endif  

  Serial.println(ESP.getFreeHeap());

  delay(1000);
}
//------------------------------------------------------------------------------
void loop() {
  sdmval.voltage = getvalue(SDM_VOLTAGE);
  sdmval.current = getvalue(SDM_CURRENT);
  sdmval.power = getvalue(SDM_POWER);
  sdmval.frequency = getvalue(SDM_FREQUENCY);

  char buff[48] = {};
  sprintf(buff, "V:%7.2f\nA:%7.2f\nP:%7.2f\nF:%7.2f\0", sdmval.voltage, sdmval.current, sdmval.power, sdmval.frequency); 
  oled.clear();

#if defined ( ONLYASCII )
  oled.print(buff);
  oled.setCursor(0,4);
  oled.clearToEOL();
#else
  oled.drawString(XMARGIN + 0, YMARGIN + (0 * FSIZE), buff);
  oled.display();
#endif

  //Serial.println(ESP.getFreeHeap());
  
  delay(1000);
}
```

igrr, 2016-07-09:
Increase in memory usage the day following the 2.3.0 release has to do with the fact that we now use full newlib library (https://github.com/esp8266/Arduino/pull/1752), and there was also an update to the SDK 1.5.4. I can agree that things are going in the direction of more and more memory use. There are a few things which may be cut down in newlib build, I'll put this on the list of things to do.

igrr, 2016-07-09:
Where should i get SSD1306AsciiBrzo.h? Google doesn't come up with any results.

luc-github, 2016-07-09:
https://github.com/reaper7/SSD1306Ascii/blob/master/src/SSD1306AsciiBrzo.h

igrr, 2016-07-10:
Ah sorry, that should have been obvious😳

reaper7, 2016-07-10:
@igrr - sorry, I forgot to add :(

devyte, 2017-10-15:
@reaper7 how does latest git behave compared to your original comment?

devyte, 2018-02-28:
Closing due to lack of feedback, and age.

closed 2018-02-28 by devyte