esp8266/Arduino #2255: Exception 29 while sending values from HTTP GET to esp12f

anujmattoo, 2016-07-10:
[image]
### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

I am working on HTTP GET Request. I have created ESP wifi as web server. It creates a access point with IP address : 192.168.4.1 When my android app is connected to web server, it sends values every 50ms on 192.168.4.1/R"VALUE". Value is integer type (0~1023). Example :- 192.168.4.1/R546

However, it is working fine when I send values every 1s, but esp crashes, when IP is hit every 5ms second. I have captured the crash detail on serial monitor. Attaching the file and code as well. Kindly help. 

### Settings in IDE

Module:  			?NodeMCU?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?80Mhz?
Upload Using:		?SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <ESP8266WiFi.h>
#include <EEPROM.h>
#include <ESP8266WebServer.h>

#define RED_LED     12

WiFiServer server(80); //Initialize the server on Port 80

char myTestUrl[100];

void setup()
{

  EEPROM.begin(512);
  Serial.begin(9600); //Start communication between the ESP8266-12E and the monitor window

  WiFi.mode(WIFI_AP); //Our ESP8266-12E is an AccessPoint
  WiFi.softAP("LED_Downlight1", "12345678"); // Provide the (SSID, password); .
  server.begin(); // Start the HTTP Server

  IPAddress HTTPS_ServerIP = WiFi.softAPIP(); // Obtain the IP of the Server
  Serial.print("Server IP is: "); // Print the IP to the monitor window
  Serial.println(HTTPS_ServerIP);
}

void loop()
{
  WiFiClient client = server.available();
  if (!client)
  {
    return;
  }

  String request = client.readStringUntil('\r');
  //Serial.print(request);

  request.toCharArray(myTestUrl, 100);

  char *Red = strstr(myTestUrl, "/R");
  if (Red)
  {
    int RedValue  = atoi(Red + 2);
    analogWrite(RED_LED, RedValue);
    Serial.print("Red: ");
    Serial.print(" ");
    Serial.print(RedValue);
    Serial.print("\n");

    int SetRedValue = (RedValue / 4);
    EEPROM.write(0, SetRedValue);

    EEPROM.commit();

    Serial.print("Red Memory Write :");
    Serial.print(" ");
    Serial.print(SetRedValue);
    Serial.print("\n \n");

  }
}
```

### Debug Messages

```
messages here
```

me-no-dev, 2016-07-10:
5ms is quite too often for the ESP to handle the requests. Each request you make actually takes 9 or 10 network packets to complete (connect, send request, get response, close) and those usually take more than 5ms.
Also please look [here](https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md) on how to decode the stack dump to something that we can read and maybe get a clue of the problem.

anujmattoo, 2016-07-10:
It might not be 5ms but I am POST-ing HTTP  request  50 times in one second to ESP. Here is the Exception decoder result.

[image]
[image]

me-no-dev, 2016-07-10:
I looked at your sketch again and in the code you are not returning anything to the browser back, so it will keep the connections open. You need to respond back to the browser, or you are breaking HTTP

anujmattoo, 2016-07-10:
Okay, what should be response ? Any example or sketch? Breaking HTTP, How?

me-no-dev, 2016-07-10:
```
HTTP/1.1 200 OK\r\nConnection: close\r\nContent-Length: 0\r\n\r\n
```
This is a minimal good response to the browser which will close the connection and clean the resources from the ESP

anujmattoo, 2016-07-10:
I updated the code with the response, but still shows the exception after POST-ing data to ESP (30 Values/per second).

`String request = client.readStringUntil('\r');
  //Serial.print(request);

  request.toCharArray(myTestUrl, 100);

  char *Red = strstr(myTestUrl, "/R");
  if (Red)
  {
    int RedValue  = atoi(Red + 2);
    analogWrite(RED_LED, RedValue);
    Serial.print("Red: ");
    Serial.print(" ");
    Serial.print(RedValue);
    Serial.print("\n");

    int SetRedValue = (RedValue / 4);
    EEPROM.write(0, SetRedValue);

    EEPROM.commit();

    String s = "HTTP/1.1 200 OK\r\nConnection: close\r\nContent-Length: 0\r\n\r\n";
    client.print(s);

  }
}`

me-no-dev, 2016-07-10:
I'll try your sketch and I'll see what I can do. One thing to note is EEPROM.commit() You will wear out the flash at the EEPROM sector really quick if you write to it with every request. the flash has around 100000 write/erase cycles

me-no-dev, 2016-07-11:
Serial baud at 9600 causes requests to take between 30 and 50 ms, which makes it hard to do 30 per second. I upped the baud to 115200 and removed EEPROM code and I get around 20-30ms per request.
That is through the STA interface, AP will be slower as it's software implemented.
Do you wait for the request to complete or are you just shooting packets towards the esp? What code do you use to send those requests?

anujmattoo, 2016-07-11:
Hey thanks @me-no-dev  for considering my problem and working on my code. As per you last 2 comment, I would like to ask few questions. **One, What is the best way/ efficient way to Write EEPROM in my code?** Any Suggestion or improvement would  be of great help. Second, Right now what I have done is, I have created ESP as web server, it works on HTTP GET-POST. The ESP creates a WiFi network (LED_Downlight1) refer, ` WiFi.softAP("LED_Downlight1", "12345678"); ` SmartPhone/Laptop gets connected to that network. When I hit `http://192.168.4.1/R100` for an instance, in browser, the LED brightness adjust according to value sent after "/R". However when sending values from an Android app, there is a slider, which send the values from 0~1023. After receiving few 100 request, depend upon slider value change interval, it throws the exception. **I am just shooting values to ESP, not waiting for any kind of response.**  I have attached the android app with this comment, right now, only first slider is working (RED one).
[WebServer.apk.zip](https://github.com/esp8266/Arduino/files/357821/WebServer.apk.zip)

me-no-dev, 2016-07-11:
you should wait for the request to complete and you should send response to the client so it closes the connection. this is how HTTP works and I can't stress enough how important that is! If you do not do things correctly you will load the network stack on the esp and will get exceptions all over the place!
As for EEPROM, maybe save every minute if value has changed, those sliders will kill your flash fast

anujmattoo, 2016-07-11:
If I wait for the response, the purpose would be defeated, it will take time for every response. I need values to be changed instantly. 
So what is way around? How can I speed up my Sending values without getting exceptions? Use websockets over HTTP?
Please suggest, I am in real trouble now. What to do?

me-no-dev, 2016-07-11:
use plain tcp or udp server. Android and ESP can do both. WebSocket is also an option

anujmattoo, 2016-07-11:
Okay, can you please send me any link of Examples of ESP8266 based on TCP/UDP Server. 
What are the advantages to use TCP/UDP Server over HTTP? Will they serve my purpose, anyhow?

me-no-dev, 2016-07-11:
```cpp
WiFiServer server(80); //Initialize the server on Port 80
```
This IS a TCP server already, but you are using HTTP client so you have to abide to the HTTP requirements. If you change the port and use different client, it will be just a tcp server, you will not need to respond and each message will take only two network packets.
If this info is not enough for you, I advise you to read on TCP/UDP servers in google so you have better understanding what such a server is. Examples are also available tight here in this repository (TelnetServer and others)

anujmattoo, 2016-07-11:
Thanks @me-no-dev  for helping me and sharing wonderful information. Cheers!

closed 2016-08-01 by igrr