esp8266/Arduino #2257: ESP8285 Development Board

kriswiner, 2016-07-10:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```
Just want to confirm that I exposed pins 9 and 10 on my ESP8285 Development Board (https://www.tindie.com/products/onehorse/esp8285-development-board/) and was able to use either one to blink and led or serve as an interrupt  from a sensor, etc. I will do more testing but it seems these two pins are indeed available GPIOs. Not sure if they come with a special function; the ESP82XX seems to have an internal MUX to allow any pin to express the available peripherals.

How can I arrange to have my ESP8285 Development Board included in the list of available boards?

igrr, 2016-07-10:
Which reset method do you use, ck or nodemcu? We have an ESP8285 entry in the list already, does it work with your board?

kriswiner, 2016-07-10:
I use ck; if you have an ESP8285 Dev Board already listed you don't need
mine I suppose.

<https://github.com/esp8266/Arduino/issues/2257#issuecomment-231616571> , or
mute the thread
W0ks5qUXvpgaJpZM4JI78Y> .
<https://github.com/notifications/beacon/AGY1qkpcO2NjV9p32Uat1H-ZMf3ziI-7ks5
qUXvpgaJpZM4JI78Y.gif>

mcmega, 2017-09-18:
Hi!
I have modelu ESP8285 WEMOS? D1 mini, but need Reset Method: nodemcu. How do it this? In Arduino only ck method.

kriswiner, 2017-09-19:
No idea... maybe ask the guy you bought it from?

mcmega, 2017-09-19:
I figured out how to fix the problem!
You must change the settings in the file boards.txt. The file is on the path (for Windows)
C: \ Users \ *** \ AppData \ Local \ Arduino15 \ packages \ esp8266 \ hardware \ esp8266 \ 2.3.0
It is necessary to add a reset method in the corresponding section. 

I'm in the Generic ESP8285 module add:
```
esp8285.menu.ResetMethod.ck = ck
esp8285.menu.ResetMethod.ck.upload.resetmethod = ck
esp8285.menu.ResetMethod.nodemcu = nodemcu
esp8285.menu.ResetMethod.nodemcu.upload.resetmethod = Nodem
```

devyte, 2017-10-15:
@kriswiner said:
or serve as an interrupt from a sensor

Did you actually test that interrupts work on these pins? Ref: #3549

kriswiner, 2017-10-15:
Which pins? I have used interrupts on most of the pins on the ESP8266/85,
yes.

devyte, 2017-10-15:
Your issue is about pins 9 and 10.

kriswiner, 2017-10-15:
Yes, I have used interrupts on pins 9 and 10.

devyte, 2017-10-15:
Those pins are usually not ready for interrupts. How did you init them? I ask because of #3549 .

kriswiner, 2017-10-15:
Nothing special, they work just fine. Not sure what your problems is. What
ESP8285 board are you using?

devyte, 2018-01-08:
@kriswiner not my problem (I don't use the 8285 board), but what was reported in #3549 . It was reported there that pins 9 and 10 didn't work with interrupts, and looking at the pin init code in our core, they are in fact not even initialized as input on startup.1
Hence, I wondered how you got them to work.

devyte, 2018-01-10:
@kriswiner your original request seems to be to just make sure a board is added in boards.txt, except that this is a variant that has pins 9 & 10 available in addition to the others. Is this correct?

kriswiner, 2018-01-10:
Yes, and the board was added months ago so safe to close this issue.

devyte, 2018-01-10:
Awesome. Closing.

closed 2018-01-10 by devyte