esp8266/Arduino #2262: SPI reading data from sensor problem

mottab, 2016-07-11:
### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.1.0-rc2

### Description

I am working with [MS5540C](http://www.fut-electronics.com/wp-content/uploads/2015/08/MS5540C_waterproof_pressure_sensor_datasheet.pdf) sensor, trying to communicate with it using SPI, but getting wrong data and the same result regardless the environment changes.

I set the SPI Frequency to 5khz  ```cpp SPI.setFrequency(500000); ```
I used to make this in arduino using this line of code ```cpp SPI.setClockDivider(SPI_CLOCK_DIV32); ```

I also uses a pwm frequency (mclk) using:
 ```cpp
analogWriteFreq(10);
analogWriteRange(256);
analogWrite(D8, 128);
 ```
I used to write that arduino uno using:
 ```cpp
TCCR1B = (TCCR1B & 0xF8) | 1 ;
analogWrite(9, 128);
 ```

Why am I getting wrong reads ??

What Should I have to change to get write reads from the sensor.

### Settings in IDE

Module:  			NodeMCU 1.0 (ESP-12E Module)
Flash Size: 		4MB
CPU Frequency:		80Mhz
Upload Using:		SERIAL
### Sketch

```cpp
#include <SPI.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h> 
#include <ESP8266WebServer.h>
#include <WiFiClientSecure.h>
#include <ArduinoJson.h>
#include "FS.h"

void setup() {
  Serial.begin(115200);
  Serial.println("");
  delay(2000);
  setupSensor();
  delay(500);
  
}

void loop() {
  readSensorData();
  delay(1000);
}

void resetsensor() //this function keeps the sketch a little shorter
{
  SPI.setDataMode(SPI_MODE0); 
  SPI.transfer(0x15);
  SPI.transfer(0x55);
  SPI.transfer(0x40);
}

void setupSensor() {
  SPI.begin(); //see SPI library details on arduino.cc for details
  SPI.setBitOrder(MSBFIRST);
//  SPI.setClockDivider(SPI_CLOCK_DIV32); //divide 16 MHz to communicate on 500 kHz (16000000/32)
  // 80,000,000 / 
  SPI.setFrequency(500000);
//  SPI.beginTransaction(SPISettings(500000, MSBFIRST, SPI_MODE0));
//  SPI.setHwCs(false);
  pinMode(clock, OUTPUT);
  delay(100);
}

int readSensorData() {
  #ifdef __AVR__
  TCCR1B = (TCCR1B & 0xF8) | 1 ; //generates the MCKL signal
  Serial.println("HAHAHA d7kt 3leek");
  #else
  analogWriteFreq(10);
  analogWriteRange(256);
  #endif
  analogWrite(D8, 128);

  resetsensor();//resets the sensor - caution: afterwards mode = SPI_MODE0!

  //Calibration word 1
  unsigned int word1 = 0;
  unsigned int word11 = 0;
  SPI.transfer(0x1D); //send first byte of command to get calibration word 1
  SPI.transfer(0x50); //send second byte of command to get calibration word 1
  SPI.setDataMode(SPI_MODE1); //change mode in order to listen
  word1 = SPI.transfer(0x00); //send dummy byte to read first byte of word
  word1 = word1 << 8; //shift returned byte 
  word11 = SPI.transfer(0x00); //send dummy byte to read second byte of word
  word1 = word1 | word11; //combine first and second byte of word

  resetsensor();//resets the sensor

  //Calibration word 2; see comments on calibration word 1
  unsigned int word2 = 0;
  byte word22 = 0; 
  SPI.transfer(0x1D);
  SPI.transfer(0x60);
  SPI.setDataMode(SPI_MODE1); 
  word2 = SPI.transfer(0x00);
  word2 = word2 <<8;
  word22 = SPI.transfer(0x00);
  word2 = word2 | word22;

  resetsensor();//resets the sensor

  //Calibration word 3; see comments on calibration word 1
  unsigned int word3 = 0;
  byte word33 = 0;
  SPI.transfer(0x1D);
  SPI.transfer(0x90); 
  SPI.setDataMode(SPI_MODE1); 
  word3 = SPI.transfer(0x00);
  word3 = word3 <<8;
  word33 = SPI.transfer(0x00);
  word3 = word3 | word33;

  resetsensor();//resets the sensor

  //Calibration word 4; see comments on calibration word 1
  unsigned int word4 = 0;
  byte word44 = 0;
  SPI.transfer(0x1D);
  SPI.transfer(0xA0);
  SPI.setDataMode(SPI_MODE1); 
  word4 = SPI.transfer(0x00);
  word4 = word4 <<8;
  word44 = SPI.transfer(0x00);
  word4 = word4 | word44;
  
  long c1 = word1 >> 1;
  long c2 = ((word3 & 0x3F) << 6) | ((word4 & 0x3F));
  long c3 = (word4 >> 6) ;
  long c4 = (word3 >> 6);
  long c5 = (word2 >> 6) | ((word1 & 0x1) << 10);
  long c6 = word2 & 0x3F;
  
  resetsensor();//resets the sensor

 //Temperature:
  unsigned int tempMSB = 0; //first byte of value
  unsigned int tempLSB = 0; //last byte of value
  unsigned int D2 = 0;
  SPI.transfer(0x0F); //send first byte of command to get temperature value
  SPI.transfer(0x20); //send second byte of command to get temperature value
  delay(35); //wait for conversion end
  SPI.setDataMode(SPI_MODE1); //change mode in order to listen
  tempMSB = SPI.transfer(0x00); //send dummy byte to read first byte of value
  tempMSB = tempMSB << 8; //shift first byte
  tempLSB = SPI.transfer(0x00); //send dummy byte to read second byte of value
  D2 = tempMSB | tempLSB; //combine first and second byte of value

  resetsensor();//resets the sensor

  //Pressure:
  unsigned int presMSB = 0; //first byte of value
  unsigned int presLSB =0; //last byte of value
  unsigned int D1 = 0;
  SPI.transfer(0x0F); //send first byte of command to get pressure value
  SPI.transfer(0x40); //send second byte of command to get pressure value
  delay(35); //wait for conversion end
  SPI.setDataMode(SPI_MODE1); //change mode in order to listen
  presMSB = SPI.transfer(0x00); //send dummy byte to read first byte of value
  presMSB = presMSB << 8; //shift first byte
  presLSB = SPI.transfer(0x00); //send dummy byte to read second byte of value
  D1 = presMSB | presLSB; 
  
  const long UT1 = (c5 * 8) + 20224;
  const long dT =(D2 - UT1);
  const long TEMP = 200 + ((dT * (c6 + 50))/1024);
  const long OFF  = (c2*4) + (((c4 - 512) * dT)/4096);
  const long SENS = c1 + ((c3 * dT)/1024) + 24576;
  
  long PCOMP = ((((SENS * (D1 - 7168))/16384)- OFF)/32)+250;
//  float TEMPREAL = TEMP/10;
  Serial.print("pressure =    ");
  Serial.print(PCOMP);
  Serial.println(" mbar"); 

  const long dT2 = dT - ((dT >> 7 * dT >> 7) >> 3);
  const float TEMPCOMP = (200 + (dT2*(c6+100) >>11))/10;
  Serial.print("temperature = ");
  Serial.print(TEMPCOMP);  
  Serial.println(" °C");
  Serial.println("************************************");
  analogWriteFreq(1000);
  analogWriteRange(PWMRANGE);
  return PCOMP; // in mbar
}
```

vicnevicne, 2016-07-12:
This is a question and (until proven wrong) not an issue with the Arduino cores, so probably it doesn't belong here...
However, I find the following line in your code a bit strange:
`SPI.setDataMode(SPI_MODE1); //change mode in order to listen`
Most of the SPI devices I know only work in a single mode (mode meaning clock idle low or high, and bits changing on rising or falling edge). See https://en.wikipedia.org/wiki/Serial_Peripheral_Interface_Bus#Clock_polarity_and_phase
Anyway, I think this is not the place to discuss further. Please search the forums on esp8266.com and post your question there if you are still in doubt :-)

mottab, 2016-07-13:
@vicnevicne Thanks, it seems like ESP8266 is unstable when going over 4khz and this is the frequency that the sensor is using in MCLK

closed 2016-07-14 by mottab

tobiges, 2021-04-06:
My ms5540c works with a 8266 and arduino uno. I have an platformio example here: [MS5540C-sensor-example](https://github.com/tobiges/MS5540C-sensor-example)

halilhassan, 2021-04-16:
> My ms5540c works with a 8266 and arduino uno. I have an platformio example here: [MS5540C-sensor-example](https://github.com/tobiges/MS5540C-sensor-example)

Do you have a wiring diagram for the 8266?

tobiges, 2021-04-16:
I updated the readme to include some wiring examples.