esp8266/Arduino #2263: Compilation error with Ardion HEAD, lack of newlibc?

middelink, 2016-07-11:
### Basic Infos

#### Hardware
Hardware:		ESP-12E
Core Version:      	2.3.0

### Description
Compilation failure after removal of various function due to use of newlibc. Sadly, newlibc is not included so when fetching HEAD, compilation fails miserable under platformio. And likely other projects too

Problem description
I would like to be able to compile projects with esp8266/Arduino :)

### Settings in IDE
platformio actually.

Module:  			NodeMCU ESP12E
Flash Size: 		4MiB
CPU Frequency:	80Mhz
Flash Mode:		qio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

*Anything which includes some serious use of ESP8266Webserver, mDNS, PubSubClient and/or NTP and likely many other libraries.*

### Debug Messages
*None, as it does not run*

### Messages
```bash
xtensa-lx106-elf-g++ -o .pioenvs/espressif_esp12_board/firmware.elf -Os -nostdlib -Wl,--no-check-sections -u call_user_start -Wl,-static -Wl,--gc-sections -Wl,-wrap,system_restart_local -Wl,-wrap,register_chipv6_phy -Wl,-T"esp8266.flash.4m.ld" .pioenvs/espressif_esp12_board/src/tmp_ino_to.o -L/home/middelin/.platformio/packages/framework-arduinoespressif/tools/sdk/lib -L/home/middelin/.platformio/packages/ldscripts -L.pioenvs/espressif_esp12_board -Wl,--start-group .pioenvs/espressif_esp12_board/libFrameworkArduinoVariant.a .pioenvs/espressif_esp12_board/libFrameworkArduino.a -lmesh -lwpa2 -lsmartconfig -lpp -lmain -lwpa -llwip -lnet80211 -lwps -lcrypto -lphy -lhal -laxtls -lgcc -lm -lstdc++ .pioenvs/espressif_esp12_board/libESP8266WiFi.a .pioenvs/espressif_esp12_board/libESP8266WebServer.a .pioenvs/espressif_esp12_board/libESP8266mDNS.a .pioenvs/espressif_esp12_board/libSPI.a .pioenvs/espressif_esp12_board/libPubSubClient_ID89.a .pioenvs/espressif_esp12_board/libMD_MAX72XX_ID430.a -Wl,--end-group
.pioenvs/espressif_esp12_board/libESP8266mDNS.a(ESP8266mDNS.o):(.text._ZN13MDNSResponder6_replyEhPcS0_t+0xc): undefined reference to `strcat'
.pioenvs/espressif_esp12_board/libESP8266mDNS.a(ESP8266mDNS.o):(.text._ZN13MDNSResponder6_replyEhPcS0_t+0x94): undefined reference to `strcat'
.pioenvs/espressif_esp12_board/libESP8266mDNS.a(ESP8266mDNS.o):(.text._ZN13MDNSResponder6_replyEhPcS0_t+0xb7): undefined reference to `strcat'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(core_esp8266_main.o):(.text._Z9init_donev+0xc): undefined reference to `printf'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(core_esp8266_main.o): In function `init_done()':
core_esp8266_main.cpp:(.text._Z9init_donev+0x3d): undefined reference to `printf'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(umm_malloc.o): In function `umm_info':
umm_malloc.c:(.irom0.text+0xb6): undefined reference to `printf'
umm_malloc.c:(.irom0.text+0x12f): undefined reference to `printf'
umm_malloc.c:(.irom0.text+0x16a): undefined reference to `printf'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(umm_malloc.o):umm_malloc.c:(.irom0.text+0x1f2): more undefined references to `printf' follow
.pioenvs/espressif_esp12_board/libESP8266WebServer.a(Parsing.o):(.text._ZN16ESP8266WebServer9urlDecodeERK6String+0x4): undefined reference to `strtol'
.pioenvs/espressif_esp12_board/libESP8266WebServer.a(Parsing.o): In function `ESP8266WebServer::urlDecode(String const&)':
Parsing.cpp:(.text._ZN16ESP8266WebServer9urlDecodeERK6String+0x76): undefined reference to `strtol'
.pioenvs/espressif_esp12_board/libESP8266WebServer.a(Parsing.o):(.text._ZN16ESP8266WebServer13_parseRequestER10WiFiClient+0x28): undefined reference to `atoi'
.pioenvs/espressif_esp12_board/libESP8266WebServer.a(Parsing.o):(.text._ZN16ESP8266WebServer13_parseRequestER10WiFiClient+0xf9): undefined reference to `atoi'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZN6StringC2Ejh+0x0): undefined reference to `utoa'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::String(unsigned int, unsigned char)':
WString.cpp:(.text._ZN6StringC2Ejh+0x1f): undefined reference to `utoa'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZN6String6concatEi+0x0): undefined reference to `itoa'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::concat(int)':
WString.cpp:(.text._ZN6String6concatEi+0x16): undefined reference to `itoa'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZNK6String7indexOfEcj+0x0): undefined reference to `strchr'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::indexOf(char, unsigned int) const':
WString.cpp:(.text._ZNK6String7indexOfEcj+0x17): undefined reference to `strchr'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZN6String11toLowerCaseEv+0x0): undefined reference to `tolower'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::toLowerCase()':
WString.cpp:(.text._ZN6String11toLowerCaseEv+0x15): undefined reference to `tolower'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZN6String4trimEv+0x0): undefined reference to `isspace'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::trim()':
WString.cpp:(.text._ZN6String4trimEv+0x2b): undefined reference to `isspace'
WString.cpp:(.text._ZN6String4trimEv+0x42): undefined reference to `isspace'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o):(.text._ZNK6String5toIntEv+0x0): undefined reference to `atol'
.pioenvs/espressif_esp12_board/libFrameworkArduino.a(WString.o): In function `String::toInt() const':
WString.cpp:(.text._ZNK6String5toIntEv+0x12): undefined reference to `atol'
collect2: error: ld returned 1 exit status
scons: *** [.pioenvs/espressif_esp12_board/firmware.elf] Error 1
```

igrr, 2016-07-11:
Probably platformio issue? If you try compiling in Arduino IDE, it will
compile fine.

middelink, 2016-07-11:
Seems that way, since it is not using platform.txt, it did not pick up the newly added path to tools/sdk/libc/... That and the one liner in the linker script.

Adding the paths to the build_flags and a simple manual edit of esp8266.flash.common.ld made it function.

```ini
[env:espressif_esp12_board]
platform = espressif
framework = arduino
board = esp12
board_flash_mode = dio
upload_resetmethod = nodemcu
build_flags = -I/home/middelin/.platformio/packages/framework-arduinoespressif/tools/sdk/libc/xtensa-lx106-elf/include -L/home/middelin/.platformio/packages/framework-arduinoespressif/tools/sdk/libc/xtensa-lx106-elf/lib -lc
upload_port = /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0
```

closed 2016-07-11 by middelink

n8henrie, 2016-09-12:
@middelink -- can you be more specific regarding the edit you had to make to `esp8266.flash.common.ld`? I'd like to get the git version working in Platformio as well.

middelink, 2016-09-14:
Greetings,

My current (working) platformio.ini has the following build_flags:
build_flags = -I$PIOHOME_DIR/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/include -L$PIOHOME_DIR/packages/framework-arduinoespressif8266/tools/sdk/libc/xtensa-lx106-elf/lib -lc

So if you run it like this, it will run out of .text space, as it will now include Arduino's libc implementation. So we need to move that to the .irom0.text segment.

$ diff -u $PIOHOME_DIR//platforms/espressif8266/ldscripts/esp8266.flash.common.ld.old $PIOHOME_DIR//platforms/espressif8266/ldscripts/esp8266.flash.common.ld
--- $PIOHOME_DIR/platforms/espressif8266/ldscripts/esp8266.flash.common.ld.old	2016-09-14 21:49:11.253437461 +0200
+++ $PIOHOME_DIR/platforms/espressif8266/ldscripts/esp8266.flash.common.ld	2016-09-14 21:43:57.568704836 +0200
@@ -168,6 +168,7 @@
     EXCLUDE_FILE (umm_malloc.o) .text*)
     *.pioenvs/*/src/*.o(EXCLUDE_FILE (umm_malloc.o) .literal*, \
     EXCLUDE_FILE (umm_malloc.o) .text*)
+    *libc.a:(.literal .text .literal.* .text.*) 
     *libm.a:(.literal .text .literal.* .text.*)
     *libgcc.a:_umoddi3.o(.literal .text)
     *libgcc.a:_udivdi3.o(.literal .text)

Done.

n8henrie, 2016-09-15:
Got it -- the platformio team [just added a more "official" way](http://docs.platformio.org/en/latest/platforms/espressif8266.html#using-arduino-framework-with-staging-version) to install the dev branch that works great, very user friendly.

Issue there / my steps: https://github.com/platformio/platform-espressif8266/issues/4#issuecomment-247194767

middelink, 2016-09-15:
Yup, just followed their steps and it seems to work. Less happy with the fact that the staging framework is installed OVER the normal one, so my "production" builds now also use the staging framework, even though they have `platform = espressif8266`, that for me is a showstopper :(