esp8266/Arduino #2265: ESP12e WiFi not working using Arduino IDE, but working well in lua

croisez, 2016-07-11:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	2.3.0

### Description

ESP12e WiFi not working using Arduino IDE, but working well in lua.
If I use the Nodemcu Firmware Programmer to put nodemcu inside my esp12e, I am able to configure WiFi in STA or in SOFTAP mode. I am able to connect to an AP, and when SOFTAP mode is started, I can see the new AP name using my smartphone.
But if I do the same in cpp, using Arduino IDE, in STA mode, the RSSI is always -31.
And in SOFTAP mode, the new AP name is not visible on my smartphone.
Someone has posted a similar issue at esp8266.com forum: http://www.esp8266.com/viewtopic.php?f=28&t=5890
The conclusion was that the programmer had chosen the wrong esp board while programming.
I don't think that I choose the wrong board. Isn't it?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		4MB (3M)
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch
Basic example from ArduinoIDE->ESP8266Wifi->NTPClient

entropia1ac, 2016-07-13:
Hi,
I've got the same issue .(ESP01/ESP12/ESP12E)

MPX4132, 2016-07-20:
Same issue on ESP03...

krzychb, 2016-07-20:
@entropia1ac,
@MPX4132,

As you are using core 2.3.0, maybe you face the issue https://github.com/esp8266/Arduino/issues/2186#issuecomment-228581052.

If you are new to esp8266/Arduino, then I would install core 2.2.0 instead of 2.3.0 and check if you are able to make ruining the following example sketches:
1. https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide#blink-test
2. https://learn.adafruit.com/adafruit-huzzah-esp8266-breakout/using-arduino-ide#connecting-via-wifi

Krzysztof

MPX4132, 2016-07-20:
Yeah, I'm in fact using 2.3.0. I tried @SupotSaeEa's workaround and it seemed to fix the issue, was about to post the comment, @krzychb beat me to the punch.

Workaround using version 2.3.0 and ESP03:
```
WiFi.setOutputPower(0);
WiFi.begin(ssid, password);
```

nneil, 2016-07-25:
The workaround doesn't work for me. I am using an ESP-201 and cannot connect to any AP using WPA2-Personal and a simple SSID and password. I chose Generic ESP8266 as the board type. NodeMCU works fine.

MPX4132, 2016-07-27:
I'm guessing the workaround fixes the bug, or set of bugs, with the newer firmware on the older models. The ESP-201 seems to be fairly new. I'm guessing that's why it's not working out for you. If I were you, I'd try to find a workaround for the ESP-12, since the one you've got seems to be closer to that model.

nneil, 2016-07-27:
I subsequently tried a WeMos D1 Mini (rev 1), and a Doit.am ESP12E DEVKIT V2 (NodeMCU clone).  I don't know where to begin to "find a workaround", other than to report the issue here.

MPX4132, 2016-07-27:
Yeah, same, I just got lucky finding a solution for mine. Just googled "site:github esp03 won't connect" or something like that, I forgot.

devyte, 2017-10-14:
@croisez is this issue still valid with latest git? does it happen with #3215 ?

parth-garg, 2017-11-09:
Select NodeMCU 1.0(ESP-12E Module) as Board with the default settings,works like a charm

arnold-b, 2017-11-25:
Hello, 

i have the same problem. On me the NodeMCU (ESP-12E) don´t work. I cant get any Wifi connection

parth-garg, 2017-11-25:
Did you flash it first using node mcu flasher,
Order a custom build containing the wifi module and flash it on the esp 12e.
Try this and then start on lua

arnold-b, 2017-11-25:
Thanks, i will try. 

 

Till now i just installed the current lua firmware not the customized one

parth-garg, 2017-11-25:
> auth/AfTd2XIfRKHj1oepMe6R_f5-pjPPoPUsks5s5-ALgaJpZM4JJuxN> . <

The default firmware doesn't have the wifi module

arnold-b, 2017-11-25:
Thanks, 

 

i still wait for the customized version

arnold-b, 2017-11-26:
Hello, 

 

i got the firmware now. 

 

You said under Arduino it´don’t work ?

 

I never used Lua before.

 

 

 

Your NodeMCU custom build finished successfully. You may now download the firmware:

- float:  <http://nodemcu-build.com/builds/nodemcu-master-15-modules-2017-11-25-09-23-38-float.bin> http://nodemcu-build.com/builds/nodemcu-master-15-modules-2017-11-25-09-23-38-float.bin

- integer: http://nodemcu-build.com/builds/nodemcu-master-15-modules-2017-11-25-09-23-38-integer.bin

 

This was built against the master branch and includes the following modules: crypto, encoder, enduser_setup, file, gpio, http, i2c, net, node, ow, spi, tmr, uart, wifi, tls.

The files are guaranteed to be available for download for 24h.

arnold-b, 2017-11-26:
I have this 2 files but i don´t know what to do

arnold-b, 2017-11-26:
In Explorer i get now every second the debug. 

 

I didn´t setup any program but he always try to connect to my wifi.

 

                wifi_event_monitor_handle_event_cb is called

reconnect

scandone

no Braun found, reconnect after 1s

 

                wifi_event_monitor_handle_event_cb is called

reconnect

scandone

no Braun found, reconnect after 1s

 

                wifi_event_monitor_handle_event_cb is called

reconnect

scandone

no Braun found, reconnect after 1s

 

                wifi_event_monitor_handle_event_cb is called

reconnect

arnold-b, 2017-11-26:
Hello, 

 

i know i´m bothering you. 

 

I think my module is broken. 

 

In debug I get now the reason NO_AP_FOUND

 

                STAMODE_DISCONNECTED

                Disconnect from SSID Braun, reason 201

 

                STA - DISCONNECTED

                SSID: Braun

                BSSID: 00:00:00:00:00:00

                reason: 201

 

 

                Queue empty

reconnect

 

V

on: parth-garg [mailto:notifications@github.com] 
Gesendet: Samstag, 25. November 2017 13:23
An: esp8266/Arduino
Cc: arnold-b; Comment
Betreff: Re: [esp8266/Arduino] ESP12e WiFi not working using Arduino IDE, but working well in lua (#2265)

 

The default firmware doesn't have the wifi module

devyte, 2017-11-26:
I am closing this due to lack of feedback from Op in over a month, and due to age.
@arnold-b if you need assistance getting started, please refer to a community forum. If you think there is a problem in the code or tools hosted in this repo, please open a new issue and fill in the details requested.
A tip off the top of my head: erase the entire flash using esptool, and only then flash a new sketch.

closed 2017-11-26 by devyte

arnold-b, 2017-11-26:
Hello, i just want to let you know. 

 

Suddenly it works. That’s really strange. 

 

I build a second Module. This one worked immediate. Then I tried the other module and suddenly this one also worked. 

 

Closed #2265 <https://github.com/esp8266/Arduino/issues/2265> .

parth-garg, 2017-11-28:
> AfTd2VujhT37akoDoCYouj7fQPtfoDABks5s6YutgaJpZM4JJuxN> . <

Sorry I didn't reply was a little busy,
Listen closely
1. Order a custom node mcu build as per your requirements.
2. When you get it, use the floating point version of that (we use the
floating point for added precision, the integer should work fine as well).
3. Use node mcu flasher to flash it to your esp
4. Before flashing bring esp to flash mode(press reset button, while
holding reset press flash button, leave flash button and then the reset
button)
5. Select your custom build and flash it on to the esp on the required
comport

parth-garg, 2017-11-28:
Also even if you're using arduino after having flashed the firmware, it
should work with arduino, though you might need to flash it again for kya
before working on that.

arnold-b, 2017-11-28:
Hello, 

 

thank you. 

 

You are right after i got it working with lua, it worked also in arduino.

zeitwesen, 2019-05-14:
not sure if it is related - I had the problem that I did not see any networks on a WifiScan nor could I open up my own AP... 

problem was, that this particular ESP8266-12 was using crystal frequency of 26Mhz... 

Fix: Arduino IDE -> Tools -> (Selected Board: Generic ESP8266 Module) -> Crystal Frequency -> 26MHz

Maybe it can help somebody

tokitahmid4, 2020-05-09:
> Select NodeMCU 1.0(ESP-12E Module) as Board with the default settings,works like a charm

thank you "parth-garg"  it works

maximo80, 2020-05-19:
> not sure if it is related - I had the problem that I did not see any networks on a WifiScan nor could I open up my own AP...
>
> problem was, that this particular ESP8266-12 was using crystal frequency [...]

@zeitwesen  I would like to thank your post..... I've been looking a solution days and days..... but ESP12F didn't connect with nothing.... I was desperate but with "ONLY" change crystal frequency to 26 Mhz everything works fine..... So THANKS a lot !!!!!!
This forum is very useful thanks to everybody.

tokitahmid4, 2020-05-19:
Bro what is is problem . tell me details .

tokitahmid4, 2020-05-19:
ESP12-E / ESP12-F   select  Board  NodeMCU 1.0 .   Do you want to connect  ESP12E with arduino  or  USE ESP12E  Standalone ?    if you connect ESP12E with arduino  ,  first you should flash ESP12E .  if you need any help ,  contract me  tokitahmid4@gmail.com

tokitahmid4, 2020-05-19:
Bro How can i help you ?

devyte, 2020-05-19:
Please don't include your email address in a post/comment.

tokitahmid4, 2020-05-19:
OK

maximo80, 2020-05-19:
thanks @tokitahmid4 I built a pcb with ESP12F but with ESP8266 arduino ide parameters didn't work, but only changing crystal frequency to 26 mhz in arduino ide, everything was oK. 
I didn't check if i put the board as nodemcu works as well. I'll do.
thanks to everybody for your help