esp8266/Arduino #2268: How can I interface SPI based device with ESP8266. Any Example?

anujmattoo, 2016-07-12:
#### Hardware
Hardware:			ESP-12
Core Version:      	2.1.0-rc2

### Description

Problem description

I can not find any SPI example based on ESP8266. Which pins to use to interface SPI with ESP-12E/F. I have to interface 7 segment led driver using SPI. Please help!

### Settings in IDE

Module:  			ESP-12E
Flash Size: 	              4MB/1MB
CPU Frequency:		80Mhz
Flash Mode:			qio
Flash Frequency:	       40Mhz
Upload Using:		OTA / SERIAL
Reset Method:		ck / nodemcu

pipi61, 2016-07-12:
spi example in arduino folder : )
c:\Arduino\hardware\esp8266com\esp8266\libraries\TFT_Touch_Shield_V2\
or find with google ili9341 library
https://www.youtube.com/watch?v=RzduRWkV2LM
https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#spi
find max7219 drivers etc
https://github.com/aniline/arduino-sketchbook

devyte, 2017-10-15:
Closing per #3655 .

closed 2017-10-15 by devyte