esp8266/Arduino #2269: wdt reset during WiFi.scanNetworks() in WIFI_AP_STA mode

dks-xueliu, 2016-07-12:
Hello

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:	NodeMCU v1.0
Core Version:  2.3.0 

### Description

In my application, I need both station and AP mode. I modify the WiFiScan example to enable WIFI_AP_STA mode and shorten delay time after each scan to 1s. The problem is that the module will reset due to wdt after uncertain times of wifi scan, and the message show below:

 ets Jan  8 2013,rst cause:4, boot mode:(3,6)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

Does anyone has anyidea ?

### Settings in IDE

Module:  			ESP-12E
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			dio
Flash Frequency:	40Mhz?
Upload Using:		SERIAL
Reset Method:		nodemcu

### Sketch

```cpp

#include "ESP8266WiFi.h"

const char WiFiSSID[] = "WiFi_12345";           // predefined ssid
const char WiFiPSK[] = "WiFi_Password";         // predefined password

void setup() {
  Serial.begin(115200);

  // Set WiFi to station mode and disconnect from an AP if it was previously connected
  WiFi.mode(WIFI_AP_STA);
  //WiFi.disconnect();
  WiFi.softAP(WiFiSSID, WiFiPSK);
  delay(100);

  Serial.println("Setup done");
}

void loop() {
  Serial.println("scan start");

  // WiFi.scanNetworks will return the number of networks found
  int n = WiFi.scanNetworks();
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.print(i + 1);
      Serial.print(": ");
      Serial.print(WiFi.SSID(i));
      Serial.print(" (");
      Serial.print(WiFi.RSSI(i));
      Serial.print(")");
      Serial.println((WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*");
      delay(2);
    }
  }
  Serial.println("");

  // Wait a bit before scanning again
  delay(1000);
}
```

### Debug Messages

```
messages here
```

pasko-zh, 2016-08-24:
I cannot reproduce your error, i.e. (soft) WDT reset. However, it could be that one iteration of your loop, especially due to `int n = WiFi.scanNetworks();` takes longer than 3.2 seconds and will thus trigger a Soft WDT reset? 
I've measured one iteration being around 2.1 seconds, independant of CPU speed and number of networks. If you would like to disable the software watchdog and would like to use the hardware watchdog alone, you refer to my comment [here](https://github.com/esp8266/Arduino/issues/2261#issuecomment-241754253). Then, you will have up to 7--8 seconds until the hardware watchdog bites.

devyte, 2017-10-15:
@dks-xueliu I also can't reproduce on my ESP12-based borads.
I suggest increasing the time between scans. The ESP has a single radio, so every time it scans it channel hops, and drags both STA and AP interfaces along for the ride.
In addition, don't delay() for so long. Instead, poll the elapsed time with millis(), and if sufficient time has passed, execute the scan.
Closing as can't reproduce.

closed 2017-10-15 by devyte