esp8266/Arduino #2270: SPI modes 2 and 3 are reversed

vicnevicne, 2016-07-12:
#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

The SPI modes 2 and 3 are reversed, at least regarding generation of MOSI signal

This can be demonstrated using the sketch below and switching between modes.
See the attached scope captures of this sketch using an ESP vs using an Arduino:
[SPImodes MOSI AVR vs ESP.zip](https://github.com/esp8266/Arduino/files/360084/SPImodes.MOSI.AVR.vs.ESP.zip)

That was discovered by chance when working on using ESP as SPI slave and I think care will have to be taken because just blindly swapping the settings could cause other issues, in particular regarding the MISO sampling logic. 
See for example http://www.esp8266.com/viewtopic.php?f=32&t=10579&start=20#p50652

### Sketch

```
#include <SPI.h>

/*
    GPIO    NodeMCU   Name  |   Uno
   ===================================
     15       D8       SS   |   D10
     13       D7      MOSI  |   D11
     12       D6      MISO  |   D12
     14       D5      SCK   |   D13
*/

int mode = SPI_MODE0;
long lastOutputTime = 0;

void setup() {
  Serial.begin(115200);
  SPI.begin();  
  Serial.println("Outputting 0b10101010 on MOSI every second.\nEnter 0,1,2 or 3 to switch SPI mode");
}

void loop() {
  // Check If mode change is required
  while (Serial.available() > 0) {
    char inputMode = Serial.read();
    if (inputMode >= '0' && inputMode <= '3') {
      switch(inputMode) {
        case '0': mode=SPI_MODE0; break;
        case '1': mode=SPI_MODE1; break;
        case '2': mode=SPI_MODE2; break;
        case '3': mode=SPI_MODE3; break;      
      }
      Serial.println();Serial.print("Switching to SPI mode ");Serial.println(inputMode);
    }
  }
  
  long now = millis();
  if (lastOutputTime < now - 1000) {
    lastOutputTime = now;
    Serial.print(".");
    SPI.beginTransaction(SPISettings(250000, MSBFIRST, mode));  
    SPI.transfer(0xAA);
    SPI.endTransaction();
  }
}
```

whatnick, 2016-11-14:
I am currently solving this using conditional compile for libraries targeted at both ESP8266 and standard Arduino.

devyte, 2017-10-15:
Closing as duplicate of  #2416 .

closed 2017-10-15 by devyte