esp8266/Arduino #2271: is there any way to refresh or restart the server ???

mushfiq-cse05-buet, 2016-07-12:
whenever i run a server such as LED on/off, after uncertain amount of time the server get lost or inaccessible from my browser, so i'm looking for a way to refresh or restart the server again ?? TIA

krzychb, 2016-07-13:
Does https://github.com/esp8266/Arduino/issues/2272#issuecomment-232256697 aswer your question or this is another issue?

mushfiq-cse05-buet, 2016-07-13:
@krzychb  no its an another issue i wanto refresh the server after a certain interval of time

mushfiq-cse05-buet, 2016-07-13:
@krzychb 
the reason behind this refreshing is , whenever i run a simple LED ON/OFF server on my esp-01, it runs smoothly for 2-3 hours, but after this interval of time the server get lost or not accessible from my browser! restarting the module by ESP.restart() is a solution but as it restart the whole module and the LED is connected to gpio, the last state of the LED also get lost! so i think i need to refresh the server after a certain interval to keep state of connected LED unchanged  TIA

krzychb, 2016-07-13:
@mushfiq-cse05-buet,

You can save state of LED in [EEPROM](https://github.com/esp8266/Arduino/blob/master/doc/libraries.md#eeprom) before executing ESP.restart().

However, I believe there is something fundamentally wrong with module, server's application or network if server gets lost or is not accessible from browser after 2-3 hours. I would connect esp-01 running server to a serial terminal, enable additinal [debugging](https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md) if required and look for clues why connection is lost. 

I have been [testing](https://github.com/krzychb/EspFire2012) several esp modules that run webserver together with led strip control. If application is compatible, configured and tested for ESP, I did not face any communication / performance issues for several days (I was not patient enough to run them for longer time).

mushfiq-cse05-buet, 2016-07-14:
@krzychb thanks fro ur reply, but is there any function which can refresh or restart the server object ???

rima888, 2017-03-20:
Hi, I have the same problem. Do you found a solution to the problem? I'm appreciate if you shared to as.

vvrona, 2017-10-07:
Siemka krzychb i have that same problem that you have, i have relay to switch light on my desk but after couple ours esp is not respond. i change a esp board for new one and now it work one day, and don`t response again. Do you solved this ?

d-a-v, 2017-10-08:
Could you please try PR #3362 and report back ?
Could you please share a simple sketch that would allow us to reproduce the problem ?
Thanks

devyte, 2017-10-15:
@mushfiq-cse05-buet waiting on feedback.

devyte, 2018-01-08:
Closing due to no feedback in over 2 months. I think this issue is the same as, or related to, #2330 .

closed 2018-01-08 by devyte