esp8266/Arduino #2272: how to reset the module periodically ???

mushfiq-cse05-buet, 2016-07-13:
is there any way to reset the module after a certain amount of time or periodically ???

krzychb, 2016-07-13:
Hi @mushfiq-cse05-buet,

To reset the module you can use ``` ESP.restart() ```.
Please check [this FAQ item](https://github.com/esp8266/Arduino/blob/master/doc/faq/readme.md#i-have-observed-a-case-when-esprestart-doesnt-work-what-is-the-reason-for-that) that is related to use of use ``` ESP.restart() ```.
To execute reset periodically you can use [TickerScheduler](https://github.com/Toshik/TickerScheduler)

Krzysztof

devyte, 2017-10-14:
@mushfiq-cse05-buet after a serial upload, you have to physically reset the ESP to assure correct functionality.
Aside from that, your instability most likely is due to some mem corruption, or too long timing.
Not the right place for such dicussion, closing per #3655 .

closed 2017-10-14 by devyte