esp8266/Arduino #2273: Bricked 3 modules during flash

shibby14, 2016-07-13:
I seem to have managed to ruin 3 of my Olimex MOD-WiFi-ESP8266-DEV modules by trying to flash them. I've been working with these exact same modules quite a long time, so I know the procedure for flashing well enough. I've been careful to connect the GPIO0, GPIO2 and GPIO15 to GND, VCC and GND when resetting the module, but right on the first try I got:

_warning: espcomm_send_command: didn't receive command response
warning: espcomm_send_command(FLASH_DOWNLOAD_DATA) failed
warning: espcomm_send_command: didn't receive command response
error: espcomm_upload_mem failed_ 

After that none of the modules accept serial input and either spew out gibberish, or simply do nothing. Flashing or erasing memory doesn't work either. Before attempting to flash the modules I made sure they were all operational, and the serial com link was functioning correctly. At this point I'm desperately looking for ideas as to why this could have happened and how to avoid ruining any more modules. The verbose was not very useful in finding out the cause:

_esptool v0.4.9 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from COM1 to COM12
	setting address from 0x00000000 to 0x00000000
	espcomm_upload_file
	espcomm_upload_mem
	setting serial port timeouts to 1000 ms
opening bootloader
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
resetting board
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
trying to connect
	flush start
	setting serial port timeouts to 1 ms
	setting serial port timeouts to 1000 ms
	flush complete
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
	read 0, requested 1
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed_

To me it seems like the module is simply not responding. Is there anything that I can try to do to get it back into working order? The setup is pretty straight forward, I have a lab power supply pumping out 3.25V to the chip plugged into a breadboard and a PL-2303 3.3 serial to USB connecting it to a win 10 machine running arduino IDE (1.6.6 and 1.6.9). This exact setup worked 4 months ago, the only thing that has changed was that I upgraded to win10 (from win7) so that basically all of the drivers, including PL-2303, stopped working. Could all of this be caused by the serial to USB connector?

me-no-dev, 2016-07-13:
Have you tried contacting Olimex and asking them if they have idea? Nothing on our side has changed to possibly lead to any such error.

shibby14, 2016-07-13:
The hardware I'm using is exactly the same I used 4 months ago, the only thing that changed is the new drivers for the serial cable and the latest version of Arduino IDE. 

Well if its not the tools then perhaps its the serial cable... I'm reading 3.5-3.6V on the Tx wire of the cable - is that something I should be worried about?

themindfactory, 2016-07-13:
Would be nicer if it was 3.3V or at the same level as the VCC on the ESP,
so if 3V on ESP 3V on the RX pin would be nice. Can not see how you can
brick them as the bootloader is in the ESP....

shibby14, 2016-07-14:
Isn't it possible that 3.6V on the Rx pin could have done this? It's a solid cable&adapter, covered in rubber&plastic, so I have no way of changing the voltage level on the Tx wire. I've been using the same cable for a long time, but now that I had to install new drivers for it I'm getting worried that it became an ESP8266 killer. 

I used this same cable to issue AT commands to one of the modules before I bricked it - and it worked. However the moment I restarted the module and attempted to flash it I got the errors and the it stopped working. Now I have three unresponsive modules, two of which output unending stream of gibberish, while the third simply sits quietly and does nothing. 

Perhaps my setup is wrong... This is how I connect the pins in order to flash the module:
[image]
Am I doing anything wrong?

pgScorpio, 2016-07-14:
Last year I had a similar problem with 2 ESP07 modules that where bricked after I tried to upload a specific sketch (at that time with ide 1.6.1 and I think it was esp sdk 1.5.4). 
I never found out why they bricked. (I suspect the sketch was  compiled for the wrong device, but uploaded to the esp by esptool anyway), but after that esptool failed to connect to the device just like you describe above.
However, I managed to revive them by erasing the flash using the **nodeMcu flasher tool**. For that purpose I created a number of blank bin files (see attached zip). 
I remember it took me some attempts before it actually worked, but finally they came to live again and maybe this will work for you too.

[blank_bins.zip](https://github.com/esp8266/Arduino/files/363697/blank_bins.zip)

shibby14, 2016-07-15:
@pgScorpio You were right! After some brute-forcing I noticed that if I changed COM ports every time I attempt to flash the modules nodemcu would actually discover MAC addresses and manage to flash the INTERNAL://BLANK file onto the modules. After that I would try the Arduino IDE again and the module would break once again. 

When I tried flashing the 2M (the specified size of Olimex flash) file you attached nodemcu would show an error after a while into the process, so I tried smaller files and eventually 64k worked. After that I have been able to move up in steps to 512k (esptool.py erase still didn't work though) and after changing the board to "Generic ESP8266 Module" in Arduino (512k, 64 SPIFFS) i have been able to flash sketches to the modules to get them to work! It appears as though I'm limiting the storage space, but hey, at least it works. So thanks for your help!

Now I'm wondering why this could have happened. Aren't all routines for flashing new firmware stored on the chip itself? Is the filesystem something only created when flashing using Arduino IDE, or is it embedded in all the generated binaries no matter what tool is used?

pgScorpio, 2016-07-15:
@shibby14 I'm glad i could be at least of some help to you.

Some additional info: (might help you even further)

1: I used the nodeMcu flasher tool to erase the flash because this tool hardly does any error checking (at least not at that  time). It just checks IF there is a response without checking the response itself. If you got an error other than 'no-response' or 'timeout' while flashing the 2M empty bin, you might need to use an older version of the tool or try other flash configuration settings (processor/flash  freq?, erasing might take longer than the tool expects).
2: Normally the flash is written as a chain of blocks and each block has its own header which also contains the block size. Erasing "random" blocks by writing an empty bin file to it might (and probably will) break this chain.
3: Normally the last block contains the device parameters, including flash type and size. (which cannot be found if the chain is broken)
4: Any tools depending on information read from the flash itself will no longer work (at least not as it should) after (partly) erasing the flash and breaking the chain of blocks.
5: Check [Espressif Getting Started Guide](http://www.espressif.com/en/support/explore/get-started/esp8266/getting-started-guide#) for more info. 
Once you have got your chip (partly) running again you can uses Espressif's Firmware Download Tool to check the flash type and if you know which flash chip is on your board you can also use this tool to write the correct settings. Most of the time using this tool to re-flash an original (AT-)Firmware from the Espressif site, with the correct device settings, will fully recover the device.  (See chapter 6 of the Getting Started Guide)

Good luck !

jjssoftware, 2016-07-19:
+1 on using the nodemcu flasher to unbrick esp boards. I've revived a couple of esp12f boards using this tool where other methods failed. I'm generally a 100% linux user but the various flavours of esptool on linux were no good when I was trying to resurrect my dead esps. Shame it's windows only at the moment

devyte, 2017-10-15:
Closing as resolved.

closed 2017-10-15 by devyte

gazialankus, 2019-01-16:
Hi @pgScorpio, sorry to resurrect an old thread. I have two of [these](https://www.aliexpress.com/item/SESP8266-ESP-12-ESP-12F-CH340G-V2-USB-WeMos-D1-Mini-WIFI-Development-Board-D1-Mini/32958249491.html?spm=a2g0s.12269583.0.0.132e2e34HEqXVc&fbclid=IwAR1dWU-Zg8_ZPzgnwWYQe3pmyzzPaTEoAkI3dKNGtiNOswz2kkOAPjIXwvI), unfortunately one is bricked. I was able to use nodemcu flasher to flash the 2M blank file @pgScorpio provided. But I can't flash anything else now. I tried many things. Here's my esptool output: 

```
λ esptool.py --port COM4 flash_id
esptool.py v2.6
Serial port COM4
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
MAC: 84:0d:8e:8c:af:da
Uploading stub...

A fatal error occurred: Failed to write to target RAM (result was 0107)
```
I get the same error with different baud rates, trying to flash, etc. 

Any suggestions?

gazialankus, 2019-01-18:
Hi @pgScorpio, thank you so much for your help! 

I found the oldest nodemcu flasher [here](https://github.com/nodemcu/nodemcu-flasher/blob/d547d55b4a5306ba4bcc1b77cecbb7de52b6f754/Win64/Release/ESP8266Flasher.exe) (Dec 7 2014) and it won't flash the 4m blank file: 
[image] 

Here's what the working device shows with Espressif's Firmware Download Tool: 
[image]

Here's what happens with the bad device: 
[image]

And here are the things that seem to work with this bad device: 

1. I can flash your 2M blank file with nodemcu flasher. I can't flash AT firmware etc that I found around. 
2. I get output in serial. I connect to it with PUTTY with 74880 baud (found by trial and error). Upon connection I get this: 
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

ets_main.c
```
when I press reset I get this:
```
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

ets_main.c
```
when I connect D3 to GND (supposed to be the flash button) I get this: 
```
 ets Jan  8 2013,rst cause:2, boot mode:(1,7)
```
(hoping that I took it to a flashing mode, I tried the things that I had tried after doing this, but nothing changed)

I tried to read Chapter 6 of the getting started guide, but I have no idea how to check the device parameters and correct them, or copy one device's flash to another. Could you tell me what to try? I'm close to declaring defeat and you're the only hope I have left :)

Thanks!

JiriBilek, 2019-01-18:
Have you connected your device via Rx, Tx and serial to USB module (e.g. FTDI)? I have bad experience with these chinese clones especially with their USB.

pgScorpio, 2019-01-18:
@gazialankus Ok, 1 final attempt (But it does not look good)

To be sure the GPIO pins should be connected as follows:

GPIO 1 (U0TXD) = mode: 
-Always connected to Vcc (= normal operation. Gnd = Factory Test mode)
GPIO 2,0,15 = bootsel: 
-GPIO 2: To Vcc via a 3.3k - 5.6k resistor (often 10K, but sometimes this does not work ok)
-GPIO 0: To Vcc via a 3.3k - 10k resistor (**This is the FLASH pin**, connect to Gnd for flash mode)
-GPIO15: Connected to Gnd (Vcc = SD card, Gnd = Serial flash).
GPIO 10,9,7 = sdio_bootsel: 
-Don't care as long as GPIO15 is connected to gnd, but pullup/pulldown's are recommended (so don't leave floating or just connected to an input).

But:
When resetting in FLASH mode you get .... rst cause:2, boot mode:(**1**,7)
where boot mode(1,*) means:
1=bootsel=UART flash mode, 
7=sdio_bootsel=don't care if bootsel < 4.

So this seems to be OK,.
But since the Espressif download tool gives "ERROR" and does not show "DETECTED INFO" nor AP and STA mac addresses your module probably has a bad connection or defective flash chip....
So finding a bad solder joint or damaged track might be your last hope (unless you wanna try to replace the flash chip too)....

PS: 

Reading back the flash can be done with running esptool.py from the command line:
i.e.:
./esptool.py -p PORT -b BAUDRATE read_flash 0 FLASH_SIZE FLASH_FILE.bin

(Fill in correct PORT, BAUDRATE, FLASH_SIZE and desired FLASH_FILE. See esptool.py documentation)

Changing the device parameters can be done with the Espressif download tool:
Either: Flash a downloaded 'entire flash' binfile downloaded from another (same model) module with DoNotChgBin checked to set the parameters from the bin file.
Or:
-Select the wanted SpiFlashConfig parameters.
-Make sure DoNotChgBin is unchecked.
-Flash any binfile that fits the device.