esp8266/Arduino #2274: rst cause:2, boot mode:(1,7)

simondid, 2016-07-13:
### Basic Infos
rst cause:2, boot mode:(1,7)
#### Hardware
Hardware:			esp-01

### Description

so i'm making a esp-01 controle my sonos device this requires me to make a http post and handle the response from the sonos device i goth the option to set the volume on the sonos working but when i'm traying to read the response my esp creashes / resets i testet alle the code and every thing works with out creashes untill i run the "getresponse" function 

i goth gpio 2 pull high at alle time 
gpio 0 is pull low on boot to upload code then a while loop checks that the state changes to high before starting to do any none setup code 

i need gpio 2 and 0 to be in a high state under normal operation for my rotary encoder

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k (64k spiffs)
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

```
#include <ESP8266WiFi.h>
String volume_start = "<s:Envelope xmlns:s=\"http://schemas.xmlsoap.org/soap/envelope/\" s:encodingStyle=\"http://schemas.xmlsoap.org/soap/encoding/\"><s:Body><u:SetGroupVolume xmlns:u=\"urn:schemas-upnp-org:service:GroupRenderingControl:1\"><InstanceID>0</InstanceID><DesiredVolume>";
String volume_end = "</DesiredVolume></u:SetGroupVolume></s:Body></s:Envelope>";
#define SONOS_PAUSE "<s:Envelope xmlns:s=\"http://schemas.xmlsoap.org/soap/envelope/\" s:encodingStyle=\"http://schemas.xmlsoap.org/soap/encoding/\"><s:Body><u:Pause xmlns:u=\"urn:schemas-upnp-org:service:AVTransport:1\"><InstanceID>0</InstanceID></u:Pause></s:Body></s:Envelope>"
#define SONOS_PLAY  "<s:Envelope xmlns:s=\"http://schemas.xmlsoap.org/soap/envelope/\" s:encodingStyle=\"http://schemas.xmlsoap.org/soap/encoding/\"><s:Body><u:Play xmlns:u=\"urn:schemas-upnp-org:service:AVTransport:1\"><InstanceID>0</InstanceID><Speed>1</Speed></u:Play></s:Body></s:Envelope>"

 #define PLAY 1
#define PAUSE 0
int vol = 0;

int brightness = 0;    // how bright the LED is, start at half brightness
int fadeAmount = 10;    // how many points to fade the LED by
unsigned long currentTime;
unsigned long loopTime;
const int pin_A = 0;  // pin 12
const int pin_B = 2;  // pin 11
unsigned char encoder_A;
unsigned char encoder_B;
unsigned char encoder_A_prev=0;

WiFiClient client;
void setup() {
  // put your setup code here, to run once:
 // put your main code here, to run repeatedly:
 String volume = volume_start + vol + volume_end;
Serial.begin(115200);
  pinMode(pin_A, INPUT);
  pinMode(pin_B, INPUT);
  while(digitalRead(pin_A)==LOW){
	  Serial.println("gpio o = low wating");
	  delay(10);
  }
  delay(1000);
  currentTime = millis();
  loopTime = currentTime;
Serial.print("connecting");
WiFi.begin("solystvej","solystvej8723241400");

while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  SetVolume(client,0);

Serial.println("don");
}

void loop() {
 currentTime = millis();
  if(currentTime >= (loopTime + 5)){
    // 5ms since last check of encoder = 200Hz
    encoder_A = digitalRead(pin_A);    // Read encoder pins
    encoder_B = digitalRead(pin_B);

    if((!encoder_A) && (encoder_A_prev)){
      // A has gone from high to low
      if(encoder_B) {
        // B is high so clockwise
        // increase the brightness, dont go over 255
        Serial.println("clockwise");

        if(brightness + fadeAmount <= 100) brightness += fadeAmount;
    SetVolume(client,brightness);
      }
      else {
        // B is low so counter-clockwise
        // decrease the brightness, dont go below 0
        Serial.println("counter clockwise");
        if(brightness - fadeAmount >= 0) brightness -= fadeAmount;
     SetVolume(client,brightness);
      }

    }
    encoder_A_prev = encoder_A;     // Store value of A for next time

    // set the brightness of pin 9:
//    analogWrite(9, brightness);

    loopTime = currentTime;  // Updates loopTime
  }
  // Other processing can be done here

}
void getVolume(WiFiClient client){
	Serial.println("Getting Volume level");

	if(!client.connect("192.168.1.118",1400)){
	    Serial.println("ERROR unable to connect to sonos");
	    return;
	  }

}
void SetVolume(WiFiClient client,int level){
   String volume = volume_start + level + volume_end;
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
char buf[512];

if(!client.connect("192.168.1.118",1400)){
    Serial.println("ERROR unable to connect to sonos");
    return;
  }

  client.println("POST /MediaRenderer/GroupRenderingControl/Control HTTP/1.1");
  client.println("Connection: close");
  client.println("Host: 192.168.1.118:1400");
    int i =403;
    int cmd = PLAY;

   sprintf(buf, "Content-Length: %d", volume.length());
   Serial.println(buf);
  client.println(buf);
  client.println("Content-Type: text/xml; charset=\"utf-8\"");
  Serial.println("response");
  sprintf(buf, "SOAPACTION: \"urn:schemas-upnp-org:service:GroupRenderingControl:1#SetGroupVolume\"");
  client.println(buf);

  client.println("Accept: */*");
//   strcpy(buf, (cmd == PLAY) ? SONOS_PLAY : SONOS_PAUSE);

   client.println(volume);
   Serial.println("response");
   getResponse();

}
void getResponse() {
	Serial.println("response code");
	int timeout = millis() + 5000;
	  while (client.available() == 0) {
	    if (timeout - millis() < 0) {
	      Serial.println(">>> Client Timeout !");
	      client.stop();
	      return;
	    }
	  }
	  while(client.available()) {
	    String line = client.readStringUntil('\r');
	    Serial.print(line);
	  }
}

```

### Debug Messages
```
....
WiFi connected
IP address: 
192.168.1.111

WiFi connected
IP address: 
192.168.1.111
Content-Length: 312
response
response
response code

Soft WDT reset

ctx: cont 
sp: 3ffef2e0 end: 3ffef750 offset: 01b0

>>>stack>>>
3ffef490:  3ffe8904 3ffe8688 3ffee6fc 40202ec4  
3ffef4a0:  3ffe8688 3ffe8688 3ffee680 40201d2a  
3ffef4b0:  00000000 00000000 3ffee6fc 40202ec4  
3ffef4c0:  3ffee6fc 3ffe8688 3ffef700 40201ebe  
3ffef4d0:  50414f53 49544341 203a4e4f 6e727522  
3ffef4e0:  6863733a 73616d65 6e70752d 726f2d70  
3ffef4f0:  65733a67 63697672 72473a65 5270756f  
3ffef500:  65646e65 676e6972 746e6f43 3a6c6f72  
3ffef510:  65532331 6f724774 6f567075 656d756c  
3ffef520:  40100022 000d4c26 3ffeda38 00000000  
3ffef530:  3ffee0b0 3ffeda38 00000000 4000050c  
3ffef540:  00000000 000cb9aa 40210c84 3ffeda38  
3ffef550:  3ffeda18 00008000 3ffed26c feefeffe  
3ffef560:  3ffed32c 40210c20 00000000 00000012  
3ffef570:  4020ee6e 4020edd5 00000002 3ffed940  
3ffef580:  4020ee15 00000000 00000012 00008000  
3ffef590:  4020f308 00000000 3ffed940 00008000  
3ffef5a0:  4020ed90 3ffed3e0 00000020 ffffffff  
3ffef5b0:  3ffed32c 00000000 40211288 40211040  
3ffef5c0:  4021ab36 3ffe8720 3ffe8714 00000000  
3ffef5d0:  4021b98b 3ffe8720 3ffe8714 00000000  
3ffef5e0:  3fff0114 00000002 3ffef64f 40203748  
3ffef5f0:  40202501 402024dd 3ffe8720 3ffee6fc  
3ffef600:  00000031 0000000a 3ffee6fc 40202e4d  
3ffef610:  3ffef64d 01c200fe 00001c00 00000000  
3ffef620:  796c6f73 65767473 3ffef64c 40202f61  
3ffef630:  00303034 201a653f 00000040 00000900  
3ffef640:  fe856d00 2031a03f efeffe40 313131fe  
3ffef650:  efeffe00 fee898fe ffdad03f 0000003f  
3ffef660:  00000009 73686c7c 73796c6f 6a657674  
3ffef670:  00000100 3ffef718 00000000 3ffef700  
3ffef680:  3ffee6fc 0000000a 00000003 40202f92  
3ffef690:  33323738 00000001 3ffe8905 40203748  
3ffef6a0:  40105072 0000000b 3ffe85db 3ffee728  
3ffef6b0:  3fffdad0 0000000d 3ffee6fc 40202e4d  
3ffef6c0:  3ffe8904 3ffee6c4 3ffee6fc 3ffee728  
3ffef6d0:  3fff12ac 0000013f 00000138 40202e78  
3ffef6e0:  6f01a8c0 00ffffff 0101a8c0 3ffee728  
3ffef6f0:  3fffdad0 3ffee6c4 3ffee6fc 40201ff3  
3ffef700:  3ffe8760 00000000 000003e8 feefeffe  
3ffef710:  3ffee680 3fff10b4 3fff0eb4 0000013f  
3ffef720:  00000138 feefeffe feefeffe feefeffe  
3ffef730:  feefeffe 00000000 3ffee720 402033fc  
3ffef740:  feefeffe feefeffe 3ffee730 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

```

krzychb, 2016-07-13:
Hi @simondid,

Have you tried to track down where exactly the code crashes?
I mean where exactly inside of ```getResponse ``` function.
If not, then [this guide](https://github.com/esp8266/Arduino/blob/master/doc/faq/a02-my-esp-crashes.md) may be helpful.

I suspect you are stuck for couple of seconds inside of ``` while (client.available() == 0) ``` loop, so the s/w watchdog timer expires and drops the stack trace.

Krzysztof

simondid, 2016-07-13:
yes i have a 5 second wait on that loop so i just trayed to set a 1 milisecond delay in the while loop witch seems to fix the problem :P 

so even after 5 seconds i have 0 replay  any idear why this is?

krzychb, 2016-07-13:
A missing newline at the end of POST request?
Incorrect `Content-Length`?

simondid, 2016-07-13:
thats the thing the sonos device executtes the command but i'm reciving no response so i'm assuming the initiere post i correct

qwerty12345asdf, 2018-10-21:
Hey, @simondid  did you find a solution?