esp8266/Arduino #2277: printf('%s', ptr) with NULL ptr don't display NULL anymore but crash

hallard, 2016-07-13:
Hi guys,

I discovered new crash on my working code a `printf("%s", NULL);`  now crash, It seems that before it printed "NULL" ?

I don't remember this problem was present (because my code always worked) and since a new git version (I'm using always latest git repo but I don't know which one done side effect because I did not recompiled this code since approx 1/2 months) it does not work anymore.

I know it's not excellent coding to use printf with null pointer (shame on me) but checking also inside printf is not a bad idea. Something has changed on this point ?

Here sketch and output
```arduino
void setup() {
  Serial.begin(115200);
  Serial.println("\r\nHello World");
}

void loop() {
  char * str = "It's me!!";

  Serial.printf("str='%s'", str );
  for (uint8_t i= 0 ; i<=5 ; i++ ) {
    delay(1000);  
    Serial.print('.');
  }

  str = NULL;
  Serial.printf("str='%s'", str );
  for (uint8_t i= 0 ; i<=5 ; i++ ) {
    delay(1000);  
    Serial.print('.');
  }
}
```

And Serial output give the following
```
Hello World
str='It's me!!'......
Exception (28):
epc1=0x402084e8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3ffef490 end: 3ffef8a0 offset: 01a0

>>>stack>>>
3ffef630:  40206c68 3ffef730 3ffe88b8 402032ec  
3ffef640:  00000009 feefeffe feefeffe feefeffe  
```

igrr, 2016-07-13:
We now use newlib for all C library functions. It doesn't handle null pointer in printf. If you think it has to, feel free to do a PR to newlib-xtensa.

hallard, 2016-07-13:
@igrr 
thanks for the information, I have no problem, just good to know, and let other know ;-)

closed 2016-07-13 by hallard