esp8266/Arduino #2280: mDNS related crash when linking with binary LWIP lib

everslick, 2016-07-14:
The following minimal sketch crashes on my (and on at least one other tested) WLAN reproducably. the crash does happen after around 2 seconds after "started" is printed. the crash does not occour, when i link against LWIP_SRC.

```
#include <Arduino.h>
#include <ESP8266mDNS.h>

const char *name = "esp";

void setup() {
  Serial.begin(115200);
  Serial.println();

  delay(1000);

  WiFi.mode(WIFI_STA);
  WiFi.begin("XXX", "xxx");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin(name)) {
    Serial.printf("MDNS: hostname set to '%s.local'\n", name);
  } else {
    Serial.print("MDNS: could not start mDNS responder\n");
  }

  Serial.println("started");
}

void loop() {
  Serial.print(".");
  delay(333);  
}
```

me-no-dev, 2016-07-14:
What do you mean by "link against LWIP_SRC" ? Using the open source? include some headers? I run source build of lwip constantly (and that produced the latest liblwip_gcc.a) with a whole slew of services - no problem so far.

me-no-dev, 2016-07-14:
oh, you say the xcc version is borked? That has igmp and other fixes missing that are in the gcc version (and the reason why gcc version is default)

everslick, 2016-07-14:
exactly. so this is a known issue then? still good to have it documented somewhere. :)

devyte, 2017-12-12:
@d-a-v since this is related to lwip, can you please take a look? If nothing can be done, then we should document what doesn't work, as observed above.

d-a-v, 2017-12-13:
@devyte I found no problem with both "Prebuilt (gcc)" lwip 1.4/2
@everslick 
This is an old issue, can you retry with master or 2.4.0rc2 ?
I added a ```MDNS.addService("MYTEST", "tcp", 82);``` and can see it with a bonjour browser on android. The esp stays up and running.

devyte, 2017-12-13:
@d-a-v I believe the issue is with espressif's xcc lwip, not with the prebuilt gcc.
At this point, I think we should retire the xcc option.

everslick, 2017-12-13:
Yes the bug only existed when linking against the xcc compiled lib. Imoh,
it is not relevant anymore.

d-a-v, 2017-12-13:
I indeed agree we can't do anything about the lwIP binary provided by espressif.

igrr, 2018-01-02:
Removed in 370e75cb4763d885ae2cb97785c85af018535eb9.

closed 2018-01-02 by igrr