esp8266/Arduino #2281: Conflict between File class defined in FS.h and SD

williamesp2015, 2016-07-14:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			ESP-12E
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		4MB
CPU Frequency:		80Mhz?
Flash Mode:			dio
Flash Frequency:	?40Mhz?
Upload Using:	 SERIAL?
Reset Method:		?ck / nodemcu?

I am suing SD card and defined File Datafile and also using SPIFFy to upload my HTML files into the ESP. When compiling says FS.h:48:7: error: previous definition of 'class fs::File'

luc-github, 2016-11-06:
there is a define for this to be used before header call
```
#define FS_NO_GLOBALS
#include <FS.h>
```
then you must add namespace fs for SPIFFS objects like 
`fs::File = currentfile = SPIFFS.open("/myfile.txt", "r");`  
but for SD : 
`File textfile = SD.open("/myfile.txt", FILE_READ);`

vniclos, 2017-05-02:
Thank's I,m beguinner and this is a great helpfull

devyte, 2017-10-15:
The referenced solution is present in latest git. Closing as resolved.

closed 2017-10-15 by devyte

PaulStoffregen, 2018-01-20:
Hi... Paul here, the guy who makes Teensy.  :-)

I know this may be a little off-topic, but I'm hoping to craft a FS.h that all core libraries can use.  My main goal is to be able to create libraries which *use* files, like a JPEG decoder, which can accept a generic File object that works with ESP spiffs and Arduino's SD & Bridge, and USB Host Shield's upcoming support for USB memory sticks, and long-term any library which provides access to files.

Can someone please point me to where we should discuss such a FS.h (if anyone's interested)?

igrr, 2018-01-20:
Hi Paul, please feel free to open a new issue in this repo for such discussion (and @devtye please don't close it as off-topic).

Alternatively, we could discuss on Arduino-dev mailing list, however personally I find Github to be more convenient.

devyte, 2018-01-20:
@igrr I don't consider it off topic :p and I just got involved myself in the Arduino discussion.
You are right, there are several different file systems, and currently they don't necessarily share an interface. Also, files from different file systems can't be treated in similar manner, because they alps don't share an interface.
The discussion is actually rather complex in the general Arduino context, because there seem to be a whole lot of different file systems out there, and with differing principles of operation.
Thing is, most Arduino developers think in terms of polymorphism: implement a base class, and derive objects specific to each type. Then, generalize by implementing methods and functions that take as argument a pointer to the base. This has overhead in memory, where we're most constrained
Paul's comment started me thinking along the lines of implementing a proof of concept here, based on templates, and with zero overhead in mind. If it works out, then we can present it to Arduino for extension to include other file systems.

duckylock, 2019-02-21:
Hi, I tried this methode, it seams to by-pass the class name conflict but i m still not able to read or write in the sd card or spiffs... in fact i m able to write a file but not to write inside this file... any idea or other same experience with this issue?

danbates2, 2019-04-14:
>  `fs::File = currentfile = SPIFFS.open("/myfile.txt", "r");`

I was banging my head at this until I removed an equals sign!

    fs::File currentfile = SPIFFS.open("/myfile.txt", "r");

Works for me.