esp8266/Arduino #2296: Error while receiving IR remote signal

scropion86, 2016-07-18:
### Basic Infos

My Hardware is based on ESP-12F and Three Relays and RGB LED MOSFET Drive , but in that case i am not using the MOSFETS here.
i connected my Hardware to WIFI using WIFI Manager library , and controlling it using Blynk library and App, also i use a IR receiver to manully control the Relays and send back the status to Blynk server/App using Virtual pin.

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.3.0

### Description

the problem is sometimes when the Remote sent IR to the Receiver connected to my Board GPIO0 the device is resetted.

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

i am using Platformio and select the board as esp12e

### Sketch

```cpp

//@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
//@@@@@@@@@@@ intialazing librarys @@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@
//@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@

//ESP8266 library
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
//For OTA
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

//####needed for WIFI Manager library###########
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager
//WiFiManager
//Local intialization. Once its business is done, there is no need to keep it around
WiFiManager wifiManager;

//###### Blynk Library ################
#define BLYNK_PRINT Serial    // Comment this out to disable prints and save space
#include <BlynkSimpleEsp8266.h>
char auth[] = "2d831d5f05eb4c809a22db721945ba7e";
WidgetTerminal terminal(V1); // V Pin For Blynk Terminal
//########IR Part ####################
#include <IRremoteESP8266.h>
int RECV_PIN = 0; //an IR detector/demodulatord is connected to GPIO pin 2
IRsend irsend(14); //@@@@@@@@@@@@@@@@@an IR led is connected to GPIO pin 0
IRrecv irrecv(RECV_PIN);
decode_results results;

//##### Simpletimer ######
#include <SimpleTimer.h>
SimpleTimer timer;

//Relay Status to be Synced with Blynk Buttons
int  Relay1 = 0 ; //GPIO 16
int  Relay2 = 0; //GPIO 4
int  Relay3 = 0; //GPIO 13

// IR Remote Emulator Multi Function buttons alternating Value
[... 479 lines cut ...]
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld

```

and decoded message is 
```
Decoding 26 results
0x40104a3c: RC_GetBlockAckTime at ?? line ?
0x401048f9: ppCalFrameTimes at ?? line ?
0x40212e20: readvdd33 at ?? line ?
0x40104e5e: rcGetRate at ?? line ?
0x402128e9: ram_pbus_debugmode at ?? line ?
0x40212d86: readvdd33 at ?? line ?
0x4020f6d8: igmp_timeout at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/igmp.c line 685
:  (inlined by) igmp_tmr at /Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/igmp.c line 668
0x40212e33: readvdd33 at ?? line ?
0x40215b09: phy_dig_spur_set at ?? line ?
0x40215b4e: phy_dig_spur_set at ?? line ?

```

igrr, 2016-07-20:
That is a bug in IrRemoteESP8266 library.
In this line:
https://github.com/markszabo/IRremoteESP8266/blob/master/IRremoteESP8266.cpp#L478
the ISR handler must be placed into IRAM, not into flash.

scropion86, 2016-09-28:
Here is the same library with the problem solved [https://github.com/alexmoz2016/IRremoteESP8266](url)

themindfactory, 2016-09-28:
The link with the problem solved still have the interrupt code in flash

RichardS

scropion86, 2016-09-29:
sorry i added the wrong branch before [here is the correct one](https://github.com/alexmoz2016/IRremoteESP8266/tree/patch-1).
the user alexmoz2016 corrected it with [this commit.](https://github.com/alexmoz2016/IRremoteESP8266/commit/51b9587deb5c97764f283b028d1c5a54d41d87dd)

i re-opened it to allow more user test it.

closed 2017-04-20 by scropion86