esp8266/Arduino #2297: Library v2.2.0 to v2.3.0 upgrade within Arduino IDE - boards 'generic' unknown

electronicsguy, 2016-07-19:
### Basic Infos

#### Hardware
Hardware:  ESP-01
Core Version:  2.3.0

### Description
I am using Arduino v1.6.9 on Windows 7 and was using esp8266 library v2.2.0 which was working perfectly. Yesterday, the IDE popped up a message saying the board has an update, so I went ahead and upgraded the library to v2.3.0. Now, I am not able to compile any sketch.

The board selected is "Generic ESP8266 Module", just like before. What is the problem?

### Debug Messages

Board generic (platform esp8266, package esp8266) is unknown

Error compiling for board Generic ESP8266 Module.

igrr, 2016-07-19:
Please check if this comment helps: https://github.com/esp8266/Arduino/issues/1387#issuecomment-228510966

electronicsguy, 2016-07-19:
@igrr Thanks Ivan! you nailed it. Is the problem with esp8266 library upgrade not doing a cleanup, or a bug in the Arduino boards manager?

igrr, 2016-07-19:
esp8266 library doesn't do any cleanup because it's just a set of files which can be compiled for the esp. It doesn't contain any scripts which run on your computer. I suppose it may be an issue with the boards manager for certain upgrade scenarios.

closed 2016-07-20 by igrr