esp8266/Arduino #2298: Add documentation for ESP8266WiFi library

igrr, 2016-07-19:
Provide methods documentation, driven by examples. I.e. not formal "doxygen" style, but rather through short code snippets which illustrate usage of different methods.

In addition to methods documentation, describe typical flows for the following actions:
- initiating connection to STA and checking result (blocking way and asynchronous way)
- setting up an AP (and getting callbacks when clients connect)
- scanning networks (blocking and asynchronous ways)
(these items may be among snippets mentioned above)

Mention persistent/non-persistent settings (i.e. "why does my Blink sketch set up an AP?").
Mention that ESP will automatically reconnect by default (there is no need to call WiFi.begin(stuff) each time connection drops).

Other classes to be documented include WiFiClient, WiFiServer, WiFiUDP, WiFiClientSecure. 
We can probably refer users to Arduino documentation, because unlike `WiFi` class, our changes to these are relatively minor. Just document additional APIs (multicast UDP, setNoDelay for TCP). For WiFiClientSecure, need to describe: a) way to verify remote server identity, and b) how to load client certificates from C arrays and file system. Also mention current limitations of WiFiClientSecure (needs lots of RAM, no TLS 1.2 support, etc).

Matthew-Bradley, 2016-07-21:
So I suppose this would take the form of various code examples demonstrating the use of the different classes in all the various different ways?

Matthew-Bradley, 2016-07-25:
@igrr I've started putting up a few examples here, which might provide a useful starting point:
https://github.com/Matthew-Bradley/ESP8266Examples

closed 2016-09-02 by igrr