esp8266/Arduino #2302: There is a function to get the board flash size ?

odilonafonso, 2016-07-19:
### Basic Infos

#### Hardware
Hardware:			Any ESP board

### Description

I want to know the module flash size.
There is a function to do this ?

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		????

Zeron, 2016-07-19:
Look at the CheckFlashConfig example that comes with the core.

https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/CheckFlashConfig/CheckFlashConfig.ino

odilonafonso, 2016-07-19:
Thank you @Zeron  !!!
Well, I will extend my problem.
I'm trying to use the area reserved for spiffs but I can not write any use of flash size options 1MB - gives error in execution.
I've tried several options without success.
Is there a way for me to know what spiffs size should I use?

odilonafonso, 2016-07-19:
I can use SPIFFS when set:
Flash Size: "512K (64K SPIFFS)"

odilonafonso, 2016-07-19:
After some tests, can use: Flash Size: "1MB (512K SPIFFS)".
Think is the correct size.

closed 2016-07-19 by odilonafonso