esp8266/Arduino #2303: TCP TIMEOUT

josep112, 2016-07-20:
I've tried several alterations but nothing works need to http.GET()  the time to be the maximum 1s, someone help me?

             http.setTimeout(1000);
             http.begin(ip+"/estado-dimmer.php?mac=DI:"+mac+"&codigo="+blynk_token);                  
             httpCode = http.GET();

SteveToulouse, 2016-07-28:
I'm not sure this is the source of the problem but this line in the HTTPClient library:

     `https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPClient/src/ESP8266HTTPClient.cpp#L924)`

resets the timeout whenever reply data becomes available. 

I'm probably missing something very obvious but, if not, I get the feeling that you can't guarantee that the whole HTTP request/reply duration will be strictly limited to the timeout you set. The timeout starts counting just after the request is sent so if the reply starts to arrive after timeout-1ms, you can have at least one more full `timeOut` duration before a timeout is detected.

devyte, 2017-10-14:
Closing per #3655 .

closed 2017-10-14 by devyte