esp8266/Arduino #2304: SDK 2.0 has been released

marvelousNinja, 2016-07-20:
http://bbs.espressif.com/viewtopic.php?f=46&t=2451

**ESP8266_NONOS_SDK_V2.0.0_16_07_19 Release Notes**:

1. Updated libphy.a to 1055, fixed an issue of large current in Light-sleep.
1. Updated AT+ to 1.3.0:
  1. Added Light-sleep wakeup command, AT+WAKEUPGPIO;
  1. Fixed abnormal AT+CWDHCPS IP allocation;
  1. Added at_sdio demo under example directory.
1. Fixed probable system default when calling cur and def interfaces.
1. Fixed the issue of high current in Deep-sleep mode without disabling SPI first.
1. Fixed an issue where the SDK would crash when switching from AP to Station mode by calling wifi_set_phy_mode.
1. Updated secondary boot loader to v1.6:
  1. Supports AT + based on SDIO;
  1. Supports entering testing mode through GPIO.
1. Added support for MXIC Flash QIO mode.
1. Fixed exception caused during TCP handshake and retransmission.
1. Fixed issues in ESP-NOW.
1. Added ESP-PAIR (Simple-Pair) feature, for APIs please refer to ESP8266 SDK API Guide.
wpa2-enterprise time function derivation and time check disable can be set by users.
1. Support for PEAP/TTLS in wpa2-enterprise, for APIs please refer to ESP8266 SDK API Guide.
1. Added mqtt demos under examples directory.
1. Other issue fixes.
1. Resolved an issue that RF_Init_data sector may be broken in stress test. Provided a function user_rf_cal_sector_set which has to be added in application by software developer. More details about user_rf_cal_sector_set refer to documentation "2C-ESP8266__SDK__API Guide". (Resolved in ESP8266_NONOS_SDK_V1.5.4.1)
1. Fix a potential risk that will cause rf not work after external reset. (Resolved in ESP8266_NONOS_SDK_V1.5.4.1)
1. Add SDIO AT support. (Resolved in ESP8266_NONOS_SDK_V1.5.4.1)
Fix a potential bug in espconn. (Resolved in ESP8266_NONOS_SDK_V1.5.4.1)

**ESP8266_NONOS_SDK Notes:**

1. esp_init_data.bin has to be downloaded into flash at least once.
1. Download blank.bin to initialize the sector stored RF_CAL parameter (set by user_rf_cal_sector_set), and download esp_init_data.bin into flash, when the system needs to be initialized, or RF needs to be calibrated again.

**AT Notes:**

1. Since the release of ESP8266_NONOS_SDK_V1.5.0, AT firmware is too large to use 4Mbit flash. Please choose flash with at least 8Mbit capacity.
1. Firmware upgrade over-the-air process is changed a bit. We will upgrade the latest AT firmware to Espressif Cloud server only after it has been tested and the overall performance is guaranteed. Users may not be able to use AT+CIUPDATE in ESP8266_NONOS_SDK_V1.5.0 and other more advanced versions.

igrr, 2016-07-20:
Thank you for taking time to copy the release notes here 👍

marvelousNinja, 2016-07-21:
WPA Enterprise is required for my use cases, so I was thinking to help a little bit more.
I am using Platformio and trying to compile my firmware with the new version of SDK (I just copied it into `~/.platformio/packages/framework-arduinoespressif/tools/sdk`).

However, some of SDK files were modified to fix some of the compilation errors: https://github.com/esp8266/Arduino/blob/4c587e28bc188efc852039606cac641768323ee3/tools/sdk/include/ets_sys.h#L160 (these prototypes are not defined in the original SDK version).

@igrr, is there any checklist for those modifications? I will probably dive into git history, but any pointers would be useful.

After copying those prototypes, I'm failing with this:
```
In file included from .pioenvs/nodemcuv2/FrameworkArduino/pgmspace.h:12:0,
from .pioenvs/nodemcuv2/FrameworkArduino/Arduino.h:240,
from .pioenvs/nodemcuv2/FrameworkArduino/Esp.cpp:21:
/Users/alexander/.platformio/packages/framework-arduinoespressif/tools/sdk/include/ets_sys.h:130:31: error: 'int_handler_t' has not been declared
```

igrr, 2016-07-21:
No there is no checklist for these changes, they have accumulated over time.
The way i do this is by diff-ing new SDK directory against the old SDK directory, and then applying the changes from patch file semi-manually to the arduino/tools/sdk tree. Usually there are only a few changes in the header files.

Bigger issue which you are going to face is that now we need to reserve an extra sector for RF calibration data somewhere (see documentation for `user_rf_cal_sector_set`). This means that linker scripts need to be changed to reserve another sector at the end of the application, before the beginning of SPIFFS. This will lead to some changes to OTA update code, because right now it assumes that the end of the application region is the beginning of SPIFFS, which will be no longer true. So, extra symbol has to be added to linker script to point to the real end of the application region.

If you are just hacking this and not interested in using EEPROM, for example, you could add a call to `user_rf_cal_sector_set` and point it to the sector used by EEPROM.

marvelousNinja, 2016-07-21:
Wrong button =) 
Thanks for the pointers. I hope I will get this running.

marvelousNinja, 2016-07-21:
A bit of an update.

I was able to compile the firmware, but it can't boot. Output from serial is cycling through this:
```
 ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v09f0c112
~ld
```

In short, here is what I've done:
* Followed your advice and compared diffs between new SDK and the old. I think, I've applied all custom patches along the way.
* Did this: `xtensa-lx106-elf-ar -d libmain.a time.o`, since compiler wasn't happy about duplicated time functions.
* In `ldscripts` platformio package I've edited `esp8266.flash.command.ld` and added new symbol (looks like it's the only new thing from SDK: `PROVIDE ( Enable_QMode = 0x400044c0 );`). I decided not to touch SPIFFS symbols for now.
* Copied the definition of `user_rf_cal_sector_set` from the examples in the SDK. I've placed it in `Esp.cpp` in the core.
* Stuck on this stuff:

  ```
  (.text+0x1cc): undefined reference to `memchr'
/Users/alexander/.platformio/packages/framework-arduinoespressif/tools/sdk/lib/liblwip.a(espconn_buf.o): In function `ringbuf_findchr':
  ```
  As a possible solution, replaced `libwip.a` and `libwip_gcc.a` from the new SDK with files from this repository (that actually worries me a bit). That allowed me to successfully compile the firmware, and load it on my ESP.

@iggr, do any of these issues remind you about something when you ported previous versions of SDK? I guess the next step would be to learn how to debug this stuff and get more information...

igrr, 2016-07-21:
Did you also remove mem_manager.o from libmain.a as mentioned in tools/sdk/lib/README.md?

Regarding replacing lwip with old version, this may not be a good idea. Just find an implementation of memchr (it's just three lines) and dump it somewhere in a C file.

marvelousNinja, 2016-07-21:
I've tried both with removing `mem_manager.o` and without it. Didn't seem to make much of a difference.

igrr, 2016-07-21:
Ok, I'm going to review code changes done in 2.0.0 sdk libraries early next week.

marvelousNinja, 2016-07-21:
Hell yes, I've got it to work!

So, indeed, I've tried to go with original `libwip.a` and defined `memchr` in one of C files.
As a result - `mem_manager` warnings started to fire around. I've removed it from `libmain.a` and sample firmware started working (it just prints 'Hello' into serial).

If that's OK, I can wrap up PR and we can coordinate what to do next? There's a chance that some libs (like SPIFFS) aren't working.

igrr, 2016-07-21:
Sure, feel free to post a PR.
Flash layout still has to be fixed though, and rf config sector setting has to be moved into core_esp8266_main.c
Also LwIP source files in tools/sdk/lwip have to be updated.

kaeferfreund, 2016-07-21:
Do we have to keep eeprom lib? I can't see any advantage over using SPIFFS apart from arduino compatibility...

igrr, 2016-07-21:
Arduino compatibility is very important, so removing EEPROM is certainly not an option :)

kaeferfreund, 2016-07-21:
Okay. Didn't know that

jpenninkhof, 2016-07-21:
Awesome to see that Espressif improved SSL/TLS and included WPA-Enterprise. Hope SSL based on mbedTLS will work with a little smaller memory footprint this time around. Anyway, thanks so much for your efforts of getting this into the Arduino SDK. Awesome job guys! Seriously appreciated!

igrr, 2016-07-21:
mbedTLS footprint is not smaller if you configure it to support 16k fragments used by most of the TLS servers, unfortunately. But at least we get TLS 1.2 support. This still has to be integrated into WiFiClientSecure.

wizzhi, 2016-08-08:
TTLS is very critical for me. I guess also for some others. So it will only be available in 2.5.0? What will be the earliest date? An chance to get it form some RC branches?

scargill, 2016-08-12:
Hi

I'm using the latest Arduino/ESP8266 - which I have to say - very good... but In converting my code - I note in my debug that this is still using 1.53 of the SDK - any idea when it will be upgraded to 2.0 and how we (I'm new to this) go about upgrading?

I'm not using the EEPROM as it has no backup in the event of power loss during update. I realised quickly that it is just as easy to use the Espressif functions which use 3 sectors to secure data in those circumstances. Really worth looking at for storing user info.

2.0???

igrr, 2016-09-09:
@Beliwars you can try the experimental version based on SDK 2.0.0:
- uninstall ESP core using boards manager
- install git version of the core using the following instructions: https://github.com/esp8266/Arduino#using-git-version
- when doing git clone in these instructions, add `-b update_sdk_2.0.0` parameter after `git clone` to check out the branch with SDK 2.0.0

Beliwars, 2016-09-09:
thanks @igrr, 

I'll clone the git version and install with the sdk_2.0.0 option, I'll let you know how everything goes.

thanks again.

regards

krzychb, 2016-09-09:
@Beliwars,

Please post your discoveries regarding `system_adc_read_fast` as well if possible. According to  [Espressif ESP8266 FAQs](https://espressif.com/sites/default/files/documentation/esp8266_faq_en.pdf)  you should be able to achieve 100k samples/sec (Wi-Fi modem turned off).

scargill, 2016-09-09:
> Hello Alex and Ivan,
>
> I'm trying to increase the sampling rate on the ESP's ADC and I noticed that on SDK 2.0.0 there is the function system_adc_read_fast and I'd like to use it to check how much can I increase the sampling rate; could you please tell me how can I upgrade the SDK on the arduino ESP core? I have v2.3.0 installed on my arduino IDE which is using SDK version 1.5.3.
>
> I'll appreciate any help you can give me, I'd rather continue using the Arduino IDE than setting up a new development environment around the SDK 2.0.0.
>
> Thanks in advance and regards.

Note that WIFI has to be disabled for this -

Beliwars, 2016-09-12:
Hello, 

I wrote a simple sketch converting 100 samples each time and then checking the time it took to perform the 100 samples; I filtered the output averaging 100 times the result and I obtained:

~528 us, 530 us (to perform the 100 ADC conversions) which led to a sample rate of about 188-189 KSPS; it was a basic test, no WIFI was configured on the ESP and the Operating clock was left @ 80MHz, I will change the system clock to 160 MHz to check results with the same sketch.

regards,

Beliwars, 2016-09-12:
Hello all, 

I just finished with the 160 MHz testing, this changes very little in the final result, since the system bus is always 80 MHz (ADC working clock = 80M/adc_clk_div) I'm using adc_clk_div = 8, which is the recommended value; so when I change the CPU working frequency I can go up to Sampling rate ~ 220 kSPS (100 samples = 453 us)

Below is the code I used: (Note I record the micros before calling system_adc_read_fast() and after I finish)

// Expose Espressif SDK functionality - wrapped in ifdef so that it still
// compiles on other platforms
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
}
#endif
int i=0;
long total = 0;
long tim = 0;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);

}

void loop() {
  // put your main code here, to run repeatedly:
  #ifdef ESP8266
    uint16_t adc_addr[100];
    uint16_t adc_num = 100;
    uint8_t adc_clk_div = 8;
    //Serial.println(system_get_sdk_version());
    long start= micros();
    system_adc_read_fast(adc_addr, adc_num, adc_clk_div);
    int tot = micros() - start;
    tim += tot;
    total += 100000000.0/tot;
    i++;
    if(i == 100){
      Serial.print("Sampling rate: ");
      Serial.println(total/100);
      Serial.print("It lasted: ");
      Serial.println(tim/100);
      i = 0;
      tim = 0;
      total = 0;
    }
  #endif

}

I also noted that as @scargill mentioned before, to use system_adc_read_fast, WiFi has to be disabled. Also, if ADC continuously sampling is needed, all interrupts have to be disabled, so PWM or NMI hardware timer can not be used when system_adc_read_fast is called.

My guess is that the achievable max sample rate would be less inside an actual application loop.

Beliwars, 2016-09-25:
Hello all, 

I've been trying to use this adc_read_fast function on an actual implementation (sampling one sample at each time and then performing some operations) I found that calling such function to convert only one sample gives a 60kSPS sampling rate.

timostr, 2016-10-10:
Problem: 
` ets Jan  8 2013,rst cause:4, boot mode:(3,7)

wdt reset
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v00000000
~ld
`
I installed as well the Beta based on the instructions in this discussion.
My problem is I can't sample more than 255 samples without running in watchdog reset.
Even if interrupts are disabled and watchdog is disabled. 
Anybody around to point me in the right direction?
`// Expose Espressif SDK functionality - wrapped in ifdef so that it still
// compiles on other platforms
#ifdef ESP8266
extern "C" {
#include "user_interface.h"
}
#endif
#define num_samples 255
uint16_t adc_addr[num_samples];
uint16_t adc_num = num_samples;
uint8_t adc_clk_div = 8;
int i = 0;
unsigned long start = 0;
unsigned long total = 0;
unsigned long tim = 0;
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
#ifdef ESP8266
  //Serial.println(system_get_sdk_version());
  start = micros();
  // uint16 *adc_addr : point to the address of ADC continuously fast sampling output. 
  // uint16 adc_num   : sampling number of ADC continuously fast sampling, range [1, 65535].
  // uint8 adc_clk_div: ADC working clock = 80M/adc_clk_div, range [8, 32], the recommended value is 8.
  wifi_set_opmode(NULL_MODE);
  system_soft_wdt_stop();
  ets_intr_lock( );  //close interrup
  noInterrupts();
  system_adc_read_fast(adc_addr, adc_num, adc_clk_div);
  unsigned int tot = micros() - start;
  tim += tot;
  total += num_samples * 1000000.0 / tot;
  i++;
  if (i == 100) {
  interrupts();
  ets_intr_unlock(); //open interrup
  system_soft_wdt_restart(); 
    Serial.print("Sampling rate: ");
    Serial.println(total / 100);
    Serial.print("It lasted: ");
    Serial.println(tim / 100);
    i = 0;
    tim = 0;
    total = 0;
  }
#endif
}`

Pedrotok, 2016-10-11:
I have a problem. I'm trying to compile the IoT_Demo example, but when I execute gen_misc.sh, it's not compiling and I'm getting this error:

**"multiple definition of `user_rf_cal_sector_set' "**

So I tried to comment the user_rf_cal_sector_set function defined on the user_main.c . When I do this, the binaries are generated, but I only see random characrters on the serial monitor.

I'm using this line at the beggining of the user_init: **uart_div_modify( 0, UART_CLK_FREQ / ( 115200 ) );** .

I was able to compile and execute programs before this new SDK 2.0.0 released.

gorghino, 2016-10-19:
@igrr I downloaded the SDK2.0.0 but If I compile, Android 1.6.11 tells:

fork/exec /home/gorgo/Scaricati/arduino-1.6.11/hardware/esp8266com/esp8266/tools/xtensa-lx106-elf/bin/xtensa-lx106-elf-g++: no such file or directory
Error compiling for board Generic ESP8266 Module.

gorghino, 2016-10-22:
@Pedrotok I succeded in compiling esp-open-sdk w/ SDK 2.0.0. I was facing "Random characters" problem with the blue esp led blinking very fast. I discovered that if It blinks like that, the esp is in "Waiting for firmware" mode.
So I reflashed my compiled openSDK with:
esptool.py --port /dev/ttyUSB0 write_flash 0xfc000 ../../esp_init_data_default.bin 0x7e000 ../../blank.bin 0xfe000 ../../blank.bin 0x0 ../../boot_v1.6.bin 0x01000 user1.1024.new.2.bin 

And I succeded in compiling a Hello world user_main.c.

I read about "Putting user_rf_cal_sector_set() in user_main.c" but If I do it, I can't compile my main because this function is already set in libmain.a

hashin, 2016-12-02:
@igrr any updates on the status of TLS 1.2? Thanks in advance!

igrr, 2016-12-02:
axTLS with 1.2 is in master I believe. How is it related to this issue?

vladislav-e-martin, 2016-12-03:
@igrr Currently, the most [up-to-date version in 2.3.0](https://github.com/esp8266/Arduino#available-versions) if you download the ESP8266 SDK using Arduino's Board Manager. So, I thought that I would be able to test the WPA2-Enterprise functions found at `#include "wpa2_enterprise.h"`. However, when I download the library using the Board Manager, I ran into the following error message: _fatal error: wpa2_enterprise.h: No such file or directory_.

Then, I followed the steps to download the SDK you presented [earlier in this thread](https://github.com/esp8266/Arduino/issues/2304#issuecomment-245849344). The error message went away (for some reason I can't get the `wifi_station_set_enterprise_username` function to actually set the username correctly...). Is there any reason that you must download version 2.0.0 of the NONOS SDK to `#include "wpa2_enterprise.h"` using `extern "C"` in an Arduino sketch? Especially since the version you recommend to install is 2.0.0 and the current stable one is 2.3.0.

igrr, 2016-12-04:
You are confusing SDK version (2.0.0) with Arduino core version (2.3.0). Arduino core version 2.3.0 doesn't include SDK 2.0.0. there is an open PR which contains an attempt of integrating SDK2.0.0 I to the Arduino core.

YvesCla, 2017-01-17:
@timostr Did you find a solution for sampling more then 255 samples? I have the same problem.

torntrousers, 2017-02-02:
The 2.0 SDK has been out for over 6 months now, would be wonderful if the ESP8266/Arduino code could use it :)

osos, 2017-02-21:
Seems something is happening: https://github.com/esp8266/Arduino/commit/ae13809c8184300aab9e3f09ef23af23d936b7ee

https://github.com/esp8266/Arduino/pull/2440

@igrr, when is expected next release including sdk 2.x ?

igrr, 2017-02-22:
Currently I'm waiting for feedback on nightly builds (#2936). They have SDK 2.0, fwiw.

andig, 2017-03-19:
It has meanwhile been merged into master.

dapug, 2017-03-26:
When will the Board Manager update to a new version?  I assume v2.3.0 (current as of this post) is NOT the one that supports ESP SDK v2.0.0.

denmorru, 2017-03-30:
There is no function like `system_get_sdk_version()` to return "Arduino IDE ESP8266 package" version,
but it is useful checking

devyte, 2017-10-14:
@igrr this is for SDK 2.0.0, and has milestone 2.5.0. Aren't we moving straight to SDK 2.1.0?

igrr, 2017-10-14:
This can be closed, yes. Going straight to 2.1.0.

closed 2017-10-14 by igrr