esp8266/Arduino #2306: Error when trying Http OTA for first time

AdamMiltonBarker, 2016-07-20:
### Basic Infos

#### Hardware
Hardware:			ESP-1
Core Version:      	?2.1.0-rc2?

### Description

I have added the httpupdater into a previously functioning sketch and receive the error documented here. Exception decoder will not work, continunally says:

ERROR: neither MoistureSense.ino.elf or MoistureSense.cpp.elf were found!
Did you forget to compile the sketch?

I have a custom setup which may be the reason. Any functions not provided in the sketch below are not needed the sketch runs on many devices fine, the only issue is the recently added code for the update.

### Settings in IDE

Module:  			Generic ESP8266 Module
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		OTA
Reset Method:		CK

### Sketch

```cpp

#include <Arduino.h>

void setup() {
  
  Serial.begin(115200);
  delay(10);
  Serial.setDebugOutput(true);
  Serial.println();
  Serial.println("System Initiating...");
        
  client.setServer(mqtt_server,mqttPort);
  readSpiffs();
  
  if ( esid != "" ) {
    
      WiFi.begin(esid.c_str(), epass.c_str());
      
      if (checkWifiConnection()) {
        
        serverType = "WC";

        Serial.println("Successfully connected to WiFi");
      
        Serial.print("SSID: ");
        Serial.println(WiFi.SSID());
        IPAddress ip = WiFi.localIP();
        Serial.print("IP Address: ");
        Serial.println(ip);
        long rssi = WiFi.RSSI();
        Serial.print("Signal strength (RSSI): ");
        Serial.print(rssi);
        Serial.println(" dBm");

        t_httpUpdate_return ret = ESPhttpUpdate.update("https://eu.techbubbletechnologies.com/updates/updater","");

        switch(ret) {
            case HTTP_UPDATE_FAILED:
                Serial.println("HTTP_UPDATE_FAILD Error");
                break;

            case HTTP_UPDATE_NO_UPDATES:
                Serial.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
                Serial.println("HTTP_UPDATE_OK");
                break;
        }
        
        
        return;
        
      } 
  }
  
  serverType = "AP";
  server=new(ESP8266WebServer);
  initiateAccessPoint();

}

void loop() {

}
```

### Debug Messages
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000bf0e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff29a0 end: 3fff2c00 offset: 01a0

>>>stack>>>
3fff2b40:  3ffe9490 3fff46e8 3fff1b1c 4020781c  
3fff2b50:  3ffea0e8 00000000 000003e8 3fff1bd8  
3fff2b60:  3fff458c 3fff45cc 3ffe9948 4020e0c5  
3fff2b70:  3fff46dc 00000200 3fff46e8 3ffe96dc  
3fff2b80:  3fff45e4 3fff1bd8 3fff1b1c 4020e1c0  
3fff2b90:  402010ae 0000000a 3fff1b1c 3fff1bd8  
3fff2ba0:  3fffdad0 00000000 3fff1b1c 4020793b  
3fff2bb0:  feefeffe feefeffe feefeffe feefeffe  
3fff2bc0:  feefeffe feefeffe feefeffe feefeffe  
3fff2bd0:  feefeffe feefeffe feefeffe 3fff1bd8  
3fff2be0:  3fffdad0 00000000 3fff1bd0 4020f6ac  
3fff2bf0:  feefeffe feefeffe 3fff1be0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

igrr, 2016-07-20:
I see you are providing an empty string for the HTTPS fingerprint. This is probably not handled correctly. 

But even if it was handled, update over HTTPS (or any other HTTPS request, for that matter) will not work without a valid fingerprint.

AdamMiltonBarker, 2016-07-20:
Thanks for the reply. Is that the answer to the error ? Can you expand on how I provide a finger print please or point me to the docs that explain this.

igrr, 2016-07-20:
No, that's not an answer. That is just an observation. It may be crashing for some other reason, but without a decoded stack trace I can't tell.
Another observation, the URL listed in sketch doesn't return a valid update binary. Instead it returns a 404 error.
We don't have any docs for the ESP8266HTTPUpdate library yet, but you can open a website in your browser, click on the green lock icon in URL bar, click "view certificate" (exact text depends on the browser and OS), then scroll through the certificate info until you find "SHA1 fingerprint". This is the fingerprint which has to be used as the second request argument.
You can also get certificate fingerprint from the command line, see for example this answer: http://stackoverflow.com/a/5165073.

AdamMiltonBarker, 2016-07-20:
Ah thanks for pointing out the url it should be hitting iot.techbubbletechnologies not eu, and thanks for the info on the fingerprint. I will make these changes and come back to you.

AdamMiltonBarker, 2016-07-20:
Re the decoded stack trace did you see the error above ? Followed the instructions but some of my sketches are not located in the sketch folder, this one included.

AdamMiltonBarker, 2016-07-20:
Also there is a custom path to the compiled binary

AdamMiltonBarker, 2016-07-20:
OK I have added the signature I still get the following error:

`
Fatal exception 28(LoadProhibitedCause):
epc1=0x4000bf0e, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000

Exception (28):
epc1=0x4000bf0e epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff2ca0 end: 3fff2f00 offset: 01a0

>>>stack>>>
3fff2e40:  3ffe9490 3fff49e8 3fff1e1c 4020781c  
3fff2e50:  3ffea3e8 00000000 000003e8 3fff1ed8  
3fff2e60:  3fff488c 3fff48cc 3ffe9c48 4020e0c9  
3fff2e70:  3fff49dc 00000200 3fff49e8 3ffe96dc  
3fff2e80:  3fff48e4 3fff1ed8 3fff1e1c 4020e1c4  
3fff2e90:  402010ae 0000000a 3fff1e1c 3fff1ed8  
3fff2ea0:  3fffdad0 00000000 3fff1e1c 4020793f  
3fff2eb0:  feefeffe feefeffe feefeffe feefeffe  
3fff2ec0:  feefeffe feefeffe feefeffe feefeffe  
3fff2ed0:  feefeffe feefeffe feefeffe 3fff1ed8  
3fff2ee0:  3fffdad0 00000000 3fff1ed0 4020f6b0  
3fff2ef0:  feefeffe feefeffe 3fff1ee0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

`

Info on setup, due to the amount of sketches I have they are located in different folders, this sketch is not in the main sketches folder. 

The output files from a build are saved to a custom location.

I get the error:

ERROR: neither MoistureSense.ino.elf or MoistureSense.cpp.elf were found!
Did you forget to compile the sketch?

What are my options to solve this ?

me-no-dev, 2016-07-20:
switch the location back do you can debug the issue. when I have time, I'll look into making the exception decoder ask for the location if the elf is not found

AdamMiltonBarker, 2016-07-20:
Ok thanks is it just the compile location or does the sketch have to be placed in the default sketches folder (which is also in a different location to default)

me-no-dev, 2016-07-20:
compile location should be enough

AdamMiltonBarker, 2016-07-20:
perfect thank you doing it now.

AdamMiltonBarker, 2016-07-20:
Yep that worked perfectly, here is the output of the first stack trace.

`
Decoding 7 results
0x4020781c: get  at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObject.ipp line 24
:  (inlined by) as  at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\ArduinoJson\src/../include/ArduinoJson/JsonObjectSubscript.hpp line 54
:  (inlined by) as  at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp line 45
:  (inlined by) operator char const*  at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\ArduinoJson\src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp line 40
:  (inlined by) readSpiffs() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 218
0x4020e0c9: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020e1c4: Print::println() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x402010ae: delay at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_wiring.c line 53
0x4020793f: String::operator!=(char const*) const at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/WString.h line 190
:  (inlined by) setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 412
0x4020f6b0: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
`

me-no-dev, 2016-07-20:
points to ArduinoJson. Maybe something in ```readSpiffs()```?

AdamMiltonBarker, 2016-07-20:
Yes I just removed the update and found I am still getting the erro (deleted log above) will check this out and get back to you, thanks again for help so far.

AdamMiltonBarker, 2016-07-20:
OK sorted that out, then tested and is working without the includes for the updater and the code for update, once added I now get this:

`
Decoding 36 results
0x4022e11f: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x4022e2b8: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x4022e56a: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x4022f0c2: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1287
0x4022f31d: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1412
0x4022e184: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x402304c8: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 227
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 260
0x4022c5a6: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 342
0x4022c9a5: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 105
0x4023799b: pbuf_free at ?? line ?
0x402379a9: pbuf_free at ?? line ?
0x4020915f: ClientContext::_consume(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 169
0x4022c3dc: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1470
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1348
0x4022c740: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 148
0x4022b0f4: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 576
0x4022c512: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x402095b4: SSLContext::connect(ClientContext*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) WiFiClientSecure::_connectSSL() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 262
0x402089da: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020f504: esp_yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40209746: WiFiClientSecure::connect(IPAddress, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 246
0x40208959: WiFiClient::connect(char const*, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020975c: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 252
0x4020bbc5: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x4010020c: _umm_free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x4010068c: free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40207166: reconnect() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 240
0x4020d2bc: HTTPClient::~HTTPClient() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 58
0x402073c0: loop at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 452
0x40210a9c: Print::write(unsigned char const*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020df61: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020df61: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020e05c: Print::println() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020e0a8: Print::println(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x40207938: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 425
0x4020f550: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
`

AdamMiltonBarker, 2016-07-20:
Seems to be a conflict between PubSub and the updater is there a way to fix that? Both are required for this project.

AdamMiltonBarker, 2016-07-21:
Any ideas guys? This error is due to adding the  updater library and update code? When they aren't included everything is working.

igrr, 2016-07-21:
Last stack trace you have posted indicates that there is no more RAM available. HTTPS needs lots of memory, so make sure you don't have other big objects allocated at the time you run updater.

AdamMiltonBarker, 2016-07-21:
The only thing that has happened is the spiffs have been read for connection details for wifi and mqtt, server was set up, I moved the mqtt setup below the update but still same.

igrr, 2016-07-21:
SPIFFS takes a few kB once initialized. How much heap memory do you have prior to starting the updater?

AdamMiltonBarker, 2016-07-21:
Ahh actually, I now get 

HTTP_UPDATE_FAILD Error
Attempting connection to TechBubble IoT JumpWay...

Exception (29):
epc1=0x4000e1b2 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000004 depc=0x00000000

ctx: cont 
sp: 3fff27e0 end: 3fff2dc0 offset: 01a0

I expect the update fail, as the script isnt complete, I did not get that message before so that seems to trying to get the update, but then this:

`
Decoding 37 results
0x4022e13b: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x4022e2d4: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x4022e586: regular_multiply at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 822
0x4022f0de: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1287
0x4022f339: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1412
0x4022e1a0: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x402304e4: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 227
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 260
0x4022c5c2: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 342
0x4022c9c1: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 105
0x402379b7: pbuf_free at ?? line ?
0x402379c5: pbuf_free at ?? line ?
0x4020917b: ClientContext::_consume(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 169
0x4022c3f8: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1470
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1348
0x4022c75c: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 148
0x4022b110: ssl_new at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 576
0x4022c52e: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 262
0x402095d0: SSLContext::connect(ClientContext*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 497
:  (inlined by) WiFiClientSecure::_connectSSL() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 262
0x402089f6: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x4020f520: esp_yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40209762: WiFiClientSecure::connect(IPAddress, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 246
0x40208975: WiFiClient::connect(char const*, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x40209778: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 252
0x4020bbe1: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x4010020c: _umm_free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1285
0x4010068c: free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266\umm_malloc/umm_malloc.c line 1731
0x40207166: reconnect() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 239
0x4020f37c: operator delete(void*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/abi.cpp line 84
0x4020d2d8: HTTPClient::~HTTPClient() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 58
0x402073cc: loop at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 453
0x40210ab8: Print::write(unsigned char const*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020df7d: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020df7d: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020e078: Print::println() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x4020e0c4: Print::println(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/Print.cpp line 76
0x40207958: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 427
0x4020f56c: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/core_esp8266_main.cpp line 43
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.2.0\cores\esp8266/cont.S line 109
`

So if this is still ram is there a way to terminate the update and free up ram?

mtnbrit, 2016-07-21:
I thought that FAILD typo was fixed several core versions ago, are you sure you have a clean core install and no left-overs from previous versions? 

possible you are using older code, much has been fixed and improved since the days of FAILD….

AdamMiltonBarker, 2016-07-21:
In preferences I have the following:

http://arduino.esp8266.com/stable/package_esp8266com_index.json

and no libraries are marked for updates in Arduino IDE

AdamMiltonBarker, 2016-07-21:
Actually sorry yes updating boards now will come back.

AdamMiltonBarker, 2016-07-21:
After update to 2.3.0 I still get 

HTTP_UPDATE_FAILD Error

AdamMiltonBarker, 2016-07-21:
Issue still remains:

`
Decoding 34 results
0x4020fb14: esp_yield at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x4022ea97: more_comps at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1072
0x4022ec30: alloc at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1106
0x4022f02d: bi_clone at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 263
0x4022fa25: bi_barrett at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1285
0x4022fc99: bi_mod_power at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1414
0x4022eafc: trim at /Users/igrokhotkov/e/axtls/e1/crypto/bigint.c line 1197
0x40230d25: RSA_public at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 242
:  (inlined by) RSA_encrypt at /Users/igrokhotkov/e/axtls/e1/crypto/rsa.c line 276
0x4022ce30: send_client_key_xchg at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 352
0x4022d231: do_clnt_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 109
0x4020925b: ClientContext::_consume(unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) ClientContext::read(char*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\include/ClientContext.h line 175
0x4022cc28: do_handshake at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1481
:  (inlined by) basic_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 1357
0x4022cfd0: do_client_connect at /Users/igrokhotkov/e/axtls/e1/ssl/tls1_clnt.c line 154
0x401004d8: malloc at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1664
0x4022cd80: ssl_read at /Users/igrokhotkov/e/axtls/e1/ssl/tls1.c line 265
0x402090dc: WiFiClient::_s_connected(void*, void*, signed char) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x402096cd: SSLContext::connect(ClientContext*, char const*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 517
:  (inlined by) WiFiClientSecure::_connectSSL(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 279
0x40208b06: WiFiClient::connect(IPAddress, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClient.cpp line 149
0x402098dd: WiFiClientSecure::connect(char const*, unsigned short) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src/WiFiClientSecure.cpp line 269
0x4020bd8d: PubSubClient::connect(char const*, char const*, char const*, char const*, unsigned char, unsigned char, char const*) at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\Arduino\_Sketches\libraries\PubSubClient\src/PubSubClient.cpp line 586
0x4010020c: _umm_free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1287
0x4010068c: free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x402071ce: reconnect() at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 239
0x4010068c: free at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266\umm_malloc/umm_malloc.c line 1733
0x4020d704: HTTPClient::~HTTPClient() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 51
0x40207434: loop at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 453
0x402110c0: Print::write(unsigned char const*, unsigned int) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x4020e351: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x4020e351: Print::write(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x4020e44c: Print::println() at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x4020e498: Print::println(char const*) at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/Print.cpp line 76
0x402079c0: setup at C:\Users\Usuario\Transporter\TECHNOLOGIES-IoT\_TechBubbleProductMasters\Devices\MoistureSense/MoistureSense.ino line 427
0x4020fb60: loop_wrapper at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/core_esp8266_main.cpp line 56
0x40100718: cont_norm at C:\Users\Usuario\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\cores\esp8266/cont.S line 109
`

AdamMiltonBarker, 2016-07-21:
This is the full code which I will delete soon, just wondering if anyone can add any advice.

`/*
 Copyright (c) 2016 TechBubble Technologies and other Contributors.
 
 The MIT License (MIT)
 Permission is hereby granted, free of charge, to any person obtaining a copy
 of this software and associated documentation files (the "Software"), to deal
 in the Software without restriction, including without limitation the rights
 to use, copy, modify, merge, publish, distribute, sublicense, and/or sell
 copies of the Software, and to permit persons to whom the Software is
 furnished to do so, subject to the following conditions:
 The above copyright notice and this permission notice shall be included in all
 copies or substantial portions of the Software.
 THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
 IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
 FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
 AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
 LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
 OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
 SOFTWARE.
 
 Contributors:
    Adam Milton-Barker  - Initial Contribution
*/

#include <FS.h> 
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#include <PubSubClient.h>
#include <WiFiClientSecure.h> 
#include <ArduinoJson.h>
#include <ESP8266mDNS.h>

#include <ESP8266HTTPClient.h>
#include <ESP8266httpUpdate.h>

ESP8266WebServer *server;

WiFiClientSecure espClient;
PubSubClient client(espClient);

const char* ssid = "TechBubbleAP";
const char* passphrase = "thereisnospoon";
String st;
String content;
String serverType ="";

String esid = "";
String epass = "";
  
String locationID = "";
String zoneID = "";
String deviceID = "";

char charBuf[50];
char deviceName[100]; 
char mqttUsername[100]; 
char mqttPassword[100]; 
char willTopic[50];

const char* mqtt_server = "iot.techbubbletechnologies.com";
int  mqttPort = 8883;
 
int plantPin = 2;

bool checkWifiConnection(void) {
  int connectAttempt = 0;
  //Serial.println("Waiting for Wifi to connect...");  
  while ( connectAttempt < 20 ) {
[... 363 lines cut ...]

void loop() {
  if(serverType=="AP"){
    server->handleClient();
  }else{ 
    
    if (!client.connected()) {
      reconnect();
    }
    
    client.loop();

    if(readPlantSense()){
      char json[] = "{\"Sensor\":\"PlantSense\",\"SensorID\":9,\"SensorValue\": \"DRY\"}";
      publishToDeviceSensors(json);  
      char jsonW[] = "{\"WarningType\":\"MOISTURE\",\"WarningOrigin\":\"PlantSense\",\"WarningValue\": \"DRY\",\"WarningMessage\": \"Plant Needs Watering\"}";
      publishToDeviceWarnings(jsonW); 
    }    
    
    delay(1000);               
    Serial.println(readPlantSense());

  }
}
`

AdamMiltonBarker, 2016-07-21:
@igrr sorry for silly question but what would be the way to see how much memory there is

mtnbrit, 2016-07-21:
> st = ""; for (int i = 0; i < networks; ++i){ st += "<option value='"; st += WiFi.SSID(i); st += "'>"; st += WiFi.SSID(i); st += " "; st += (WiFi.encryptionType(i) == ENC_TYPE_NONE)?" ":"*"; st += "</option>"; } st += "";
> std::unique_ptr buf(new char[size]);
> server->on("/",  <applewebdata://86360EE5-3F98-4185-A7AE-3B34F5E27B1B>{

You must still have some stale versions of old libs in your build if you are still getting FAILD. Not saying this is the cause of your crash, but its a red flag that you are running an older code.

search your machine for the file ESP8266HTTPUpdate.cpp. grep the file for FAILD, you should not find it.

AdamMiltonBarker, 2016-07-21:
Check my answer and code above, the response is hardcoded ;)

AdamMiltonBarker, 2016-07-21:
It is hardcoded see above

AdamMiltonBarker, 2016-07-21:
Any ideas on how to amend this on the code above guys ?

mtnbrit, 2016-07-21:
not sure what you mean by “hardcoded”. In my experience its easy to end up with multiple conflicting versions of the IDE, libraries and core. Your best bet might be to purge your system of any cores, boards, libraries, arduino IDE installs, and start over with a fresh latest IDE, then just add 2.3.0 board and your code and go from there.

AdamMiltonBarker, 2016-07-21:
The response that is being printed to serial is hardcoded, the spelling mistake is in the sketch not the library.

```
        t_httpUpdate_return ret = ESPhttpUpdate.update("https://iot.techbubbletechnologies.com/updates/updater","29 7C 82 C9 05 E2 A5 7F 03 2D A3 74 62 E7 7A 3C 24 F6 9F E7 58 87 61 84 60 A8 BF 1F 4D E6 6D 84 78 1D 22 4C D2 B8 A4 F2 EB 66 D5 02 45 F8 7B 92 75 77 B1 18 12 A2 22 3B C8 EF A8 41 BC 73 B2 60 BF 30 A2 78 DE CE 83 8A 17 01 52 9A F0 19 F1 47 CB CF 20 18 0B 01 48 B6 2D F5 49 EA B6 A8 15 C0 E5 A8 9A B1 5F D6 4A 39 EF 70 68 1D F6 DA 15 AD 22 81 5F BA 9E AF FA 86 5C 28 19 D0 B1 00 48 66 A1 11 24 2E 32 83 5F 2F 79 0C D6 D9 75 78 03 70 2E FA 24 5A 8A 7B 82 AB 9C 8D 13 DF C5 EA 33 37 0B E8 2C B0 E4 90 EF 46 90 C5 CD EF FD 79 EA 25 6A 1B D9 2D 27 CE 17 1F 6A B6 F5 EE 11 34 D5 3B CA A8 A9 10 26 BE 5A EC B6 DA 4C C1 53 CF DE 81 F6 76 0B C8 1D 54 EE 41 C0 7F DB 0E D4 B1 B8 5C 57 25 31 3E 9D 81 80 B3 3C 9B 8E 9B 28 58 B1 BC 32 AD A8 B0 25 4D 92 36 AB 8A 18 57 5E 47 4A 85");

        switch(ret) {
             case HTTP_UPDATE_FAILED:
               Serial.println("HTTP_UPDATE_FAILD Error");
               break;

            case HTTP_UPDATE_NO_UPDATES:
                Serial.println("HTTP_UPDATE_NO_UPDATES");
                break;

            case HTTP_UPDATE_OK:
               Serial.println("HTTP_UPDATE_OK");
               break;
        }```

AdamMiltonBarker, 2016-07-21:
Any ideas on stopping the ram from running out on the above sketch for the updater to work? If that is the issue with the crash.

AdamMiltonBarker, 2016-07-22:
I went through another sketch as got annoyed with this one, I removed all of the unnecessary serial prints and replaced any needed ones with Serial.println(F()), cleaned up everything and checked all worked fine, as soon as I placed the update code in it started crashing. This is on an ESP01 if it makes and difference, would really like to get to the bottom of this if possible.

retsifp, 2016-07-25:
I'm not sure this is the problem, but you didn't provide the SHA1 fingerprint but the certificate signature...

So maybe setting the fingerprint to the correct "E1 D2 56 D8 9B 14 F2 8E 3D 36 D9 35 BC B3 4F 19 45 C1 97 98" will fix it...

AdamMiltonBarker, 2016-07-25:
Ok thanks will check that now.

AdamMiltonBarker, 2016-08-05:
I have moved to an ESP 12f and no longer get the error but do get connection refused, any ideas on what may be the issue

AdamMiltonBarker, 2016-08-05:
It is now making the sensor readings as NAN

AdamMiltonBarker, 2016-08-06:
I give up with this after light sleep it crashes. Clearly still issues with this feature.

AdamMiltonBarker, 2016-08-06:
Hi guys. I have moved to an ESP 12f and no longer get the error but do get connection refused, any ideas on what may be the issue.

I have removed the htaccess so it goes directly to the .php page.
I have updated the code to use the .php location.
I have changed the fingerprint to the SHA1 fingerprint.
I have added the default NOT UPDATED header to test.

It is still providing the error:

HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused

AdamMiltonBarker, 2016-08-06:
Anyone ?

AdamMiltonBarker, 2016-08-07:
@igrr any particular reasonthis is now being ignored ?

igrr, 2016-08-07:
Hi, I'll take a look again once I'm in the office on Monday.
I see you haven't addressed my last comment about memory usage. HTTPS will fail to connect if you don't have enough heap memory. Please check your heap usage and log the value of ESP.getFreeHeap() before making the call to http update. This method is mentioned in reference docs...

AdamMiltonBarker, 2016-08-07:
Thanks I replied asking how to do it but now if you check the issue I linked I can now do nothing anyway. After erasing the flash, changing to 4m as the script suggested and reuploading / formatting spiffs ut constantly crashes in a 12f.

igrr, 2016-08-07:
It happens if wifi config gets corrupted. Can you try erasing the whole flash using esptool.py?

AdamMiltonBarker, 2016-08-07:
I already did that.

AdamMiltonBarker, 2016-08-07:
Doing again now to try and am trying 4m /1m instead of 3

AdamMiltonBarker, 2016-08-07:
Same again. it didn't crash this time on starting the AP but then when I enter the wifi details etc into the form and press submit it crashes and this is the stack:

```
Decoding 57 results
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
0x40224d7c: user_uart_wait_tx_fifo_empty at ?? line ?
0x40105347: wdt_feed at ?? line ?
0x40105328: wdt_feed at ?? line ?
0x40105341: wdt_feed at ?? line ?
0x4010533e: wdt_feed at ?? line ?
``

AdamMiltonBarker, 2016-08-07:
And again same issue.

igrr, 2016-08-08:
Running the sketch you have posted in the linked issue, i got the following output. It appears that connection fails during TLS handshake. 

```
connected with #!/bin/test, channel 11
dhcp client start...
.ip:192.168.77.241,mask:255.255.255.0,gw:192.168.77.1
.Successfully connected to WiFi
[HTTP-Client][begin] url: https://iot.techbubbletechnologies.com/updates/updater.php
[HTTP-Client][begin] host: iot.techbubbletechnologies.com port: 443 url: /updates/updater.php
[HTTP-Client][begin] httpsFingerprint: E1 D2 56 D8 9B 14 F2 8E 3D 36 D9 35 BC B3 4F 19 45 C1 97 98
please start sntp first !
State:	sending Client Hello (1)
Alert: handshake failure
Error: SSL error 40
Alert: unexpected message
Alert: close notify
[HTTP-Client] failed connect to iot.techbubbletechnologies.com:443
[HTTP-Client][returnError] error(-1): connection refused
[HTTP-Client][end] tcp is closed
HTTP_UPDATE_FAILD Error (-1): HTTP error: connection refused
```

Wireshark is useful tool to debug such issues. Running wireshark and filtering by protocol (ssl) and device IP address, we see what's going on:
```
1236	3.243890	192.168.77.241	37.48.75.198	TLSv1.1	210	Client Hello
1316	3.460705	37.48.75.198	192.168.77.241	TLSv1.1	149	Alert (Level: Fatal, Description: Handshake Failure)
```

First packet decoded:
```
Secure Sockets Layer
    TLSv1.1 Record Layer: Handshake Protocol: Client Hello
        Content Type: Handshake (22)
        Version: TLS 1.1 (0x0302)
        Length: 88
        Handshake Protocol: Client Hello
            Handshake Type: Client Hello (1)
            Length: 84
            Version: TLS 1.1 (0x0302)
            Random
            Session ID Length: 0
            Cipher Suites Length: 4
            Cipher Suites (2 suites)
                Cipher Suite: TLS_RSA_WITH_AES_128_CBC_SHA (0x002f)
                Cipher Suite: TLS_RSA_WITH_AES_256_CBC_SHA (0x0035)
            Compression Methods Length: 1
            Compression Methods (1 method)
                Compression Method: null (0)
            Extensions Length: 39
            Extension: server_name
                Type: server_name (0x0000)
                Length: 35
                Server Name Indication extension
                    Server Name list length: 33
                    Server Name Type: host_name (0)
                    Server Name length: 30
                    Server Name: iot.techbubbletechnologies.com
```
To which the server then replies with a handshake failure message.

Let's check the cipher suites supported by your server.
https://www.ssllabs.com/ssltest/analyze.html?d=iot.techbubbletechnologies.com
(scroll down to cipher suites).

As you see from the list, two cipher suites supported by axTLS library on the ESP side (TLS_RSA_WITH_AES_128_CBC_SHA and TLS_RSA_WITH_AES_256_CBC_SHA) are not among the list of cipher suites supported by your server. This causes handshake failure, because if the server and client have no cipher suites in common, they can't talk to each other.

You may want to check your server configuration to enable one of these suites. Alternatively, if you can not or do not want to change TLS config for the whole server, you may set up a TLS proxy which will handle talking to ESPs only. The rest of your site will run with same TLS settings, but only the endpoints needed by ESP will be passed through a proxy. In this case the proxy should be configured to support these two mentioned cipher suites.

pieman64, 2016-08-08:
I don't want to hijack the thread but I have been following it for several reasons:

1. Some of my ESP's have corrupt flash so I tried erase_flash with esptool.py but it fails with **timed out waiting for packet header** (are my ESP's dead)?

2. I struggle to connect to one of my servers that has https. At first I was overlooking the fingerprint. I like the ssl labs reports and it turns out I have the same certificate issuer as @AdamMiltonBarker.

However my rating is F rather than A+ because of this https://blog.cloudflare.com/yet-another-padding-oracle-in-openssl-cbc-ciphersuites/ (I will try to fix this).

Most importantly I will check the cipher suites as advised by @igrr.

pieman64, 2016-08-08:
@igrr can you just confirm that the axTLS is the only library available for the ESP that offers "security" and therefore we must enable TLS_RSA_WITH_AES_128_CBC_SHA and TLS_RSA_WITH_AES_256_CBC_SHA cipher suites on our domains?

igrr, 2016-08-08:
It's not the only library. mbedTLS is also supported in the non-OS SDK. It has support for more cipher suites and it can also run TLS 1.2. However it has not been integrated into Arduino yet.

For axTLS, the server needs to talk TLS_RSA_WITH_AES_128_CBC_SHA or TLS_RSA_WITH_AES_256_CBC_SHA. As i have mentioned above, you don't have to lower security rating of the whole server to support ESPs. Having a TLS proxy for the endpoints used by ESPs is also an option. For this proxy you can even use self-signed certificates because at the moment we validate certificates by fingerprint, so having a certificate chain up to one of root certificates doesn't really matter.

AdamMiltonBarker, 2016-08-08:
If I allow those cypher suites it will make the server vulnerable, it is not the server that needs fixing it is the library. Notice the A+ ? That is higher than Google and Facebook and has been for many years.

AdamMiltonBarker, 2016-08-08:
@pieman64 the rating on my server is because I spend a lot of time researching vulnerabilities and I ensure that my servers and products are secure for my clients ;)

igrr, 2016-08-08:
I'd love to hear why you say that these cipher suites are insecure. Particular implementations of CBC are insecure (e.g. OpenSSL). Recent versions of OpenSSL don't have this vulnerability, so there is nothing to stop one from using those without affecting security of the clients. As I have mentioned above, you may set up a dedicated proxy (e.g. HAproxy) to be used by the ESPs so that these cipher suites do not affect the rating of your primary domain.

Note that I don't argue the fact that we should keep using these cipher suites :). In fact there is an open issue about updating to the SDK 2.0 and mbedTLS, which supports a lot more cipher suites. mbedTLS brings some other issues, for instance you will have to support fragment length negotiation extension on your server. If you can enable fragment length negotiation, then mbedTLS may be a good choice for your application. It is already supported on the ESP8266, just not in the Arduino environment. Please check Espressif IOT SDK and esp-open-rtos if you want to get mbedTLS on ESP8266 without waiting for it to be integrated into the Arduino environment.

AdamMiltonBarker, 2016-08-08:
@igrr I will update the ciphers and see how it affects the rating as far as I was aware I am currently supporting the only safe combination of cipher suites, if I am wrong then that is great news. I did reply before I read your message, but using self signed is obviously an insecure option, it defies all the purpose of using SSL as you never know if someone is playing MITM, I will check out your details as well you posted but now the immediate issue is the issue with crashing when using 4m.

igrr, 2016-08-08:
Self-signed or signed by a certificate authority — this does not matter with the ESP because validation is performed not by checking certificate chain, but by comparing certificate hash with the known value. MITM is not an issue because you have essentially pinned the correct certificate on the client.

Regarding server score, it will certainly go down. But as I have mentioned above, server score doesn't equate to server security. Some cipher suites have been marked as insecure because some widespread implementations are insecure. This doesn't prevent you from updating OpenSSL to a version which is not vulnerable and using these ciphers.

[3 more replies]

closed 2016-08-09 by AdamMiltonBarker