esp8266/Arduino #2308: UDP broadcast problem

gl3nni3, 2016-07-21:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:	esp8266(esp-01)
Core Version:      	2.1.0-rc2

### Description

Hello guys i'm trying to use a udp broadcast but the message is never sent.

### Settings in IDE

Module:  			Generic ESP8266 Module
Flash Size: 		512k
CPU Frequency:		80Mhz
Flash Mode:			DIO
Flash Frequency:	40Mhz
Upload Using:		SERIAL
Reset Method:		ck

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

char ssid[] = "ssid";  //  your network SSID (name)
char pass[] = "password";       // your network password

unsigned int localPort = 12345;      // local port to listen for UDP packets

/* Don't hardwire the IP address or we won't get the benefits of the pool.
 *  Lookup the IP address for the host name instead */
IPAddress ntpServerName(255,255,255,255);
// A UDP instance to let us send and receive packets over UDP
WiFiUDP udp;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, pass);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println(ntpServerName);
  Serial.println("Starting UDP");
  udp.begin(localPort);
  Serial.print("Local port: ");
  Serial.println(udp.localPort());
}

void loop()
{
  sendNTPpacket(ntpServerName); 
  while(udp.available()!=0)Serial.write((char)udp.read());
  delay(1000);
}

unsigned long sendNTPpacket(IPAddress& address)
{
  Serial.println("sending NTP packet...");
  char bericht[] = "Please send priority?";
  Serial.write("broadcast ip is:");
  Serial.println(address);
  udp.beginMulticast(address,WiFi.localIP(), 12345); 
  udp.write(bericht);
  udp.endPacket();
}

### Debug Messages

wifi evt: 7 is the only thing outputted in debug mode

me-no-dev, 2016-07-21:
first you have reversed the multicast address and interface address, then multicast is not broadcast ;)

```cpp
udp.beginPacket(IPAddress(255,255,255,255), 12345);
udp.write(bericht);
udp.endPacket();
```

gl3nni3, 2016-07-21:
thank you for the response :D.
I already fixed my code by a example i found online: http://www.iotfrog.com/en/articles/article/227

for anyone else having this problem this is my code now:

/*

 Udp NTP Client

 Get the time from a Network Time Protocol (NTP) time server
 Demonstrates use of UDP sendPacket and ReceivePacket
 For more on NTP time servers and the messages needed to communicate with them,
 see http://en.wikipedia.org/wiki/Network_Time_Protocol

 created 4 Sep 2010
 by Michael Margolis
 modified 9 Apr 2012
 by Tom Igoe
 updated for the ESP8266 12 Apr 2015 
 by Ivan Grokhotkov

 This code is in the public domain.

 */

#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

char ssid[] = "ssid";  //  your network SSID (name)
char pass[] = "password";       // your network password

unsigned int localPort = 12345;      // local port to listen for UDP packets

/* Don't hardwire the IP address or we won't get the benefits of the pool.
 *  Lookup the IP address for the host name instead */
IPAddress ntpServerName(192,168,27,255);
// A UDP instance to let us send and receive packets over UDP
WiFiUDP udp;

void setup()
{
  Serial.begin(115200);
  Serial.println();
  Serial.println();

  // We start by connecting to a WiFi network
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, pass);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  Serial.println(ntpServerName);
  Serial.println("Starting UDP");
  udp.begin(localPort);
  Serial.print("Local port: ");
  Serial.println(udp.localPort());
}

void loop()
{
  sendNTPpacket(ntpServerName); 
  delay(3000);
}

unsigned long sendNTPpacket(IPAddress& address)
{
  Serial.println("sending NTP packet...");
  Serial.write("broadcast ip is:");
  Serial.println(address);
  byte udpBuffer[50];
  byte message[] = "Please send priority?";
  udp.beginPacket(address, localPort);
  udp.write(message, sizeof(message));
  udp.endPacket();
}

closed 2016-07-21 by gl3nni3

pieman64, 2016-07-21:
@gl3nni3 what are the 2 library files you have in the #include's?
Maybe edit your post, highlight the sketch and press <> so it is in a readable format.

gl3nni3, 2016-07-21:
@pieman64 I updated my code thanks for the advice @pieman64

wyzarddoc, 2016-07-21:
@gl3nni3 
code seems to compile when I copied from above but in email update the previous code was sent without the includes??? a github issue?? thanks for the work
Doc