esp8266/Arduino #2315: WiFiUDP not receiving data anymore

sandrolutz, 2016-07-22:
#### Hardware
Hardware:			ESP-12E
Core Version:      	        2.3

### Description

I updated the Arduino core from 2.2 to 2.3. My code worked very well with release 2.2. Every packet has been received correctly. Unfortunately with release 2.3, the same code does not receive any udp packets anymore. There are no exceptions thrown. 

### Settings in IDE

Module:  			  NodeMCU 1.0
Flash Size: 		  4MB/1MB
CPU Frequency:	  80Mhz
Upload Using:		  SERIAL
Reset Method:		  nodemcu

### Sketch

```cpp

#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <WiFiUdp.h>

WiFiUDP udp;

void setup() {
    Serial.begin(115200);
    WiFi.mode(WIFI_AP_STA);
    IPAddress ip(12, 0, 0, 1); // Gateway IPAddress is 12.0.0.1
    IPAddress subnetmask(255, 255, 255, 0);
    WiFi.softAPConfig(ip, ip, subnetmask);
    WiFi.softAP("ESP8266-Example", "ThisIsNotSave");
    
    udp.begin(1032);
}

void loop() {
    // Read udp data
    int packetSize = udp.parsePacket();
    if(packetSize > 0) {
        Serial.println("Received some data!");
        if(packetSize > 6 && packetSize-2 == (udp.read() << 8) | udp.read()) {
            // read packet
            int msgLength = packetSize-7;
            uint32_t packetID = (udp.read()<<24) | (udp.read()<<16) | (udp.read()<<8) | udp.read();
            Serial.println("New message received!");
        }
    }
    udp->flush();
}
```

wyzarddoc, 2016-07-22:
just tried to run your code above and got the same problem with 2.3.0 something has changed

wyzarddoc, 2016-07-22:
@sandrolutz 
Just tried to run your code above in arduino ide 1.6.4 with the following results
hope this hellps

Using library ESP8266WiFi in folder: /home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi 
Using library WiFi101 in folder: /home/doc/Arduino/libraries/WiFi101 

/home/doc/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/include -I/home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lwip/include -I/tmp/build6450578341156264690.tmp/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DARDUINO=10604 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ESP8266 -DARDUINO_BOARD="ESP8266_NODEMCU" -DESP8266 -I/home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/cores/esp8266 -I/home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/variants/nodemcu -I/home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi/src -I/home/doc/Arduino/libraries/WiFi101/src /tmp/build6450578341156264690.tmp/udp-test.cpp -o /tmp/build6450578341156264690.tmp/udp-test.cpp.o 
udp-test.ino: In function 'void loop()':
udp-test.ino:30:16: error: expected '}' at end of input
Multiple libraries were found for "WiFiUdp.h"
 Used: /home/doc/Arduino/libraries/WiFi101
 Not used: /home/doc/.arduino15/packages/esp8266/hardware/esp8266/2.3.0/libraries/ESP8266WiFi
Exception in thread "Thread-19" java.lang.ArrayIndexOutOfBoundsException: -447
	at javax.swing.text.Utilities.getTabbedTextWidth(Utilities.java:249)
	at javax.swing.text.Utilities.getTabbedTextWidth(Utilities.java:208)
	at processing.app.syntax.JEditTextArea._offsetToX(JEditTextArea.java:596)
	at processing.app.syntax.JEditTextArea.scrollTo(JEditTextArea.java:478)
	at processing.app.syntax.JEditTextArea.scrollToCaret(JEditTextArea.java:443)
	at processing.app.syntax.JEditTextArea.select(JEditTextArea.java:1237)
	at processing.app.Editor.statusError(Editor.java:2781)
	at processing.app.Editor$BuildHandler.run(Editor.java:2040)
	at java.lang.Thread.run(Thread.java:745)

me-no-dev, 2016-08-01:
I tested your exact sketch agains the GIT version of this repo (not against 2.3.0) and I found it to be working fine both in AP and STA modes. Belo is my test python script that I used
```python
import socket
import sys

try:
    s = socket.socket(socket.AF_INET, socket.SOCK_DGRAM)
except socket.error:
    print 'Failed to create socket'
    sys.exit()
 
host = '12.0.0.1';
port = 1032;
 
while(1) :
    msg = raw_input('Enter message to send : ')
    try :
        s.sendto(msg, (host, port))
     
    except socket.error, msg:
        print 'Error Code : ' + str(msg[0]) + ' Message ' + msg[1]
        sys.exit()
```

sandrolutz, 2016-08-18:
Thanks for your help.
I got it working, but I don't really know, what the exact problem was.

closed 2016-08-18 by sandrolutz