esp8266/Arduino #2316: Unable to install ESP8266 board using the board manager

siddhantsahay, 2016-07-22:
I am not able to install the ESP 8266 board using the board manager in Arduino IDE. The message I am getting is shown in the image attached with this message. Please help. 
[image]

igrr, 2016-08-01:
Does this same error happen every time you try to install? Can you copy the full error message text and upload it somewhere?

siddhantsahay, 2016-08-01:
I am having this problem on one particular laptop which is running windows 7. I tried installing the new Java platform and even after that the problem still persists.

siddhantsahay, 2016-08-01:
I ll try copy pasting it here. But its the same error message that is repeated given in the picture.

"Caused by...................................................23 more "

rickylee64, 2016-08-02:
I also can't install esp8266 using board manager. I'm using: 
Windows 7, 32-bit. 
Java 8/101. 
I'm using a proxy. Details entered into IDE.
I've turned off windows firewall. 
I'm running Arduino 1.6.10. 
The "Boards Manager" seems to get stuck at "Downloading platforms index..." sometimes, and I have to use task manager to close the IDE.
When it does finish, I get the error message below. 
Any help appreciated.

Skipping contributed index file C:\Users\c845010\AppData\Local\Arduino15\package_esp8266com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Illegal character ((CTRL-CHAR, code 0)): only regular white space (\r, \n, \t) is allowed between tokens
 at [Source: java.io.FileInputStream@fb0edb; line: 138, column: 10] (through reference chain: ContributionsIndex["packages"]->java.util.ArrayList[0]->ContributedPackage["platforms"]->java.util.ArrayList[1]->ContributedPlatform["boards"])
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\bootloaders. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\cores. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\doc. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\libraries. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\package. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\tests. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\tools. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\variants. Is it pre-1.5?
WARNING: Error loading hardware folder c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com
  No valid hardware definitions found in folder esp8266com.
Skipping contributed index file C:\Users\c845010\AppData\Local\Arduino15\package_esp8266com_index.json, parsing error occured:
com.fasterxml.jackson.databind.JsonMappingException: Illegal character ((CTRL-CHAR, code 0)): only regular white space (\r, \n, \t) is allowed between tokens
 at [Source: java.io.FileInputStream@4682e5; line: 138, column: 10] (through reference chain: ContributionsIndex["packages"]->java.util.ArrayList[0]->ContributedPackage["platforms"]->java.util.ArrayList[1]->ContributedPlatform["boards"])
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\bootloaders. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\cores. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\doc. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\libraries. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\package. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\tests. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\tools. Is it pre-1.5?
Could not find boards.txt in c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com\variants. Is it pre-1.5?
WARNING: Error loading hardware folder c:\users\c845010\ricky\2016\arduino2016\arduino\hardware\esp8266com
  No valid hardware definitions found in folder esp8266com.

igrr, 2016-08-02:
@rickylee64 
1. Why are you having esp8266 core in your arduino\hardware\esp8266com directory? This is not a valid way to install the core. You need to remove that. If you do want to place the core into IDE hardware directory, it has to go into arduino\hardware\esp8266com\esp8266.
2. IDE complains about invalid characters in C:\Users\c845010\AppData\Local\Arduino15\package_esp8266com_index.json. Did you try opening this file with text editor to check if it was downloaded correctly?

devyte, 2017-10-14:
Per last comment, closing as user error. If you still need assistance, please refer to a community forum. For more details see #3655 .

closed 2017-10-14 by devyte