esp8266/Arduino #2318: When i boot ESP8266 i'm getting on my arduino MEGA serial monitor.

aditgupta100, 2016-07-23:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

igrr, 2016-08-01:
It appears that something is wrong with your serial port connection. Exception line has all the characters messed up. I can imagine this may also affect upload process, so the code in flash may have been uploaded incorrectly.

aditgupta100, 2017-01-08:
No, all connections are fine. I've tried with AT firmware and the same module was working perfectly fine. Now when i flashed NodeMCU firmware 0.9.x version nothing works.
It is giving Fatal exception 0(IllegalInstructionCause).

In the Esplorer IDE i'm getting 
Can't autodetect firmware , because proper answer not received (may be unknown firmware).

When i pull CH_PD high-low-high at this point then fatal exception comes.

 I'm not sure about the locations of all flash address and flash size but i hope .Bin firmware isn't corrupted (downloaded from github).

I have gone through instructions here : nodemcu.readthedocs.io/en/latest/en/#getting-started

taeraeyttaejae, 2017-02-22:
any update on this issue ?

i have the same issue with one nodemcu 1.0 using arduino ide 1.8.1.

using anything causes the fatal exeption, and it prints out with 74880 baud rate.

devyte, 2017-10-14:
@aditgupta100 said:

> nodemcu_integer_0.9.5_20150318.bin and NodeMCU 0.9.5 build 20150318 powered by Lua 5.1.4

That is the Lua firmware. What does that have to do with this firmware?

If you were trying to flash a firmware based on this repo over the Lua one, I suggest erasing the entire flash first (i.e.: with esptool), and only then flashing this one.

Not the right place to ask for assistance, please refer to a forum. Closing per #3655 .

closed 2017-10-14 by devyte