esp8266/Arduino #2320: CLI example to generate SPIFFS binary file?

gordonthree, 2016-07-23:
I'm hoping someone can share an example of how to generate the SPIFFS binary blob manually, so that I can use it with the httpupdate library? I've tried to "catch" the binary file the arduino IDE generates, but it only lasts a second before the ide deletes it.

#### Hardware
Hardware:			ESP-12F, ESP-07, ESP-01 (1mbyte version)
Core Version:      	2.3.0

furyfire, 2016-07-27:
PlatformIO builds and uploads the SPIFFS image if you supply it with 
`platformio run -t uploadfs`

igrr, 2016-08-01:
The tool used to build SPIFFS images is [mkspiffs](https://github.com/igrr/mkspiffs). You need to pass it correct options for block size, page size, and FS size. These values vary depending on flash size you have selected in Arduino IDE. You can take a look into boards.txt to find these values.
https://github.com/esp8266/Arduino/blob/master/boards.txt#L158-L163

For example, if you have 4M(1M SPIFFS) option selected in IDE, you need to run mkspiffs as follows:

```
mkspiffs -p 256 -b 8192 -s $((0x3FB000 - 0x300000)) -c path/to/files/dir/ spiffs-image.bin
```

Keep in mind that if you upload an incompatible SPIFFS image (built with different geometry options), SPIFFS.begin() will erase it and create a new image with correct options.

closed 2016-08-01 by igrr

kapyaar, 2017-05-01:
Hi, I am trying to generate the spiffs image bin file, and I have a 4M(1M SPIFFS) setting. I downloaded the latest release, and when I try 
`
mkspiffs -p 256 -b 8192 -s $((0x3FB000 - 0x300000)) -c path/to/files/dir/ spiffs-image.bin`

I get the following error. 

`PARSE ERROR: Arguiment: -s(--size) Could n't read argument value from string '$((0x3FB000'`

Is that above line supposed to be used AS IS, or was that a example that I need to modify? thanks for any help.

JamesGKent, 2017-05-02:
i think the idea is to tell mkspiffs the size of the bin to generate, and to calculate that you take the start address from the end address, the command listed above should do that at the time it's run if your shell supports it, if not you will need to calculate it (as you have done already)

mrfaptastic, 2018-06-19:
If anybody comes across this thread I suggest they read: https://github.com/esp8266/Arduino/issues/2625

Summary: It's easier to capture the SPIFFS file created by the Arduino IDE as I suggested.

Deepakkumar2102, 2023-02-13:
Hi , i am trying to build SPIFF file using mkspiffs   command
mkspiffs -c data -p 256 -b 8192 -s 0x100000   spiffs-image.bin.
but it is not working after upload .