esp8266/Arduino #2321: Auto size spiffs file system. Universal binary creation.

mmiscool, 2016-07-23:
I would really like to have an option to produce a generic binary that can be flashed to any device and auto size the spiffs file system to the max size.

Enhancement request.

igrr, 2016-07-28:
If the device already has smaller SPIFFS partition flashed (e.g. someone was using a 1M binary on 4M flash), would you like the partition to be preserved? Or is it okay to just create new partition which fits actual flash size?

mmiscool, 2016-07-30:
I would like it to max out the size of the file system automatically.

igrr, 2016-08-01:
I.e. it has to overwrite older small partition, correct?

mmiscool, 2016-08-01:
Yes. Wipe out the old smaller one. That would be great.

mmiscool, 2016-08-15:
Somewhat related. Don't know if it is possible. I have run in to trouble compiling for 1 megabyte modules. The file system takes up half of the module and the firmware the other half. Unfortunately my firmware is now over the allowable size limit. Would be nice if this allowed for the size of spiffs system to shrink either tightly using space unused or reserved by the firmware or to size to the next step down for rigid size of the spiffs.

mmiscool, 2016-10-20:
Any word on this becoming a possibility?

devyte, 2018-11-27:
From what I can tell, this requires some rather deep and fundamental changes to code, ld scripts, and IDE menu.
Removing milestone for now.

d-a-v, 2021-03-31:
This will be fixed by #6690.
Pushing to v3.0.1.

closed 2022-02-10 by d-a-v