esp8266/Arduino #2322: Support streaming audio

DimIsaev, 2016-07-24:
Is possible to implement support **streaming  adio**  in ESP
Example support  in SDK

> This is an example of how to use the I2S module inside the ESP8266 to output sound. In this case, it is used to output decoded MP3 data (actually, more accurately: MPEG2 layer III data): the code described here basically is a webradio streamer which can connect to an Icecast server, take the MP3 data the server sends out, decode it and output it over the I2S bus to a DAC. The MP3 decoder has been tested for bitrates up to 320KBit/s and sample rates of up to 48KHz.

https://github.com/espressif/esp8266_mp3_decoder

@igrr

me-no-dev, 2016-07-25:
https://github.com/esp8266/Arduino/blob/master/cores/esp8266/i2s.h
I2S is supported :)
Here is an example with ffmpeg streaming PCM audio to the ESP
https://gist.github.com/me-no-dev/e8b5131f840822cdd14b

earlephilhower, 2018-03-09:
I'm going to close this as it's pretty old, I2S is supported (at least output...I'm waiting on my Google AIY MEMS I2S microphone to add input), and if you *really* want streaming audio you can check out my ESP8266Audio library here on GitHub or other folks' libs.  Nothing in Arduino core to support.

closed 2018-03-09 by earlephilhower