esp8266/Arduino #2323: Support for >4MB flash

KK2345, 2016-07-24:
Not really issue, but enhancement.
ESP8266 can support up to 128Mb (16MB) flash memory size, but in Arduino IDE, you can use 4MB max. Some people get it working: 
http://www.instructables.com/id/ESP-03-Upgrade-Flash-Memory-to-128-M-Bit/?ALLSTEPS

I also found thi in boards.txt
`# disabled because espressif's bootloader refuses to write above 4M
# generic.menu.FlashSize.8M=8M (7M SPIFFS)
# generic.menu.FlashSize.8M.build.flash_size=1M
# generic.menu.FlashSize.8M.build.flash_ld=eagle.flash.8m.ld
# generic.menu.FlashSize.8M.build.spiffs_start=0x100000
# generic.menu.FlashSize.8M.build.spiffs_end=0x800000
# generic.menu.FlashSize.8M.build.spiffs_blocksize=8192
# generic.menu.FlashSize.16M=16M (15M SPIFFS)
# generic.menu.FlashSize.16M.build.flash_size=1M
# generic.menu.FlashSize.16M.build.flash_ld=eagle.flash.16m.ld
# generic.menu.FlashSize.16M.build.spiffs_start=0x100000
# generic.menu.FlashSize.16M.build.spiffs_end=0x1000000
# generic.menu.FlashSize.16M.build.spiffs_blocksize=8192
`

Are you planning to get it working or is it just impossible?

far5893, 2016-07-25:
Seems that bootloader(and esptools)  is limited to 4Mb, but i suppose if you format spiffs in place(or load it via OTA)  it can handle 15Mb of space.

me-no-dev, 2016-07-31:
here is some info
https://github.com/esp8266/Arduino/pull/2351
Needs updated esptool and to be tested :)

igrr, 2016-08-01:
I have merged everything, esptool will also be updated soon.

closed 2018-01-02 by igrr