esp8266/Arduino #2324: SPIFFS filename length > 30, causes unexpected behaviour

kiralikbeyin, 2016-07-24:
#1088

Problem still exists.

FS File: /fonts/glyphicons-halflings-regu, size: 106.19KB

orginal file name : glyphicons-halflings-regular.woff2

igrr, 2016-08-01:
Was this file created on the ESP or was it uploaded using ESP8266 Sketch Data Upload tool?

kiralikbeyin, 2016-08-01:
Sketch Data Upload tool. Not so important because i will use this https://github.com/me-no-dev/ESPAsyncWebServer i can rewrite file names in sketch and really fast fast fast:)

Thanks Ivan.

devyte, 2017-10-14:
@igrr do you know if the issue is still valid in the sketch data upload tool?

igrr, 2017-10-14:
Could be, but likely not in the upload tool — either in mkspiffs or the
code here.

kiralikbeyin, 2017-10-14:
Yesterday i couldnt upload a firmware file xxxxxxxxxxxxxxxxxxxxxxxxxxx.bin (ota update)

Renamed to shortname.bin no problem with short file names.

earlephilhower, 2020-01-01:
This would be a mkspiffs issue, if it exists.  There are tests for this specific case on in the host FS testing.  Closing as very old and not core related.  If someone does find this issue in the current mkspiffs tool, please open an issue on the mkspiffs repo or the esp-quick-toolchain.

closed 2020-01-01 by earlephilhower