esp8266/Arduino #2325: Disable LED Flash at Startup / Serieal Output

p1337, 2016-07-24:
Hello there, it´s not really an issue, but something that costs energy for nothing.

If we restart the ESP8266 (12-e) then the LED flashes 1 time at the boot up because there is an serial output at 74880 baud:

` ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x42
csum 0x42
~ld 
`

Could you make variable, that eliminates the output at the startup?
Or, where could I find the code to delete the serial output part?

Testato, 2016-07-27:
+1

Not only for the energy problem, but also in all the project where you need drive another system by serial, the boot text may create problem and start ghost commands

igrr, 2016-08-01:
All lines except for the last one (`~ld`) come from bootloader which is in ROM of ESP8266. Because it's in ROM, there is no way to change or disable that output.

@Testato if you need to keep serial output clean (to talk to a GPS unit, for instance), you may connect it to pins 13 and 15, and call `Serial.swap();` after `Serial.begin`. UART will be remapped to pins 13 and 15, and you won't get any garbage there.

closed 2016-08-01 by igrr