esp8266/Arduino #2327: Low wifi bandwith, normal?

jej, 2016-07-25:
Hi,

I am testing a Wemos Mini with the FSBrowser example (2.3.0). When downloading a file, I get a stable bandwith of 35kb/s.

I tried WIFI_AP and WIFI_STA modes, different browsers, to change some parameters like 80/160 Mhz, etc, I got the same result for ever. The board is very close from the computer and connected with USB. I would expect a bit more...

Is it a normal behaviour?

jej, 2016-07-26:
I answer myself. I use now the async webserver and it's far quicker and well implemented... Now download is about 500kb/s.

https://github.com/me-no-dev/ESPAsyncWebServer

igrr, 2016-07-26:
You may also want to try the latest git version. WiFiClient has been optimized a bit, streaming from  a file is now much faster than it used to be. Async web server will still give you higher performance though because it can handle multiple requests in parallel.

jej, 2016-07-26:
Yes, I am very satisfied also with the async web design pattern.

Is WifiClient involved in AP mode?

closed 2016-08-01 by igrr