esp8266/Arduino #2329: Time.h or DS1307RTC.h not compatible?

befunhung, 2016-07-26:
#### Hardware
Hardware: NodeMCU Dev Kit 0.9
Core Version: Arduino IDE 1.6.9

### Description

Compiled w/o error but output weird message in serial monitor

### Settings in IDE

Module:  			?NodeMCU 0.9(ESP-12 Module)?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			??
Flash Frequency:	?80Mhz?
Upload Using:		?SERIAL?
Reset Method:		?nodemcu?

### Sketch

#include <Time.h> 
#include <Wire.h> 
#include <DS1307RTC.h>  // a basic DS1307 library that returns time as a time_t
char *dayOfWeek[] = {"", "Sun", "Mon", "Tue", "Wed", "Thu", "Fri", "Sat"};
void setup()  {
  Serial.begin(9600);
  setSyncProvider(RTC.get);   // the function to get the time from the RTC
  if(timeStatus()!= timeSet)
     Serial.println("Unable to sync with the RTC");
  else
     Serial.println("RTC has set the system time");     
}

void loop()
{
   digitalClockDisplay(); 
   delay(1000);
}

void digitalClockDisplay(){
  // digital clock display of the time
  Serial.print(year());
  Serial.print("-");
  if (month() < 10) {
    Serial.print('0');
  }
  Serial.print(month());
  Serial.print("-");
  if (day() < 10) {
    Serial.print('0');
  }
  Serial.print(day());
  Serial.print(" ");
  Serial.print(dayOfWeek[weekday()]);
  Serial.print(' ');
  if (hour() < 10) {
    Serial.print('0');
  }
  Serial.print(hour());
  Serial.print(':');
  if (minute() < 10) {
    Serial.print('0');
  }
  Serial.print(minute());
  Serial.print(':');
  if (second() < 10) {
    Serial.print('0');
  }
  Serial.print(second());
  Serial.println();
}

### Debug Messages

Compiled w/o error while output to serial monitor :

est Jan  8 2013, rst cause:2, boot mode:(3, 6)

load 0x4010f000, len 1834, room 16
tail 8
chksum 0x2d
csum 0x2d
v60000318
~ld

### The following sketch compiled and run w/o fail

#include "Wire.h"
#define DS1307_I2C_ADDRESS 0x68

char *weekDay[] = {"", "Sun", "Mon", "Tue", "Wed", "Thu", "Fri", "Sat"};

// Convert normal decimal numbers to binary coded decimal
byte decToBcd(byte val)
{
  return ( (val/10*16) + (val%10) );
}

// Convert binary coded decimal to normal decimal numbers
byte bcdToDec(byte val)
{
  return ( (val/16*10) + (val%16) );
}

/*
// 1) Sets the date and time on the ds1307
// 2) Starts the clock
// 3) Sets hour mode to 24 hour clock
// Assumes you're passing in valid numbers
void setDateDs1307(byte second,        // 0-59
                   byte minute,        // 0-59
                   byte hour,          // 1-23
                   byte dayOfWeek,     // 1-7
                   byte dayOfMonth,    // 1-28/29/30/31
                   byte month,         // 1-12
                   byte year)          // 0-99
{
   Wire.beginTransmission(DS1307_I2C_ADDRESS);
   Wire.write(0);
   Wire.write(decToBcd(second));    // 0 to bit 7 starts the clock
   Wire.write(decToBcd(minute));
   Wire.write(decToBcd(hour));      // If you want 12 hour am/pm you need to set
                                   // bit 6 (also need to change readDateDs1307)
   Wire.write(decToBcd(dayOfWeek));
   Wire.write(decToBcd(dayOfMonth));
   Wire.write(decToBcd(month));
   Wire.write(decToBcd(year));
   Wire.endTransmission();
}
*/

// Gets the date and time from the ds1307
void getDateDs1307(byte *second,
          byte *minute,
          byte *hour,
          byte *dayOfWeek,
          byte *dayOfMonth,
          byte *month,
          byte *year)
{
  // Reset the register pointer
  Wire.beginTransmission(DS1307_I2C_ADDRESS);
  Wire.write(0);
  Wire.endTransmission();
  
  Wire.requestFrom(DS1307_I2C_ADDRESS, 7);

  // A few of these need masks because certain bits are control bits
  *second     = bcdToDec(Wire.read() & 0x7f);
  *minute     = bcdToDec(Wire.read());
  *hour       = bcdToDec(Wire.read() & 0x3f);  // Need to change this if 12 hour am/pm
  *dayOfWeek  = bcdToDec(Wire.read());
  *dayOfMonth = bcdToDec(Wire.read());
  *month      = bcdToDec(Wire.read());
  *year       = bcdToDec(Wire.read());
}

void setup()
{
  byte second, minute, hour, dayOfWeek, dayOfMonth, month, year;
  Wire.begin();
  Serial.begin(115200);
  
  // Change these values to what you want to set your clock to.
  // You probably only want to set your clock once and then remove
  // the setDateDs1307 call.
  // second = 45;
  // minute = 06;
  // hour = 15;
  // dayOfWeek = 6;
  // dayOfMonth = 22;
  // month = 7;
  // year = 16;
  // setDateDs1307(second, minute, hour, dayOfWeek, dayOfMonth, month, year);
}

void loop()
{
  dispDateTime();
  delay(1000);
}

void dispDateTime() {
  byte second, minute, hour, dayOfWeek, dayOfMonth, month, year;

  getDateDs1307(&second, &minute, &hour, &dayOfWeek, &dayOfMonth, &month, &year);
  Serial.print("20");
  printDigits((int) year);
  Serial.print("/");
  printDigits((int) month);
  Serial.print("/");
  printDigits((int) dayOfMonth);
  Serial.print(" ");
  Serial.print(weekDay[dayOfWeek]);
  Serial.print(" ");
  printDigits((int) hour);
  Serial.print(":");
  printDigits((int) minute);
  Serial.print(":");
  printDigits((int) second);
  Serial.println("");
}

void printDigits(int digits) {
  // utility for digital clock display: prints leading 0
  if (digits < 10)
    Serial.print('0');
  Serial.print(digits);
}

befunhung, 2016-07-26:
### include file missing
#include <Time.h> 
#include <Wire.h> 
#include <DS1307RTC.h>  // a basic DS1307 library that returns time as a time_t

befunhung, 2016-07-26:
include file missing

Time.h
Wire.h
DS1307RTC.h

devyte, 2017-10-14:
@befunhung I don't understand this issue. Please explain better.
Is the issue still valid in latest git?

earlephilhower, 2019-01-16:
We don't ship the DS1307RTC library so there's nothing we can do.  Closing this as a 3rd party issue.

closed 2019-01-16 by earlephilhower