esp8266/Arduino #2330: Esp8266 IP Address not reachable after a while

thehellmaker, 2016-07-26:
Hi All,
ESP abecomes unavailable after sometime intermittently where it says 
Connection to http://192.168.1.4:80 refused
org.apache.http.conn.HttpHostConnectException: Connection to http://192.168.1.4:80 refused

The same happens from the web browser and then it starts working randomly. Checking the logs of the ESP device itself there is no crash.

Am i missing something in the setup of the server which can keep it alive all the time.

I thought this was a webserver problem  but it seems like and ESP issue.
Here is the related issue https://github.com/me-no-dev/ESPAsyncWebServer/issues/54 

Cheers,
Akash A

thehellmaker, 2016-07-26:
A similar issue https://github.com/esp8266/Arduino/issues/1137 was reported on dec 2015

thehellmaker, 2016-07-26:
A similar issue also reported in 
http://internetofhomethings.com/homethings/?p=426

thehellmaker, 2016-07-26:
As mentioned in https://github.com/me-no-dev/ESPAsyncWebServer/issues/54 I have already tried the approach in the link http://www.esp8266.com/viewtopic.php?p=12809 and its still not working.

Now will analyse using wireshark myself

thehellmaker, 2016-07-26:
Wireshark is recieving ARP broadcast from the module every second because of the fix. 

Here is the packet content.
100	7.995514	Espressi_1a:66:47	Broadcast	ARP	42	Gratuitous ARP for 192.168.1.6 (Request)
Frame 100: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
Ethernet II, Src: Espressi_1a:66:47 (5c:cf:7f:1a:66:47), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
    Destination: Broadcast (ff:ff:ff:ff:ff:ff)
        Address: Broadcast (ff:ff:ff:ff:ff:ff)
        .... ..1. .... .... .... .... = LG bit: Locally administered address (this is NOT the factory default)
        .... ...1 .... .... .... .... = IG bit: Group address (multicast/broadcast)
    Source: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        Address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
Address Resolution Protocol (request/gratuitous ARP)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    [Is gratuitous: True]
    Sender MAC address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
    Sender IP address: 192.168.1.6
    Target MAC address: 00:00:00_00:00:00 (00:00:00:00:00:00)
    Target IP address: 192.168.1.6

thehellmaker, 2016-07-26:
For a device to which ARP is responding here is the sequence 

1. Request 
1732	208.713855	IntelCor_c5:37:30	Espressi_1a:66:47	ARP	42	Who has 192.168.1.6? Tell 192.168.1.7
2. Response
1733	208.734013	Espressi_1a:66:47	IntelCor_c5:37:30	ARP	42	192.168.1.6 is at 5c:cf:7f:1a:66:47

1. Request Body
Frame 1732: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
    Interface id: 0 (\Device\NPF_{641ED2C7-4125-43D0-BEF1-205ACE40B627})
    Encapsulation type: Ethernet (1)
    Arrival Time: Jul 26, 2016 21:05:15.359512000 India Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1469547315.359512000 seconds
    [Time delta from previous captured frame: 0.374458000 seconds]
    [Time delta from previous displayed frame: 0.374458000 seconds]
    [Time since reference or first frame: 208.713855000 seconds]
    Frame Number: 1732
    Frame Length: 42 bytes (336 bits)
    Capture Length: 42 bytes (336 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:arp]
    [Coloring Rule Name: ARP]
    [Coloring Rule String: arp]
Ethernet II, Src: IntelCor_c5:37:30 (18:5e:0f:c5:37:30), Dst: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
    Destination: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        Address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
        Address: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
Address Resolution Protocol (request)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: request (1)
    Sender MAC address: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
    Sender IP address: 192.168.1.7
    Target MAC address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
    Target IP address: 192.168.1.6
2. Response Body
Frame 1733: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
    Interface id: 0 (\Device\NPF_{641ED2C7-4125-43D0-BEF1-205ACE40B627})
    Encapsulation type: Ethernet (1)
    Arrival Time: Jul 26, 2016 21:05:15.379670000 India Standard Time
    [Time shift for this packet: 0.000000000 seconds]
    Epoch Time: 1469547315.379670000 seconds
    [Time delta from previous captured frame: 0.020158000 seconds]
    [Time delta from previous displayed frame: 0.020158000 seconds]
    [Time since reference or first frame: 208.734013000 seconds]
    Frame Number: 1733
    Frame Length: 42 bytes (336 bits)
    Capture Length: 42 bytes (336 bits)
    [Frame is marked: False]
    [Frame is ignored: False]
    [Protocols in frame: eth:ethertype:arp]
    [Coloring Rule Name: ARP]
    [Coloring Rule String: arp]
Ethernet II, Src: Espressi_1a:66:47 (5c:cf:7f:1a:66:47), Dst: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
    Destination: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
        Address: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Source: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        Address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
        .... ..0. .... .... .... .... = LG bit: Globally unique address (factory default)
        .... ...0 .... .... .... .... = IG bit: Individual address (unicast)
    Type: ARP (0x0806)
Address Resolution Protocol (reply)
    Hardware type: Ethernet (1)
    Protocol type: IPv4 (0x0800)
    Hardware size: 6
    Protocol size: 4
    Opcode: reply (2)
    Sender MAC address: Espressi_1a:66:47 (5c:cf:7f:1a:66:47)
    Sender IP address: 192.168.1.6
    Target MAC address: IntelCor_c5:37:30 (18:5e:0f:c5:37:30)
    Target IP address: 192.168.1.7

thehellmaker, 2016-07-26:
Found a very interesting thing.
I am using a Windows 7 OS to debug this issue and here are the findings
1. ESP is responding to ARP queries where destination is the ESP MAC address
1918	151.364565	IntelCor_c5:37:30	Espressi_1a:66:47	ARP	42	Who has 192.168.1.6? Tell 192.168.1.7
1919	151.371335	Espressi_1a:66:47	IntelCor_c5:37:30	ARP	42	192.168.1.6 is at 5c:cf:7f:1a:66:47

2. ESP is not responding to broadcast ARP pings using nmap. 
3459	254.010073	IntelCor_c5:37:30	Broadcast	ARP	42	Who has 192.168.1.6? Tell 192.168.1.7

I will look into the arp query responder code in the codebase

thehellmaker, 2016-07-26:
Looks like the arp requests are completely handled by lwIP project which is what this project is depenent on.
@me-no-dev looks like you imported the project as dependency 4 months back. And i did a diff with the latest version of the project 1.4.1 of lwIP and seems like some broaddcast functionality was added which is not there in the version imported. Did you import the latest version ?

me-no-dev, 2016-07-26:
lwip comes from espressif and not me :) I just tweaked some stuff here and there (not broadcast but multicast). Latest lwip is wip :)

thehellmaker, 2016-07-28:
Upgrade to open source Lwip(1.4.1) from 1.3.2 port as suggested by @igrr  The module is still responding to ARP requests.. Waiting to see if it stops.

igrr, 2016-08-01:
Can you make a diff between 1.3.2 and 1.4.1 in the part which deals with ARP? Maybe we can backport the fix instead of updating all of lwip for now.

thehellmaker, 2016-08-02:
Stopped responding to ARP requests on 1.4.1 as well. 
The gratuitous ARP that is being sent is not being handled by android devices. Deep diving into the code base to debug further.

51056	1693.352539	Espressi_88:7f:7e	Broadcast	ARP	42	Gratuitous ARP for 192.168.1.12 (Request)
Frame 51056: 42 bytes on wire (336 bits), 42 bytes captured (336 bits) on interface 0
Ethernet II, Src: Espressi_88:7f:7e (5c:cf:7f:88:7f:7e), Dst: Broadcast (ff:ff:ff:ff:ff:ff)
Address Resolution Protocol (request/gratuitous ARP)

thehellmaker, 2016-08-02:
How ever restarting the module takes a new ip address and it starts responding to ARP requests.

thehellmaker, 2016-08-07:
Now I am seeing that the IP Address is also in use by another device which is obvious as ESP didn't respond to ARP request. But ESP has been sending gratuitous ARP and here is the wire shark capture

5495591	37123.051429	00:e1:40:46:09:6c	Broadcast	ARP	42	Gratuitous ARP for 192.168.1.5 (Request) (duplicate use of 192.168.1.5 detected!)

thehellmaker, 2016-08-12:
This is not an issue with ARP as most people have pointed out. This has something to do with the wireless connectivity stability.

I see debug logs right after the module stops responding to ARP saying
wifi evt: 7
add 1
aid 1
station: 40:88:05:b1:29:eb join, AID = 1
wifi evt: 5
wifi evt: 7
bcn_timout,ap_probe_send_start

This seems to be the root cause. I have attached the full log here. 
https://drive.google.com/open?id=0B8DXcb9GfNuARW54YWFsVHhJbnc

thehellmaker, 2016-08-12:
Attaching Enums that the event numbers point to 
https://github.com/esp8266/Arduino/blob/db5e20f23770e1be307348633dc497f689493996/tools/sdk/include/user_interface.h ( line 368)
https://github.com/esp8266/Arduino/blob/de166c9dd73bd1da0baa35b2a62695035196018a/libraries/ESP8266WiFi/src/ESP8266WiFiType.h (Line 51)

Both map to same enum values..

mtnbrit, 2016-08-12:
What make, model and firmware is your AP? Have you tried a different brand or model of wifi AP? they are not all equal by far.

ClaudioHutte, 2016-08-12:
So you think the wifi connectivity instability lead to the inability of responding ARP broadcast requests?
A reception problem since transmission seems to be ok, correct?

thehellmaker, 2016-08-13:
@ClaudioHutte You are partially right. Here are my observations

1. Reciever seems to be mainly affected because post this gratuitous ARP from other module is not being recieved as well but Gratuitous ARP is being sent to other modules though

2. How ever if you see the log below it seems like... The module tries to rejoin it gets a wifi evt: 5 which is connected post which it recieves the Gratuitous ARp from other modules for just a few seconds post which it disconnects with 
```
err already associed!
station: 98:0c:a5:b8:de:91 leave, AID = 1
```
Log
```
add 1
aid 1
station: 98:0c:a5:b8:de:91 join, AID = 1
wifi evt: 5
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
Got ARP Input 
nHere for arpwifi evt: 7
Got ARP Input 
nHere for arpGot ARP Input 
nHere for arpwifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
Got ARP Input 
nHere for arpGot ARP Input 
nHere for arpGot ARP Input 
nwifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
Got ARP Input 
nHere for arpGot ARP Input 
nHere for arpGot ARP Input 
nHere for arpwifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
Got ARP Input 
nHere for arpwifi evt: 7
Got ARP Input 
nHere for arpGot ARP Input 
nHere for arpwifi evt: 7
Got ARP Input 
nHere for arpwifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
err already associed!
station: 98:0c:a5:b8:de:91 leave, AID = 1
rm 1
wifi evt: 6
add 1
aid 1
station: 98:0c:a5:b8:de:91 join, AID = 1
```

3. somewhere between multiple join and leave attempts you'll also see 
`max connection!`

4. And ofcourse a bunch of 

```
bcn_timout,ap_probe_send_start
bcn_timout,ap_probe_send_start
```

Just to explain my setup I have 2 esp8266 12 f modules 
http://www.thaieasyelec.com/products/wireless-modules/wifi-modules/esp8266-12f-wifi-serial-transceiver-module-detail.html 

1. I setup gratuitous ARP to send arp broadcast pings into the network every second as @ClaudioHutte  pointed out in the beginning
2. When the modules connect for the first time every second the module prints 
```
Got ARP Input 
nHere for arp
```
3. Along with this ARP recieve there are other wifi events. 

4. At some point the logs mentioned in point 2 stop. (After close to 48 hours) and there are a bunch of other events which happen before this terminates.

I have attached the complete log to the link 
https://drive.google.com/file/d/0B8DXcb9GfNuARFZGdy1USGNPbFk/view

ClaudioHutte, 2016-08-13:
I never tested two units as you done, though I incurred into the same troubles with ESP8266-12 and a TP-link router located quite far (two stories below mine). I would like to do some tests the same way you've done, but I will be busy into other works for the next two weeks.
What happens if the "every second gratuitous ARP send" workaround is stopped/skipped?

thehellmaker, 2016-08-13:
Before you mentioned about the gratuitous ARP i hadn't added it into the code base. Even then the module stopped responding like we discussed here https://github.com/me-no-dev/ESPAsyncWebServer/issues/54 

But i haven't collected the logs without Gratuitous ARP but I'm sure its the same issue though.

For the module to eventually stop responding it always take 36 hours + .

alirezza, 2017-05-08:
By upgrading to SDK 2.1.0  #3215  this problem will be solved.

pouriap, 2017-06-20:
Thanks for the efforts to create the `update_sdk_2.1.0` branch. 

But I'm still having the ARP issue even when using that branch:

[image]

Can anyone confirm that their ARP issue has been resolved by using that branch?

IvanBayan, 2017-06-20:
I tried new sdk and still have arp issue.

vks007, 2017-06-25:
I also have this same issue. 
I am using a webserver on the ESP which connects to my router in the STA mode. The router assigns a fix IP to the ESP (192.168.1.54). All works good but after some time (typically a few hours to a day) the ESP webserver stops responding. I tried pinging the IP address at this point and its unreachable. To see memory footprint I added log calls within the ESP which calls a googlesheet URL and logs all relevant info. All that keeps working fine. Memory foot print is also normal. So while my ESP is able to reach the internet it's IP address is not reachable from within the local network. 
If I reset my ESP or turn my modem ON/OFF (to again assign the IP address) the issue goes away for a few hours.
I have tried the simple webserver from the examples and it behaves the same way so my program is not what is causing this. I have also tried this on a SONOFF,Electrodragon ESP relay module, ESP 01 module , ESP 12E module - they all behave the same.
Can somebody guide me on what should I be looking at here,

mtnbrit, 2017-06-25:
Can you ping the ESP from the router itself? 

It looks like the arp issue.

Have you eliminated the access-point/router as being the cause by trying a different make/model?

vks007, 2017-06-26:
Hi @mtnbrit , I am in the process of trying out a different router, will a few days to verify this. 
Also, while the above condition happened some times, with my testing since yday most of the times, I am able to ping the ESP while the webserver does not respond. At other times, I am sometimes able to get a response from my iPhone browser while the desktop browser throws - connection reset error. 
I also, started recycling the webserver every 10 min (I mean recreate the webserver object via reset method) and I was still able to get into a state where the ESP behaves normally but webserver stops responding. 
Is there someway I can debug the state of the webserver object, I can log that and figure out what is not responding. Maybe I can tweak the source files and read some members and log them, if they are private, make them public just for this purpose. But I am not sure what things would tell me something about the webserver object.

vks007, 2017-07-11:
An update on my issue: I am not sure how does ARP work and what can be done about it but I changed by router and my ESP has been working perfect since then. Its been 4 days and it has not been unreachable even once. So there is something with my existing router that causes it to be unreachable. What a waste of time this has been for me. I have been trying to implement a solution to this for many weeks now. Phew!

mikrodunya, 2017-08-17:
Hi all. I have the same issue with ESP8266 webserver example. I can see ESP8266 MAC address on modem's page but i can't see it's IP address. Also i cant reach ESP8266's IP over browser. How can i solve this problem?
Thanks.

vks007, 2017-08-19:
Hi @mikrodunya , I would suggest you try another router and see if the issue persists. If you dont have one, loan it form a friend for a day or two :) . It is certianly a ARP issue with the router.

mikrodunya, 2017-08-19:
I am suspected my router too. I will try it on another router. 
Thanks.

lexelby, 2017-08-20:
I have seen improvement by having my esp devices connect to an access point running on my computer using hostapd.

That said, I still maintain that we can't just take the easy road here and blame the router.  _only_ esp8266 devices connected to my router exhibit any arp issues.

lexelby, 2017-08-20:
... And by "still maintain", I mean that I mentioned that in another similar issue here -- I forget which one. :)

devyte, 2017-10-19:
All, does PR #3362 help with this?

pouriap, 2017-10-21:
I completely fail to understand why this issue is not being addressed.
Is it a rare thing? Is my ESP device the problem? Will buying another one solve this?
I have been waiting for a fix for almost a YEAR now. Struggling with this completely broken device.
It simply does not answer ARP request. I cannot reach it half of the time. I have to refresh my browser for minutes until it finally responds.
You cannot blame the router for this because there are like 10 wifi devices in my house connected to this router all the time and none of them have ever had a similar problem.
I've tried many solutions during this one year but none of them work. The ESP just stops responding to ARP and it's completely random. Sometimes right after I turn it on it will be unresponsive. Sometimes it will take a while. 
How can this be my router's fault? How is it possible that apparently very few people are having this issue?

d-a-v, 2017-10-21:
@pouriap PR #3362 is an update to lwip-v2. Could you try it and report here ?

pouriap, 2017-10-21:
@d-a-v Thank you for answering.
I'm not sure how I should use a pull request. I searched your repositories and found your [lwip2v2 ](https://github.com/d-a-v/Arduino/tree/lwip2v2)branch. Is that what I'm supposed to download?
(BTW am I doing this right? Shouldn't there have been a link to it in the pull request? It took me a while to find it).
Also, I already have the `esp8266 v2.3.0` board installed via the boards manager. Do I have to build/make anything? Or do I just copy your branch to the esp8266 folder and overwrite it?

d-a-v, 2017-10-22:
Cloning my lwip2v2 branch would do it for a try.

d-a-v, 2017-10-22:
But if you are using v2.3.0 from board manager, you should try current master or v2.4.0rc2 first ([link](https://github.com/esp8266/Arduino/releases))

pouriap, 2017-10-22:
Thanks @d-a-v 
Just flashed with v2.4.0rc2 with no luck. Became unresponsive after ~1 hour.
Going to try with your lwip2v2 branch.

mtnbrit, 2017-10-22:
It might be useful to start building a list of access points that have this issue, would you care to report the make model and firmware version? Have you tried a different AP?

pouriap, 2017-10-22:
@mtnbrit That's exactly what I was thinking about today. I thought maybe someone would be able to see a pattern in the routers and figure out the problem. We should post our WiFi settings as well. Like encryption method, etc.
I'm not familiar with github workflow though, so I'm not sure how/where we should create this list.
Here is my router model and settings:
(I think for people with publicly accessible IPs posting their firmware version could be a security issue)
>**Huawei D-100 3G/4G Desktop Modem**
Firmware version: 01.01.02.082
Radio Channel: 			Auto
Working Mode: 			802.11b/g/n
Bandwidth:				20M/40M
RTS Threshold:			2347
CTS Protection Mode:	Auto
Preamble Length: 		Short Preamble
SSID Broadcast:			Enable
Authentication:			WPA/WPA2-Personal Mixed Mode
Encryption:				TKIP/AES

Though the Huawei D-100 that comes up on the internet is not the one I have. I think it might be a model specific to my ISP? (Irancell)

I have not tried a different AP because it would be a useless thing to do. Even if it worked with another router(which I suspect it would given how rare this issue is), I can't afford to buy a new router just because the ESP doesn't work with my current one. Also I'm pretty sure my ISP wouldn't let me use 3rd party hardware.
I guess I'll just have to keep refreshing until it responds unless the cure is found. 😞 

Speaking of which, I think the **lwip2v2** isn't fixing it either. I flashed it this afternoon and it has gone into brief period of unresponsiveness. Its behavior is always inconsistent so I cannot tell for sure until a few days have passed. But my intuition and past experiences tell me that it's going to go fully unresponsive by tomorrow or in a few days.

jogyl, 2017-10-23:
@mtnbrit I have the same issue as @pouriap and others (but mine are all gone once the webserver stops responding) . My webbserver stops responding after a couple of minutes but it is still responsive using serial and UDP broadcasts. I have tried 2.4.0-rc2 down to 2.2.0 with the same results. I am running 4 Node MCU ESP-12 that goes blackout regardless of how they are powered (USB or local PSU).

I am running 2 Unify access points in mesh with a Unify router (EdgeRouter PoE v1.9.7+hotfix.4):
model: UAP
version: 3.9.3.7537
radio: 802.11b/g/n
channel: 6 and 11

I will setup a second wifi using some other Netgear router and post back my results.

vks007, 2017-10-23:
@jogyl , I have tried every possible combination for my ESPs , powering it differently etc but I was always able to get this issue with my DLink router. While everything works fine with my TP Link router. I have been running on my TP Link router for many months now and I have never had any issues. 
One important test I did was to reach out to the internet form the ESP. My ESP was able to continuously reach the internet while it was not reachable from the local network. This indicates that the issue isnt with the web server but something between the ESP and the router. 
I havent tried tweaking the DLink router settings to see if some setting solves the issue with the ESP - its too time consuming to try each setting for a couple of hours.

jogyl, 2017-10-23:
@vks007, ok. You have problems using some DLink router, @pouriap is on a Huawei and I am using products from Ubiquiti and there are others in other forums. It seems like our devices still can send data (you using TCP connecting to an external service and me using UDP). Our devices stops responding to networks request (regardless of where in the network stack the problem is located).

The solution cannot be that we all get TP Link routers, can we at least agree upon that there seems to be some problem and try to work to isolate it? 

DLink, Huawei and Ubiquity are all pretty large companies, going for that there is some special way that these companies have implemented their networking is what kills the ESPs communication does not feel right. Is there some way we can do structured testing using same versions of the firmware and using the same sketch etc? @igrr and co, is there any way we can help out and give better feedback?

pouriap, 2017-10-24:
I can confirm now that the **lwip2v2** is not fixing the issue.

[image]

FYI this has been captured from a third machine, not the one sending the requests. So the router is actually sending the broadcast across the network but the ESP isn't responding to it.

pouriap, 2017-10-24:
A question: Do your ESPs respond after a while? Because mine does respond after I keep pinging it(or keep hitting the reload button in the browser). Sometimes it takes 30 seconds, sometimes one minute, sometimes more, but it does respond eventually. In the first image in one of my earlier posts you can see the ESP responding eventually in the highlighted row, but it has taken it one minute to do so.
Do yours do the same thing?

jp112sdl, 2017-10-24:
@pouriap I can confirm the behaviuor you described.

My ESPs always responds to a ping, when I keep pinging it continously. 

If I stop pinging and waiting for a while (don't know exactly how long... a few hours or so), the ESP is unreachable for the first 10 or 20 pings. Then the ESPs answers with a very high response time (> 300ms) for some replies before it is getting to a normal level.

While ping is not available, the webserver running on the ESP is not responding, too.

FYI: I am using an Apple TimeCapsule as AP.

Now I'm pinging my ESPs with nagios every 5 minutes. Seems to keep them alive.

mikrodunya, 2017-10-24:
Same problem here.

jogyl, 2017-10-24:
@pouriap, my ESPs allways stops respoding after little less then 2 minutes and they never wake up. They are still broadcasting their UPD presence with no interrupts.

@jp112sdl, I tried your ping-thing but it made no difference on my network. After they stop responding I get 15 "...unreachable" and 3 "...timeout" over an over (ping -t in Windows).

@vks007, I have set up another network (a Windows 10 laptop as mobile hotspot) and connected an ESP to that network. It has been responding fine for a couple of hours so the error is indeed network dependent.

I am not good at packet logging but if there is anything I can setup to log and compare now that I have two networks with the same sketches and firmware running on two different networks with very dramatically different results, I'd be happy to help out...?

pouriap, 2017-10-24:
So a very curious thing just happened.
I was trying to manually send an ARP request to the ESP, only I couldn't!
It does not answer to my manual ARP requests at all. And the only reason it works after a restart is because when it is starting up it sends a gratious ARP and my computer learns it's MAC address.
I'm confused, because when I do the manual ARP no matter how long I keep requesting, it does not answer. But when I ping it, it does answer eventually. 
Now the interesting part: I have another ESP, and this other ESP **does** respond to my manual ARP requests! I'm flashing the same code into them. The example web server.

Can you guys send manual ARP requests to your device and see if it responds?
You can do it either using the Linux utility `arping`:
`arping IP-OF-ESP`
Or using the `nping` utility, which is part of `nmap`. ([Available ](https://nmap.org/download.html)for Linux and Windows):
` nping --arp arp --arp-target-ip IP-OF-ESP IP-OF-ROUTER`

As far as I can remember this other ESP also goes unresponsive after a while, but now I'm not so sure. I'm going to leave it on for a few days to see if it goes unresponsive.

pouriap, 2017-10-24:
@d-a-v how do I enable debug output for lwip?
Now that I can reproduce 100% of the time with manual ARP requests, I'm trying to enable the debug output for `etharp.c`. It has interesting debug messages that could possibly help us trace the cause. But when I add `#define LWIP_DEBUG 1` to `lwipopts.h` I get this error when compiling the sketch:
>
		In file included from C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/arch.h:43:0,

		from C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/debug.h:35,

		from C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/opt.h:46,

		from C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiServer.cpp:35:

		C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiServer.cpp: In member function 'int8_t WiFiServer::_accept(tcp_pcb*, int8_t)':

		C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/arch/cc.h:80:45: error: 'ETS_ASSERT' was not declared in this scope

		 #define LWIP_PLATFORM_ASSERT(x) ETS_ASSERT(x)

													 ^

		C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/debug.h:66:3: note: in expansion of macro 'LWIP_PLATFORM_ASSERT'

		   LWIP_PLATFORM_ASSERT(message); } while(0)

		   ^

		C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0/tools/sdk/lwip/include/lwip/tcp.h:335:36: note: in expansion of macro 'LWIP_ASSERT'

		 #define          tcp_accepted(pcb) LWIP_ASSERT("pcb->state == LISTEN (called for wrong pcb?)", \

											^

		C:\Users\Pouria\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.3.0\libraries\ESP8266WiFi\src\WiFiServer.cpp:155:5: note: in expansion of macro 'tcp_accepted'

			 tcp_accepted(_pcb);

			 ^

Couldn't find anything on the internet about this.

d-a-v, 2017-10-24:
in ```lwipopts.h```,
```
#define LWIP_DBG_TYPES_ON               LWIP_DBG_ON
```
and at least for ARP:
```
#define ETHARP_DEBUG                    LWIP_DBG_ON
```
add this too:
```
#define ETS_ASSERT(x)   do { os_printf("Assertion \"%s\" failed at line %d in %s\n", x, __LINE__, __FILE__); *(int*)0=0; } while(0)
```

IvanBayan, 2017-10-24:
I see that most peoples which have that issue have it on ESP with webserver, i have that issue when ESP works as simple client which just send data over TCP once per 5 min. 
When it happens i see that my server (which placed in same lan) is lost HW address of ESP ($ ip neighbour).
ESP trying to send data, but TCP connection can't be established, because server can't find HW address of ESP (it sends ARP requests but ESP doesn't answer on them).

pouriap, 2017-10-24:
@d-a-v Thank you. I was missing the last part.
It compiled but I can't get anything meaningful from it. When I only enable the debug for `ETHARP_DEBUG` nothing is sent to the serial output. 
I then turned on `SOCKETS_DEBUG`, `ICMP_DEBUG`, `INET_DEBUG`, `IP_DEBUG`, `TCP_DEBUG`, and a bunch of others. Then I pinged the device and opened a web page on the device but still nothing was being sent to serial output.
I tried turning all the debug modes on, which resulted in a LOT of data being logged to the serial output continuously, but it's garbled text. Similar to when the baud rate is not correct. I tried all standard baudrates but it's still garbled text. I tried adding `#define UART_REDIRECT 0` as instructed [here](https://community.nxp.com/docs/DOC-330865) but it didn't help either. I tried removing serial output in my sketch but it didn't help either.

Is there anything else I need to do to get the debug output on the serial monitor?

jogyl, 2017-10-24:
@pouriap, this is me nping-ing one of my ESPs just around the time it fails. First one sucessfull ping (after a reset of the device) and then seconds later a fail. And then its not respodning to anything any more...

C:\...\nmap-7.60>nping --arp arp --arp-target-ip 192.168.2.143 192.168.2.1
Failed to resolve given hostname/IP: arp.  Note that you can't use '/mask' AND '1-4,7,100-' style IP ranges

Starting Nping 0.7.60 ( https://nmap.org/nping ) at 2017-10-24 21:12 W. Europe Daylight Time
SENT (2.5170s) ARP who has 192.168.2.143? Tell 192.168.2.48
RCVD (2.7000s) ARP reply 192.168.2.143 is at 18:FE:34:E0:BD:E9
SENT (3.6330s) ARP who has 192.168.2.143? Tell 192.168.2.48
RCVD (3.7190s) ARP reply 192.168.2.143 is at 18:FE:34:E0:BD:E9
SENT (4.6330s) ARP who has 192.168.2.143? Tell 192.168.2.48
RCVD (4.7410s) ARP reply 192.168.2.143 is at 18:FE:34:E0:BD:E9
SENT (5.6330s) ARP who has 192.168.2.143? Tell 192.168.2.48
RCVD (5.6630s) ARP reply 192.168.2.143 is at 18:FE:34:E0:BD:E9
SENT (6.6330s) ARP who has 192.168.2.143? Tell 192.168.2.48
RCVD (6.6890s) ARP reply 192.168.2.143 is at 18:FE:34:E0:BD:E9

Max rtt: N/A | Min rtt: N/A | Avg rtt: N/A
Raw packets sent: 5 (210B) | Rcvd: 5 (230B) | Lost: 0 (0.00%)
Nping done: 1 IP address pinged in 6.69 seconds

C:\...\nmap-7.60-win32\nmap-7.60>nping --arp arp --arp-target-ip 192.168.2.143 192.168.2.1
Failed to resolve given hostname/IP: arp.  Note that you can't use '/mask' AND '1-4,7,100-' style IP ranges

Starting Nping 0.7.60 ( https://nmap.org/nping ) at 2017-10-24 21:12 W. Europe Daylight Time
SENT (2.5140s) ARP who has 192.168.2.143? Tell 192.168.2.48
SENT (3.6320s) ARP who has 192.168.2.143? Tell 192.168.2.48
SENT (4.6320s) ARP who has 192.168.2.143? Tell 192.168.2.48
SENT (5.6320s) ARP who has 192.168.2.143? Tell 192.168.2.48
SENT (6.6400s) ARP who has 192.168.2.143? Tell 192.168.2.48

Max rtt: N/A | Min rtt: N/A | Avg rtt: N/A
Raw packets sent: 5 (210B) | Rcvd: 0 (0B) | Lost: 5 (100.00%)
Nping done: 1 IP address pinged in 7.64 seconds

d-a-v, 2017-10-24:
@pouriap try with 
```
(LWIP_DBG_ON|LWIP_DBG_TRACE|LWIP_DBG_STATE|LWIP_DBG_FRESH|LWIP_DBG_HALT)
```
instead of just ```LWIP_DBG_ON```

mtnbrit, 2017-10-24:
Some wifi APs have a “kick” function that disconnects a client and forces it to re-associate, can you try that when it loses connectivity and see if it becomes reachable after re-associateing to the AP? 

Or even try restarting the AP, does that get it back pinging again?

jogyl, 2017-10-24:
@mtnbrit, I tried that as my APs has a "controller" sw. When I hit "reconnect" they, don't... I can see how the connection statistics goes away and does not get refreshed and seconds later they are removed from the list of connected clients on that AP. The intresting thing is that it stopped them from doing UDP broadcasts that was working until I "kicked" them.

There is also something that sets them appart from other devices on my wifi, the ESPs are listed as "power save: enabled" no other devices are. On issue #460 I see some discussion about it from the standpoint to have the devices consume as little power as possible. Is this something? That some APs support this mode but they are not waking up as expected and on other networks it's not supported so they just keep on chatting...?

pouriap, 2017-10-24:
@d-a-v That didn't work either. It keeps resetting right after it connects to the AP:

	......
	Soft WDT reset

	ctx: sys 
	sp: 3ffffc30 end: 3fffffb0 offset: 01b0

	>>>stack>>>
	3ffffde0:  00000092 00000017 3fff268c 3ffee878  
	3ffffdf0:  00000000 00000000 00000020 00000000  
	3ffffe00:  3fff31cc 402470a4 3fff2a4c 40227124  
	3ffffe10:  3ffeb0e0 401042a4 3ffee878 3fff2a50  
	3ffffe20:  0002617f 00000000 3fff31cc 40228324  
	3ffffe30:  4021e0a1 40219f4c 3fff2f7c 00000063  
	3ffffe40:  402470a4 3fff2a50 00000080 00000011  
	3ffffe50:  00000000 0b464298 00004978 00000000  
	3ffffe60:  00000000 00000003 00000004 3fff2a50  
	3ffffe70:  3fff31cc 402470a4 3fff319c 40228354  
	3ffffe80:  3fff2a4c 00000000 00000000 3fff2a50  
	3ffffe90:  3fff31cc 402470a4 3fff319c 40226bd4  
	3ffffea0:  3fff2a4c 3fff01e6 3fff00e4 40106abc  
	3ffffeb0:  3fff31cc 3fff2a4c 00000043 3fff3234  
	3ffffec0:  00000001 3fff2a4c 3fff315c 00000000  
	3ffffed0:  3fff266e 3fff2a4c 3fff315c 40222049  
	3ffffee0:  3fff0184 00000000 3fff2a4c 40106abc  
	3ffffef0:  40226d48 0000002c 3fff268c 3ffe8004  
	3fffff00:  402470a8 3fff315c 3fff2a4c 40222fda  
	3fffff10:  3fff0184 000013ca 3fff268c 4021a177  
	3fffff20:  00000006 3fff02c8 00000010 3fff312c  
	3fffff30:  3ffef6d4 3fff2690 4021a1c5 3fff00a0  
	3fffff40:  402185a1 00000009 00000010 00000005  
	3fffff50:  3fff01e6 00000200 40218ba5 3fff00a0  
	3fffff60:  4021950b 00000003 00000008 00000001  
	3fffff70:  3fff2f44 4021c06f 00000000 00000000  
	3fffff80:  3ffee878 4021ba16 3fffdab0 00000000  
	3fffff90:  3fffdcc0 3ffeb1b8 00000000 4020452b  
	3fffffa0:  3ffeb1b8 40000f49 3fffdab0 40000f49  
	<<<stack<<<
	[04]B1…þLG[11]H6Êÿ

Let's review the steps once more. These are the things I'm doing:
- clone the `esp8266-2.4.0-rc2` branch (also tried with 2.3.0 from board manager)
- add/change these in the lwipopts.h file:
`#define LWIP_DEBUG 1 `
`#define ETS_ASSERT(x)   do { os_printf("Assertion \"%s\" failed at line %d in %s\n", x, __LINE__, __FILE__); *(int*)0=0; } while(0)`
`#define LWIP_DBG_TYPES_ON               LWIP_DBG_ON`
`#define ETHARP_DEBUG                    LWIP_DBG_ON`
(Tried with and without the `#define LWIP_DEBUG 1 `)
(Does it matter where in the file I put these? I'm adding them in the DEBUG section of the file)
- `cd tools\sdk\lwip\src`
- `make clean`
- `make install`
- rename and copy the generated `liblwip_src.a` file over the `liblwip_gcc.a` in `tools\sdk\lib`
- compile and upload
(Tried with code that has `Serial.begin(9600)` in it. And with code that doesn't use `Serial`)

Absolutely no debug info is outputted.

d-a-v, 2017-10-24:
- The ```*(int*)0=0``` is supposed to intentionally segfault/reset (trigger gdb). Asserts are not supposed to be raised in a normal behaviour but you can comment this out for a try.
- Also I'm not sure about ```LWIP_DBG_HALT``` (that you don't seem to use) but in case, you can try and keep it out from the other ```LWIP_DBG_*```.
- Does it reset if you are recompiling and using fresh lwip without all those defines ?
- Can you add a simple ```os_printf("hello\n");``` right after ```Serial.begin()``` as a sanity check since this is what ```LWIP_PLATFORM_DIAG(x)``` does (if ```LWIP_DEBUG``` is defined)
- About where, I think ```lwip/include/arch/cc.h``` is the place to be.

[265 more replies]

closed 2019-09-11 by devyte